esp8266/Arduino #1136: undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()

GalloTeo, 2015-12-03:
I struggled a lot around this problem 

code like this 
#include <string>
void pippo () {
	std::string  s; 
}

was generating error 
undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()

in order to fix it i had to force the compiler to generate the string class 
i changed the string file 
hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\string

into 
//

#ifndef _GLIBCXX_STRING
#define _GLIBCXX_STRING	1

#pragma GCC system_header

#include <bits/c++config.h>
#include <bits/stringfwd.h>
#include <bits/char_traits.h>  // NB: In turn includes stl_algobase.h
#include <bits/allocator.h>
#include <bits/cpp_type_traits.h>
#include <bits/localefwd.h>    // For operators >>, <<, and getline.
#include <bits/ostream_insert.h>
#include <bits/stl_iterator_base_types.h>
#include <bits/stl_iterator_base_funcs.h>
#include <bits/stl_iterator.h>
#include <bits/stl_function.h> // For less
#include <ext/numeric_traits.h> 
#include <bits/stl_algobase.h> 
#include <bits/range_access.h>
#include <bits/basic_string.h>
#include <bits/basic_string.tcc> 

// in order to fix this 
// cancme2.ino:(.text._Z5pippov+0x4): undefined reference to `std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()
// i had to force the compiler to generate the concrete class 
template class std::basic_string < char > ;

#endif /* _GLIBCXX_STRING */

and now works. 

i wanted to commit the change but it seems that folder is not under git. how i can i contribute with this fix?
thks

igrr, 2015-12-06:
With the change in c6c7d24, you can actually add this line into your sketch, not the header file.

```c++
#include <string>
template class std::basic_string<char>;
void setup() {
  std::string s = "test";
  Serial.begin(115200);
  Serial.println(s.c_str());
}
void loop() {}
```

Changing libstc++ header files is not a good idea, unless you are actually developing libstdc++.

closed 2015-12-06 by igrr

GalloTeo, 2015-12-07:
hello igrr thank you for you answer, i was not aware of that and i was actually changing other header to fix exactly that. 
What about instead the definition problem ? it seems that defining the classes through typedef does not force compiler to generate the class and i have to specify it. What is your suggestion ?