esp8266/Arduino #1138: Static IP not honored in STA mode unless WiFi.config called AFTER WiFi.begin

drmpf, 2015-12-03:
Test procedure
Flash ESP8266
start as AP for web page config do not set staticIP 
power cycle
connect to AP using DHCP  -> get 10.1.1.7 assigned
power cycle
start as AP for web page config  and set static IP 10.1.1.201 saved in EEPROM
power cycle
connect to AP calling WiFi.config(...) then WiFi.begin
print IP -> 10.1.1.7

however if 
connect to AP calling  WiFi.begin and THEN WiFi.config(...) 
print IP -> 10.1.1.201

see http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html  for sample sketch

Links2004, 2015-12-05:
if tested it with this and its working:

```cpp
/*
 * test_wifi_config.cpp
 *
 *  Created on: 05.12.2015
 */
#include <ESP8266WiFi.h>

void setup() {
    Serial1.begin(921600);

       //Serial.setDebugOutput(true);
       Serial1.setDebugOutput(true);

       Serial1.println();
       Serial1.println();
       Serial1.println();

       Serial1.println("WiFi delete all config...");
       // delete all existing WiFi stuff in EEPROM
       WiFi.disconnect();
       WiFi.softAPdisconnect();
       WiFi.mode(WIFI_OFF);
       delay(500);

       Serial1.println("WiFi set new config...");
       // connect static
       WiFi.mode(WIFI_STA);
       WiFi.config(IPAddress(192,168,1,2),IPAddress(192,168,1,1),IPAddress(255,255,255,0),IPAddress(192,168,1,1));
       WiFi.begin("wifi", "passpasspass");

       Serial1.println("Connecting ...");
       while (WiFi.status() != WL_CONNECTED) {
         delay(500);
         Serial1.print(".");
       }

       Serial1.println("");
       Serial1.println("WiFi connected");
       Serial1.println("IP address: ");
       Serial1.println(WiFi.localIP());

}

void loop() {

}
```

```cpp
WiFi connected 
IP address:  
192.168.1.2 
```

can you post, a simple example to getting the bug?

Note: my WiFi network has a other IP range.

drmpf, 2015-12-05:
Thanks for checking this. Let me get back to you with a smaller code set and after I try 2.0 (may take a while)
Reflashing the board removes the problem. (but keeps the IP setting written to EEPROM)
My code is different in that the switch between AP and STA mode is done via a power cycle reset.
A push button determines which mode is used on start up.
Currently hacking it by calling WiFi.config( ) both before and after WiFi.begin

Links2004, 2015-12-22:
@drmpf have you found the reason?

drmpf, 2015-12-22:
Sorry other bugs of my own making have been keeping me occupied.  
I am currently avoiding this issue by calling WiFi.config(..) AFTER WiFi.begin()

Links2004, 2015-12-30:
shut be fixed with #1323, there where a problem with the modes,
if WiFi STA mode where not enabled before the IP config it ignore the IP configuration.
linked to: #1316

commit 64326f95 (2015-12-30): rework mode management
may fixes: #1138
closed 2015-12-30 by Links2004