esp8266/Arduino #1141: WiFi.softAP(ssid, password) does not seem to work..

shirish47, 2015-12-04:
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>

/* Set these to your desired credentials. */
const char *ssid = "ESPap";
const char *password = "thereisnospoon";

ESP8266WebServer server(80);

/* Just a little test message.  Go to http://192.168.4.1 in a web browser
 * connected to this access point to see it.
 */
void handleRoot() {
	server.send(200, "text/html", "<h1>You are connected</h1>");
}

void setup() {
	delay(1000);
	Serial.begin(4800);
	Serial.println();
	Serial.print("Configuring access point...");
	/* You can remove the password parameter if you want the AP to be open. */
	WiFi.softAP(ssid, password);

	IPAddress myIP = WiFi.softAPIP();
	Serial.print("AP IP address: ");
	Serial.println(myIP);
	server.on("/", handleRoot);
	server.begin();
	Serial.println("HTTP server started");
}
void loop() {
	server.handleClient();
}

I am using the staging version of ESP8266... the thing is that ESP is not showing the name as given ssid.. it shows AI-thinkers.. some thing like that and open network.

igrr, 2015-12-04:
Duplicate of #1123?

shirish47, 2015-12-04:
seems like that.. but what is solutions..??

shirish47, 2015-12-04:
is it not fixable? a hardware issue??

igrr, 2015-12-04:
As mentioned down that link, it is fixed in git and in stable version. Just not yet in staging.

zenmanenergy, 2015-12-05:
It's working for me, but the first request is REALLY slow. It takes 30 seconds for the softap example to work the first time, then every request after that is fast.

shirish47, 2015-12-08:
OK ya stable version works fine.

gosewski, 2016-02-18:
It is pretty obvious but I think it is worth to be mentioned. When the password is too short (less than 8 characters) the WiFi.softAP(ssid, password) function desn't work. Of course there is no warning during compilation.

Links2004, 2016-02-18:
warning during compilation is not possible,
but the function return false (but to most user dont check the return value), 
and when debug is enabled a error will be shown on serial.

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp#L98-L102

gosewski, 2016-02-18:
Yes, you are absolutely right about the warning during the compilation and I fully agree that most users don't check the return value (probably because this return value isn't checked even in the example sketches). Therefore I wrote that it is worth to be mentioned.

closed 2016-02-27 by igrr

khalilst, 2016-06-14:
@gosewski , Your tip solved my problem. Thanks

gosewski, 2016-06-15:
@khalilst ,You're welcome. I am glad to hear that.

NateZimmer, 2016-06-29:
@gosewski  , thanks... had a password of "test" and it wasn't working. Increased the length and it worked just fine. Perhaps update the example sketch with a comment on the password length?

ym58, 2016-11-23:
Sorry to post again on the same issue, but I am also unable to have my ESP8266 actively change its AP SSID/PW.

Needless to say that I have updated the related .h libraries (via the Arduino interface, not from the Github cause there is no button in the Github to download a library .ZIP file !)
My ESP8266WiFi.h is vers 1.0.0
My ESP8266WebServer.h is also 1.0.0
My WiFiClient.h does not show in the Arduino library manager (... !...)

The AP starts up and works quite right but its SSID and PW are never set the way I wan't them to be set !
The code below is only an excerpt of AP-relevant sections
```
 `(......)
WiFiServer server(80);
void setup() 
{
  initHardware();
  setupWiFi();
  server.begin();
}
void loop() 
{
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  // Read the first line of the request
  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();
  (......)
void setupWiFi()
{
  WiFi.mode(WIFI_AP);
  WiFi.softAP("MyNewSSID", "abcdef0123456");
}
  (......)`
```
Unfortunately, the ESP/AP keeps giving me an old SSID/PW set that is written in its memory (EEPROM ?) and does not take into account the "MYNewSSID"/"abcdef0123456" set !

What's wrong ?

MatrixMike, 2016-11-23:
regarding comment about no button to download ZIP file - please look for
GREEN 'CLONE OR DOWNLOAD' button on RHS of screen and nearby you will
find 'Download ZIP' .  On some old browsers or browsers that don't
support this feature is might not be there. I use Firefox and from about
V 45 it has worked well.

ym58, 2016-11-23:
Hello Mike,
Sorry, no such button ...
[image]

MatrixMike, 2016-11-23:
> tmp23

my 'gentle' comment is that you are using a browser that does not take
full advantage (or enough of) the github web smarts.
To prove this to yourself - download and run a late (recent) live ubuntu
or mint iso DVD image and test on your machine -
the button DOES exist - I had the problem until I realised I selected a
browser that works with it.
Q Which browser do you use ?   and for fun which OS / IDE do you use to
develop on?

NateZimmer, 2016-11-23:
Heh... the issue is your not at the top level of the repo. You can't zip just 1 folder to my knowledge. 
https://github.com/esp8266/Arduino <-- should appear at this link.

ym58, 2016-11-24:
@NateZimmer 
You're my man :-) !
The .ZIP can be found only at top level, my mistake ...

danielicy, 2018-08-23:
I know this post is quite old now but I'm having the same issue. I did verify the password is longer than 8 characters. the only way I'm being able to connect to the AP is when setting the mode WIFI_AP_STA and connecting to a router as station. I don't wan't to be forced to do this. any suggestions? @igrr

wbenish, 2018-11-13:
I had problems using the ESP8266 as an access point and solved the problem by including the statement:
WiFi.mode(WIFI_AP);
See  https://github.com/esp8266/Arduino/issues/4963

fonsmol, 2018-12-04:
> It is pretty obvious but I think it is worth to be mentioned. When the password is too short (less than 8 characters) the WiFi.softAP(ssid, password) function desn't work. Of course there is no warning during compilation.

This should be accepted as one of the solutions. Thnx!

IDispose, 2019-02-17:
It appears the password has to start with a character. When I used 12345678 as password SSID is not available. Changed it to welcome123 and AP works as expected.

DAVIDBETAN2000, 2019-05-29:
AT COMMANDS RETURNING

gibo77, 2020-01-10:
Can you point me to a link about how to make the SSID name unique? I am thinking of using
"AnyName_"+ the last three mac address. Example AnyName_3BT6RT. The last three mac address will make this name unique, in case if you have more than one out there.

mohamedanwer006, 2020-05-05:
> It appears the password has to start with a **character**. When I used 12345678 as password SSID is not available. Changed it to **welcome123** and AP works as expected.

This fix the problem thank you❤