esp8266/Arduino #1142: Lack of information on REFERENCE on 2.0.0 stable

asetyde, 2015-12-04:
Why ESP.x function are not described on reference  ? where can i looking for information about it ?

igrr, 2015-12-04:
Some are described in http://esp8266.github.io/Arduino/versions/2.0.0/doc/libraries.html#esp-specific-apis.

asetyde, 2015-12-04:
ok but i think it’s not right place . 

This function are not describe and i think can be interesting to maintain efficient my device ! 
i suppose because i not ve information :

When i clean ESP.  eraseConfig ?
 				flasherasesecor ?
				getresetinfo ?
			        reset ?
				
wdt				wdtfeed ?

igrr, 2015-12-04:
Sure our documentation coverage is pretty low now. Many important things aren't yet documented (ESP8266WebServer, for instance).

You can either check the source code for the functions you are interested in, ask a question [in the chat room](http://gitter.com/esp8266/Arduino), or wait until someone else adds this documentation.

asetyde, 2015-12-04:
Many thanks

closed 2015-12-04 by asetyde