esp8266/Arduino #1143: PROGMEM F() FPSTR() optimization test

hallard, 2015-12-04:
Hi guys,

I'm searching the best way to optimize my flash size to avaoid redundant string stored in it using FPSTR macro, so be sure to understand, I've done some tests that I don't really understand.

I tried to use the same a "10 char" string 10 times I've done different tests 

First one, trying basic, just print string
```arduino
#define STR "0123456789" 
const char FP_TEST[] PROGMEM = STR;
void setup() {}
void loop() {
  Serial.print(STR);  Serial.print(STR);  Serial.print(STR);  Serial.print(STR);  Serial.print(STR);
  Serial.print(STR);  Serial.print(STR);  Serial.print(STR);  Serial.print(STR);  Serial.print(STR);
}
```
Compilation says Use is Flash=202508 RAM=33158, ok not let's test with F() macro, I should use more flash and less RAM (correct ?)
```arduino
#define STR "0123456789" 
const char FP_TEST[] PROGMEM = STR;
void setup() {}
void loop() {
  Serial.print(F(STR));  Serial.print(F(STR));  Serial.print(F(STR));  Serial.print(F(STR));  Serial.print(F(STR));
  Serial.print(F(STR));  Serial.print(F(STR));  Serial.print(F(STR));  Serial.print(F(STR));  Serial.print(F(STR));
  }
```
Compilation says Use is Flash=202652 RAM=33142, ok we eat 144 more flash and win only 16 bytes of RAM, ok not let's test with FPSTR() macro, I should use now less flash and approx same RAM (correct ?)

```arduino
#define STR "0123456789" 
const char FP_TEST[] PROGMEM = STR;
void setup() {}
void loop() {
  Serial.print(FPSTR(FP_TEST));  Serial.print(FPSTR(FP_TEST));  Serial.print(FPSTR(FP_TEST));  Serial.print(FPSTR(FP_TEST));  Serial.print(FPSTR(FP_TEST));  
  Serial.print(FPSTR(FP_TEST));  Serial.print(FPSTR(FP_TEST));  Serial.print(FPSTR(FP_TEST));  Serial.print(FPSTR(FP_TEST));  Serial.print(FPSTR(FP_TEST));  }
```
Compilation says Use is Flash=202508 RAM=33142, ok we won back our 144 bytes flash without touching RAM, seems to be the best way, just for fun let's test compiler used the PROGMEM string without FPSTR() macro

```arduino
#define STR "0123456789" 
const char FP_TEST[] PROGMEM = STR;
void setup() {}
void loop() {
  Serial.print(FP_TEST);  Serial.print(FP_TEST);  Serial.print(FP_TEST);  Serial.print(FP_TEST);  Serial.print(FP_TEST);
  Serial.print(FP_TEST);  Serial.print(FP_TEST);  Serial.print(FP_TEST);  Serial.print(FP_TEST);  Serial.print(FP_TEST);
```
Compilation says Use is Flash=202508 RAM=33142, ohhhh same thing than FPSTR, so here is my misunderstanding, what's the real role of FPSTR ?

Thanks for any explanation

Links2004, 2015-12-04:
FPSTR has noting to do with flash size usage.
the reason for it is that the access to the flash need to be aligned to 4 byte.
the print functions can not tell if ```const char FP_TEST``` is ram or flash.
for this FPSTR is needed, if it is used the access will be automatically aligned to 4 byte.
https://github.com/esp8266/Arduino/blob/342c4ae6fb847bfc787f80b89a2bb888d942dc32/cores/esp8266/Print.cpp#L62
https://github.com/esp8266/Arduino/blob/342c4ae6fb847bfc787f80b89a2bb888d942dc32/cores/esp8266/pgmspace.h#L77-L84

if you not do this you can have luck and it work, but the change that it work is 1 of 4.

hallard, 2015-12-04:
@Links2004 
thanks, got it to alignment, but not sure what I need to do ?
in case of undeclared, 
`Serial.print(F("Test"))` or `Serial.print(FPSTR("Test"))` ?

For declared string such as `const char FP_TEST[] PROGMEM = "Test";`
`Serial.print(FP_TEST)` or `Serial.print(FPSTR(FP_TEST))` ?

Links2004, 2015-12-04:
```Serial.print(F("Test"));``` will work 

```Serial.print(FPSTR("Test"))``` make not difference to ```Serial.print("Test")``` and it will not be in flash.

for ```const char FP_TEST[] PROGMEM = "Test";```
```Serial.print(FP_TEST)``` is bad (no alignment) and ```Serial.print(FPSTR(FP_TEST));``` will work.

hallard, 2015-12-04:
Excellent, thanks for this clean explanation, so with very little strings says 1,2 or 3 char, best to set them in RAM because they all eat 4 bytes in Flash, good to know ;-)

closed 2015-12-05 by hallard

tzapu, 2015-12-23:
if i might trouble you for some more assistance with this, it really is doing my head in...
i have a function 
```
template <typename Generic>
void WiFiManager::DEBUG_PRINT(Generic text) {
  if(_debug) {
    Serial.print("*WM: ");
    Serial.println(text);    
  }
}
```
if i call it like 
```
DEBUG_PRINT(F("Request redirected to captive portal"));
```
will my strings be in PROGMEM/flash ok? or am i wasting memory somehow...

rochamorelli, 2017-02-09:
Hey @tzapu, do you mind giving me feedback of your own question after this long time?
I have the same doubt.
BTW, great work on WifiManager :smile: