esp8266/Arduino #1146: Usage with Arduino Nano

Triscus, 2015-12-04:
Hello,

I would like to use the libraries and Sketches to connect through HTTPS, but I 'only' got an Arduino Nano where the ES8266 chip is onboard, so the upload isn't working, when I change the Board to 'Generic ESP8200'. 

Someone out here can give me a tip how to get this working? 

Regards

Triscus

Links2004, 2015-12-04:
can you link to the board?
without knowing the schematic its hard to say if a direct upload is easy possible.
with some soldering is will sure work :)

Triscus, 2015-12-04:
The board ist hardwired with the arduino. I can send AT commands to the ESP-board via SoftwareSerial Port 11 and 12

Links2004, 2015-12-04:
for upload you need more then RX TX.
without a schematic I can not give you any advise, 
howto using this Project on the ESP.

Triscus, 2015-12-04:
I haven't found any schematics for this setup, I only found pictures: http://www.elektor.de/pretzel-board-iot-wifi-board-7596

penfold42, 2015-12-05:
It's got a header with all the pins on an esp01 board.

If you hold the 32u4 in reset they probably float and you could use them directly.

Another option is to write a sketch for the 32u4 that acts as a serial to serial relay.

Links2004, 2015-12-05:
[image]

simply connect the ESP01 part like this:
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#esp-to-serial

note:
GPIO15 and CH_PD are connected on the board itself.

the nano processor (ATmega328) needs to be in reset
or you flash a firmware that not use pin 11 and 12.

Triscus, 2015-12-05:
Thank you for your replies, I will see if I can figure it out :)

closed 2016-06-23 by igrr