esp8266/Arduino #1149: pulseIn cause Nodemcu to reset with some pins.

Leen15, 2015-12-05:
Hi, i was trying to make working a SRF04 with my nodemcu and i notice something strange that reboot the board. So i tried to minimize my code for find the problem, and appear that pulseIn is causing the board to reset with some pins. (i don't know how).

With this code, and nothing connected to the board:

    const int ECHO_PIN = D3; // WORKING WITH: D1,D2,D8 -  NOT WORKING WITH: D3,D4,D5,D6,D7,D9,D10       
    void setup() {
       // initialize serial communication:
        Serial.begin(115200);       
        pinMode(ECHO_PIN,INPUT);  
    }
    
    void loop() {
      long duration;
      duration = pulseIn(ECHO_PIN,HIGH);
      Serial.println(duration);  
      delay(100);
    }

I read in serial values of type 1 or 2 and the board keep running with ECHO_PIN set to D1,D2, or D8, but with pins D3,D4,D5,D6,D7,D9,D10 I receive instead something like:

    1000001
    1000001
    1000000
    
    ctx: cont 
    sp: 3ffe9aa0 end: 3ffe9c90 offset: 01b0
    
    >>>stack>>>
    3ffe9c50:  00000000 00000000 00000016 3ffe9cbc  
    3ffe9c60:  3fffdc20 00000000 3ffe9cb4 40201d90  
    3ffe9c70:  00000000 00000000 00000000 402018ea  
    3ffe9c80:  00000000 00000000 3ffe8c70 40100118  
    <<<stack<<<
    
     ets Jan  8 2013,rst cause:2, boot mode:(3,6)
    
    load 0x4010f000, len 1264, room 16 
    tail 0
    chksum 0x42
    csum 0x42
    ~ld
    
and it reset every 3-4 prints.

Did i miss something in the sketch? Why only 3 pins work?

PS: I tried with 3 different nodemcu boards, ESP12E.

Thanks

framb00s, 2016-01-17:
Tnx for the info.

I get strange results with dustPin set to D3
void loop() {
  duration = pulseIn(dustPin, LOW);
  if(duration!=0){                                                    // number positive readings
    i++;
  }

when I run this program on a arduino (nano or uno) it works OK
but on the nodeMCU I get false readings.
I'll try with D1,3 or 8 and see if this works.

DiApostola, 2016-02-09:
I'll echo this bug as well - running Arduino 1.6.7 with the ESP8266 2.0 branch, using the so-called ESP-201 module. As above, trying to run the pulseIn command on any port but the ones above results in a failure to launch the sketch, and even on those ports it will not allow the wifi to work  properly and will fail after 3-4 readings, causing the module to reset.

Arrowana, 2016-05-15:
I tried latest master with D2, D3, D4 and gave up. pulseIn doesn't look to give correct results (I set a very long timeout and try to switch LOW and HIGH manually.)

earlephilhower, 2019-07-26:
I just ran the above test and tried all the reported-failing pins on a D1 Mini and all seems to be well.  Given the age and lack of detail, and the fact that it can't be repro'd, closing.  If someone does hit this, please do open a new issue with decoded dumps/etc.

closed 2019-07-26 by earlephilhower

FernandoGarcia, 2022-02-23:
I have faced this problem today using a NodeMCU.
Really this function causes crash with some pins.
Doesn't work with pin D3 but worked with D1.