esp8266/Arduino #1152: Capture stack trace and reset after startup?

penfold42, 2015-12-05:
Is there any way to capture the reset reason and stack trace after a restart in my sketch ?

I want to send this over the network rather than capture it on the serial interface.

Thanks !

Links2004, 2015-12-05:
for the  reset reason:
```cpp
ESP.getResetInfo();
//or
ESP.getResetInfoPtr();
```
https://github.com/esp8266/Arduino/blob/342c4ae6fb847bfc787f80b89a2bb888d942dc32/cores/esp8266/Esp.h#L128-L129
https://github.com/esp8266/Arduino/blob/f1a8287bb380cd1fc09a3e0c5c86c9ac7de65788/cores/esp8266/Esp.cpp#L301-L312

igrr, 2015-12-05:
You can also register a crash callback ([see this commit](https://github.com/esp8266/Arduino/commit/baa13f8c841654fdb83a67fe1d41ecc31921813e))
Within it, you can save the stack somewhere in flash, and load it on next restart. You have to be careful with what you do in that callback though.

penfold42, 2015-12-06:
Thanks guys,

What does "careful" mean ?
I'm guessing this is call as little as possible because you don't know what is still working ?

What happens if the crash handler crashes ?

igrr, 2015-12-06:
Pretty much, yes. Don't use dynamic memory allocation, don't use blocking functions (delay, Serial, network), don't spend too much time inside the interrupt handler (because hardware watchdog timer is still ticking).
If you cause another exception inside your handler, you can still rely on hardware WDT to reset the ESP in a few seconds.

TheAustrian, 2016-04-13:
Are there any plans to incorporate this saving to flash of the stack as a ready-made function or at least an example or some more pointers on how to implement such a thing? It seems a bit over my head since I didn't dabble with memory before... 

Background: I would like to be able to send a stack dump over the net after a reset, since I have an ESP that basically gets a hardware WDT reset every 10-20 hours and which isn't easily accessible.

djoele, 2016-04-30:
A couple of months ago I wrote a custom crash callback function that would save the stack to flash memory. Then at startup the stack was read from the flash memory so I know what caused the crash.

So globally I have the following structure:

void setup() 
  EEPROM.begin(EEPROM_MAX_ADDR);
  stack = loadStack();
 ....

Then I have this function:
extern "C" void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end ){   
  ...
  ...
  eeprom_erase_all();
  eeprom_write_string(0, buf);
  EEPROM.commit();
}

I notice that I always get the same stack at reboot. Investigation turns out that that the stack does not get written by the custom_crash_callback.

I know it is working because I isolated the write to EEPROM lines and reading the EEPROM then show different data. The weird thing is that some months ago it was als working insiide the crash_callback_function.

Might any code in the ESP8266 core have changed so this behaviour is different now?

djoele, 2016-05-19:
I got it working. Below is the code that I used. I'm not the best coder, so any comments are welcome. At least it is working, now still have to make it better.

What it is doing:
1. Get the complete failure message at a crash, including the stack trace
2. Write this message to eeprom
3. At startup read the message from eeprom
Then you have it available and can do anyting with it.

```
//This function gathers the stack itself from starter to ender
void getStack(uint32_t starter, uint32_t ender){
  char stack_self[1200] = "";
  char stack_self2[46];
  const char stack_begin[15] = "\n>>>stack>>>\n";
  const char stack_end[13] = "<<<stack<<<\n";
  strcat(stack_self, stack_begin);
  
  for (uint32_t pos = starter; pos < ender; pos += 0x10) {
      uint32_t* values = (uint32_t*)(pos);
      //rough indicator: stack frames usually have SP saved as the second worde
      bool looksLikeStackFrame = (values[2] == pos + 0x10);
      sprintf(stack_self2, "%08x:  %08x %08x %08x %08x %c\n", pos, values[0], values[1], values[2], values[3], (looksLikeStackFrame)?'<':' ');
      strcat(stack_self, stack_self2);
      ESP.wdtFeed();
  } 
  strcat(stack_self, stack_end);
  strcpy(buf2, stack_self);
}
```

```
//This function is called at a crash, mosty a copy and adaptation of the original crash function
//Basically what it does is building up the error message, including the stack dump
extern "C" void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end ){  
  register uint32_t sp asm("a1");
  cont_t g_cont __attribute__ ((aligned (16)));
  char result[2000];
  char exception[300];
  char cont[14];
  char nctx[14];
  char spi[50];
  
  if (rst_info->reason == REASON_EXCEPTION_RST) {
     sprintf(exception,"Exception (%i):\nepc1=0x%08x epc2=0x%08x epc3=0x%08x excvaddr=0x%08x depc=0x%08x\n",
            rst_info->exccause, rst_info->epc1, rst_info->epc2, rst_info->epc3, rst_info->excvaddr, rst_info->depc);        
  }
    
  uint32_t cont_stack_start = (uint32_t) &(g_cont.stack);
  uint32_t cont_stack_end = (uint32_t) g_cont.stack_end;
  uint32_t stack_end2 = stack_end;

  uint32_t offset = 0;
  if (rst_info->reason == REASON_SOFT_WDT_RST) {
      offset = 0x1b0;
  }
  else if (rst_info->reason == REASON_EXCEPTION_RST) {
      offset = 0x1a0;
  }
  else if (rst_info->reason == REASON_WDT_RST) {
      offset = 0x10;
  }
  if (sp > cont_stack_start && sp < cont_stack_end) {
      sprintf(nctx, "\nctx: cont \n");
      stack_end2 = cont_stack_end;
  }
  else {
      sprintf(nctx, "\nctx: sys \n");
      stack_end2 = 0x3fffffb0;
  }
  sprintf(spi, "sp: %08x end: %08x offset: %04x\n", stack, stack_end2, offset);

  if (rst_info->reason == REASON_EXCEPTION_RST) {
    strcat(result, exception);
  }
  strcat(result, cont);
  strcat(result, nctx);
  strcat(result, spi);
  getStack(stack, stack_end);
  strcat(result, buf2);
  strcpy(buf,result);

  eeprom_erase_all();
  eeprom_write_string(0, buf);
  EEPROM.commit();
} 
```

Then in setup():

```
EEPROM.begin(EEPROM_MAX_ADDR);
SPIFFS.begin();
stack = loadStack();
```

This is the loadStack function:

```
String loadStack(){
  char *rinfo;
  String reset;
  reset = ESP.getResetInfo();
  rinfo = &reset[0];
  char rr[2500];

  eeprom_read_string(0, buf, EEPROM_MAX_ADDR);
  String stack = urlencode(buf);
  const char find[4] = "ctx";
  const char find2[10] = "Exception";
  const char * stackkie = stack.c_str();
  char *ret;
  ret = strstr(stackkie, find2);
  if (ret==NULL){
    ret = strstr(stackkie, find);
  }
  
  strcpy(rr, rinfo);
  if (ret != NULL){
    strcat(rr, (const char *)ret);
  }
  String bla = urlencode(rr);
  eeprom_erase_all();
  EEPROM.commit();
  return bla;
}
```

djoele, 2016-05-25:
For the interested, with the above code I got results. See below for a SOFT_WDT result and an EXCEPTION result. When these are decode with the ESP Exception Decoder they give the expected results.

2016-05-25 19:38:00 Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x4000bf80 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000ctx%3A+sys+%0A+sp%3A+3fff3d20+end%3A+3fffffb0+offset%3A+01b0%0A+++%0A+3fff3d20%3A++3ffe92a4+000004d5+000004d5+4020e6fd++%0A3fff3d30%3A++3ffe8ca4+00000001+3fff1618+4020e7e8++%0A3fff3d40%3A++3ffe8c80+00000007+3fff1618+4020e855++%0A3fff3d50%3A++00000002+00000001+3ffe8ca4+4020709a++%0A3fff3d60%3A++00000000+00000000+00000000+4010068c++%0A3fff3d70%3A++00000000+00000000+3fff6044+4020ca32++%0A3fff3d80%3A++3fff6044+3fff0d7c+3fff6044+4020ca6e++%0A3fff3d90%3A++00000000+00000000+00000000+4020f274++%0A3fff3da0%3A++3fff6044+3fff0d7c+3fff0d3c+4020cafd++%0A3fff3db0%3A++3fff45d4+0000000f+00000007+402095bc++%0A3fff3dc0%3A++00000000+00000004+00000004+00000001++%0A3fff3dd0%3A++00000002+00000004+0000000e+3fff2e60++%0A3fff3de0%3A++00000000+00000000+3fff0d3c+3fff2e58++%0A3fff3df0%3A++00000001+3fff0d60+3fff0d3c+4020cd0b++%0A3fff3e00%3A++3ffe9180+00000000+000003e8+4020a7f8++%0A3fff3e10%3A++00000000+3fff63ec+000003e8+4020a876++%0A3fff3e20%3A++3fffdad0+00000000+3fff2e51+4020866b++%0A3fff3e30%3A++00000000+00000000+00000000+00000000++%0A3fff3e40%3A++00000000+00000000+402072dc+402083a0++%0A3fff3e50%3A++00000000+00000000+00000001+3fff2e58++%0A3fff3e60%3A++3fffdad0+00000000+3fff2e51+4020fc34++%0A3fff3e70%3A++feefeffe+feefeffe+3fff2e60+40100718++%0A+%3C%3C%3Cstack%3C%3C%3C%0A+%3Cnull%3E

2016-05-25 19:37:17 Fatal exception:28 flag:2 (EXCEPTION) epc1:0x402071ae epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000Exception+%2828%29%3A%0Aepc1%3D0x402071ae+epc2%3D0x00000000+epc3%3D0x00000000+excvaddr%3D0x00000000+depc%3D0x00000000%0A+%90%3B%FF%3F%90%3B%FF%3F%04+%0Actx%3A+cont+%0A+sp%3A+3fff3d50+end%3A+3fff3e80+offset%3A+01a0%0A+++%0A+3fff3d50%3A++00000002+00000001+00000000+402071ac++%0A3fff3d60%3A++32317830+36353433+00000000+4010068c++%0A3fff3d70%3A++00000000+00000000+3fff60c4+4020ca32++%0A3fff3d80%3A++3fff60c4+3fff0d7c+3fff60c4+4020ca6e++%0A3fff3d90%3A++00000000+00000000+00000000+4020f274++%0A3fff3da0%3A++3fff60c4+3fff0d7c+3fff0d3c+4020cafd++%0A3fff3db0%3A++3fff45d4+0000000f+00000006+402095bc++%0A3fff3dc0%3A++00000000+00000004+00000004+00000001++%0A3fff3dd0%3A++00000002+00000004+0000000e+3fff2e60++%0A3fff3de0%3A++00000000+00000000+3fff0d3c+3fff2e58++%0A3fff3df0%3A++00000001+3fff0d60+3fff0d3c+4020cd0b++%0A3fff3e00%3A++3ffe9180+00000000+000003e8+4020a7f8++%0A3fff3e10%3A++00000000+3fff65fc+000003e8+4020a876++%0A3fff3e20%3A++3fffdad0+00000000+3fff2e51+4020866b++%0A3fff3e30%3A++00000000+00000000+00000000+00000000++%0A3fff3e40%3A++00000000+00000000+402072dc+402083a0++%0A3fff3e50%3A++00000000+00000000+00000001+3fff2e58++%0A3fff3e60%3A++3fffdad0+00000000+3fff2e51+4020fc34++%0A3fff3e70%3A++feefeffe+feefeffe+3fff2e60+40100718++%0A+%3C%3C%3Cstack%3C%3C%3C%0A

djoele, 2016-05-26:
I want to do some more testing with these functions. I have the ability to call a URL like http://<ESP IP>:80/crash and then execute code that causes a crash.

Does the community have some example code that causes crashes? Then I can use that to do thorough testing.

This is what I have now:

server.on("/crash", [](){
    if(!server.authenticate(www_username, www_password))
      return server.requestAuthentication();
    server.send(200, "text/plain", "ESP8266 is about to crash with EXCEPTION..");
    char linea[]="0x123456",**ap;
    int num;
    num=strtol(linea,ap,0);
    printf("%d\n%s",num,*ap);
    int k;
  });

  server.on("/crash2", [](){
    if(!server.authenticate(www_username, www_password))
      return server.requestAuthentication();
    server.send(200, "text/plain", "ESP8266 is about to crash with SOFT_WDT..");
    while (true){
      serverClient.println("Crashing...");
    }
  });
  server.on("/crash3", [](){
    if(!server.authenticate(www_username, www_password))
      return server.requestAuthentication();
    server.send(200, "text/plain", "ESP8266 is about to crash with WDT..");
    ESP.wdtDisable();
    while (true){
      serverClient.println("Crashing...");
    }
  });

djoele, 2016-06-02:
Hello interested people, I created a minimal Arduino sketch with the following functionality:

1. At startup it connects to Wifi and registers 5 crash functions that can be called
2. The reset reason and stack (if present) are printed to Serial at startup
3. A custom crash callback stores the stack in EEPROM

Simply do the following:
- Open the sketch
- Set your ssid and password
- Fire up the sketch

Then call the next URL's in a row, for example by using curl:
1. wget <your ESP IP>:80/crash
2. wget <your ESP IP>:80/crash2
3. wget <your ESP IP>:80/crash3
4. wget <your ESP IP>:80/crash4
5. wget <your ESP IP>:80/crash5

The stack printed at startup can be given to ESP Exception Decoder and there you have your stack in human readble format.
 
The output I receive is attached. It can be seen that after startup the reasot reason and stack are available.
[crash_test_output.txt](https://github.com/esp8266/Arduino/files/296263/crash_test_output.txt)
[minimal.zip](https://github.com/esp8266/Arduino/files/296272/minimal.zip)

djoele, 2016-06-02:
@TheAustrian and @penfold42: could you have a try with this sketch I created?

djoele, 2016-06-04:
Cool! This morning I had a crash and got the crash below in my inbox. So, is seems to be working fine. I was able to decode it in ESP Exception Decoder.

Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4022a136 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000008 depc:0x00000000
ctx: cont sp: 3fff3c70 end: 3fff3f40 offset: 01a0
3fff3c70:  402090cc 3fff6984 00000001 4022a0f0  3fff3c80:  3ffefbe0 00000000 00000000 3fff3d70  3fff3c90:  00000002 0000000f 3fff3cd0 3fff6a43  3fff3ca0:  3fff6a41 07077c0a 00001387 00000030  3fff3cb0:  4023705b 6db5cf70 ebff7ce3 00000000  3fff3cc0:  3fff6984 00080000 00000003 4022affd  3fff3cd0:  00020315 3fff6a02 3fff852c 00000940  3fff3ce0:  00000002 00000015 0000000b 00000004  3fff3cf0:  3fff6984 3fff6a41 00000004 00000000  3fff3d00:  00000000 00080000 3fff6984 4022b61e  3fff3d10:  00000001 3fff675c 00000015 401004d8  3fff3d20:  3ffe9c8c 00000028 3fff6654 00000000  3fff3d30:  3fff3d70 0000000e 00000010 00001387  3fff3d40:  07077c0a 00000000 3fff6984 4022b67a  3fff3d50:  07077c0a 3fff6984 3fff6654 40208cc2  3fff3d60:  3fff675c 3fff4034 3fff6654 402090cc  3fff3d70:  00000000 3fff4034 3fff4034 4020842a  3fff3d80:  a2a75754 00000000 3fff3dc0 00000000  3fff3d90:  000001bb 3fff4034 3fff675c 40209409  3fff3da0:  3ffe9740 a2a75754 3ffe9740 a2a75754  3fff3db0:  000001bb 00000000 3fff650c 4020a760  3fff3dc0:  00000000 00000001 3fff3e40 4020e138  3fff3dd0:  3fff650c 00000000 3fff650c 4020abc0  3fff3de0:  00000000 00000000 00000000 00000000  3fff3df0:  00000000 00000000 00000000 00000000  3fff3e00:  3ffe9078 00000012 00000012 4010020c  3fff3e10:  000001bb 3fff3e9c 3fff3e40 00047c50  3fff3e20:  000001bb 3fff3e9c 3fff650c 4020ac66  3fff3e30:  000001bb 3fff3e9c 3fff650c 402070e8  3fff3e40:  00000000 00000000 00000000 00000000  3fff3e50:  00000000 00000000 00000000 00000000  3fff3e60:  00000000 3ffe8dec 3fff3e90 4020e31c  3fff3e70:  3fff3e90 3ffe8dec 0000012c 4020e344  3fff3e80:  3fff3e9c 3ffe8dec 0000012c 40207236  3fff3e90:  3fff63dc 0000001f 00000014 3fff6384  3fff3ea0:  0000004f 00000046 00000000 00000000  3fff3eb0:  00000000 3fff64b4 0000004f 00000046  3fff3ec0:  3fff3e9c 3ffe8d0c 3fff2d18 402096fe  3fff3ed0:  4072c000 00000000 00000008 3fff2f18  3fff3ee0:  3fff2d08 402072d4 3fff2d08 402072fd  3fff3ef0:  3fff2d08 402072d4 3fff2d08 40209984  3fff3f00:  00000000 070779f2 000003e8 402099d4  3fff3f10:  3fffdad0 00000000 3fff1420 40207653  3fff3f20:  feefeffe 00000000 3fff2f10 4020ebac  3fff3f30:  feefeffe feefeffe 3fff2f20 40100718

TheAustrian, 2016-06-04:
@djoele Awesome, thanks! I'll have a go at it in a few days and report back.

djoele, 2016-06-15:
@TheAustrian: have you been able to check the sketch?

djoele, 2016-06-23:
Hello everybody, this question is over half a year old.
I provided a sketch that is working, at least in my situation.

The original poster has not responded and one person has but did not reply with some results of testing the sketch.

Is there anybody interested or should we marked the question as closed?

TheAustrian, 2016-06-23:
Hi! I'm still interested and have tested the sketch you provided and it seems it works (thanks for that and sorry I did not immediately respond!), but I haven't had the time to  adapt the sketch to my needs i.e. either integrate it into my existing program and transmit the crash info with websocket to the website I use or log it in a database on my server. 

Work has been a pain lately (everyone wants everything ready before the holidays, it seems)...

djoele, 2016-06-25:
Good that it worked. Let us know when you're done putting it in your own sketch.
Perhaps this functionality could be converted into some general functionality, as written by you on Apr 13.

djoele, 2016-07-11:
@TheAustrian: did you find any time to work on your sketch?

Has anybody else tried it?

krzychb, 2016-07-11:
Hi @djoele,

I believe your sketch is a very useful tool.

I have encountered cases like @penfold42, when I like to debug some exception / ESP restarts that occur infrequently / at remote location so it has no sense or it is not practical to watch at serial monitor waiting for it to happen.

This thread and your sketch in particular inspired me to write a library to convert what you have developed into some general functionality. I am tied up with some other tasks now but I am planning to return to it in next coupe of weeks. If anybody is interested then share your thoughts.

Thank you for development of this functionality and sharing!

Krzysztof

djoele, 2016-07-11:
@krzychb: I appreciate it that you pick up where I left. Looking forward to see the results of the library you are making. Perhaps the code as I wrote it could be improved as well.

djoele, 2016-08-04:
@krzychb: have you made any progress on your library?

krzychb, 2016-08-05:
Hi @djoele, 
I jump on another exciting project and did not have a chance to get to this one.
Thank you for getting me motivated. 
I think I can aim for end of August to release preliminary version for review.
Krzysztof

djoele, 2016-08-05:
No problem :) Looking forward to your first release.

krzychb, 2016-08-15:
Hi @djoele,

I have prepared preliminary version of library to capture and save crash information https://github.com/krzychb/EspSaveCrash.

There are also examples attached. They allow to crash ESP and then show captured data on a serial monitor. There is also an example to show crash information with a web browser. I plan to redo this one and prepare a more compelling version :smile:

Krzysztof

djoele, 2016-08-17:
This is done in a good way, I think.
I tried your examples and for my cause they are working.

I created something similar for your web browser functionality. I created a call that is handled when my server tries to get the crash information from the ESP device.

Good job!

krzychb, 2016-08-17:
Thank you for quickly checking it and good news!
Krzysztof

devyte, 2017-10-19:
I think having this in its own repo is ok. Thank you, I use it myself.
Closing as resolved.

closed 2017-10-19 by devyte

popok75, 2018-12-28:
I'm not sure if I'm careful enough in that callback but here is a little lib to save esp8266 exception trace in a spiffs file, very useful if you run a server ;) , based on Krzysztof work :
[crashsaverspiffs.zip](https://github.com/esp8266/Arduino/files/2714969/crashsaverspiffs.zip)

brainelectronics, 2020-01-04:
@popok75 I've developed your code a little bit further. May you and other interested hackers checkout my [EspSaveCrashSpiffs](https://github.com/brainelectronics/EspSaveCrashSpiffs) lib. It's based on @krzychb EEPROM lib, but uses the filesystem for creating crash log files everytime a crash occurs. Thereby the files stay small and you can almost create as many crashes as you want

arihantdaga, 2020-01-23:
Is there any way or any function which will be called right before esp shuts down because of power loss. 
I have an application where i want to store cumulative energy consumed by an appliance in the EEPROM just before power goes, so that when the power comes back, i can start from the energy value till now. 
custom_crash_callback is called only when it crashes, i am looking for something which will be called everytime before device shuts down. I am sure something is done because ESP. getResetInfo can tell that last restart reason was power on/power off. So its writing this somewhere.

d-a-v, 2020-01-23:
> can tell that last restart reason was power on/power off.

Maybe because something is *not* written before boot (default value on cold boot means power-on, other values are written right before reboots).

Anyway I personally am not aware such function and I'm pretty sure there's none.

I would use a 3v3powersupply->schottky diode->supercapacitor-on-vcc,
and an interrupt management on a pin connected to 3v3power-before-diode to trigger the emergency function.
An electronician may comment :)