esp8266/Arduino #1155: HardwareSerial::begin can't be called twice to change baud rate

Imroy, 2015-12-05:
HardwareSerial::begin() creates a '_uart' object by calling uart_init(), but doesn't check to see if it has already been created. Thus it malloc's without freeing the old one first. And it doesn't respect whatever settings have already been made.

In my case, I just need to be able to change the baud rate after having swapped Serial to the alternative pins. I've modified begin() to allow changing settings on an existing _uart object, which appears to be how the original Arduino code is meant to be used. A much simpler change_baudrate() method is also possible.

asetyde, 2015-12-05:
this malloc has not a free after ?

drmpf, 2015-12-22:
I am currently using
```
  Serial.end();
  delay(10);
  Serial.begin(storage.baudRate);
```
to change baud rates

closed 2016-02-27 by igrr