esp8266/Arduino #1156: ESP8266 + Servo

yeralin, 2015-12-05:
Cannot make it work.

Used example code piece: Hello World web server
``` C++
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <Servo.h>
Servo myservo; // ADDING LIBRARY AND INITIALIZING SERVO

const char* ssid = "Test";
const char* password = "Test";

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  myservo.attach(0); // ATTACHING 0 PIN (TRIED 2 and 9) NO PROGRESS
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

//BASICALLY MY THOUGHT WAS WHEN I CALL 192.168.1.X/inline
//SERVO SHOULD ROTATE
  server.on("/inline", []() { 
    int pos;
    for (pos = 0; pos <= 180; pos += 1) // goes from 0 degrees to 180 degrees
    { // in steps of 1 degree
      myservo.write(pos);              // tell servo to go to position in variable 'pos'
      delay(15);                       // waits 15ms for the servo to reach the position
    }
    for (pos = 180; pos >= 0; pos -= 1) // goes from 180 degrees to 0 degrees
    {
      myservo.write(pos);              // tell servo to go to position in variable 'pos'
      delay(15);                       // waits 15ms for the servo to reach the position
    }
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
}
```

Links2004, 2015-12-13:
whats the exakt problem?
any serial debug log to share?
what ESP do you use?

yeralin, 2015-12-18:
@Links2004 Here is the thing. Everything uploads/compiles fine, but when I access 192.168.1.x/inline it is loading, then shows me This works well, but my servo is not rotating 
[image]

Makuna, 2016-01-25:
Is it OK to call delay () inside the on () handler?  If interrupts are turned off then servo will not function either.

igrr, 2016-01-25:
I'm not sure about this particular issue, but running servo from WebServer handler certainly works.
Here's a [description of a demo](http://tech.scargill.net/esp8266-arduino-netio-server-demo/) and a [sketch](https://gist.github.com/igrr/9ef4d5c74355503e3b1f).

closed 2016-01-25 by yeralin