esp8266/Arduino #1159: Panic .../Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:404 ax_port_realloc

Apple456, 2015-12-06:
Trying to execute a simple post but always get this exception...

anyone knows how to fix it?

Panic .....Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:404 ax_port_realloc

ctx: cont 
sp: 3fff24f0 end: 3fff28c0 offset: 01b0

>>>stack>>>
3fff26a0:  40106cc6 3fff4b50 00000000 3fff656d  
3fff26b0:  36363636 3fff8bf8 3fff8c04 40207ef1  
3fff26c0:  3fff656d 00000000 3fff4b50 4021edfa  
3fff26d0:  3fff656d 00000000 3fff4b50 402201da  
3fff26e0:  00000001 00000001 3fff6560 4021e38e  
3fff26f0:  33c58093 927501d2 54640173 00000003  
3fff2700:  00000010 3fff656d 3fff4bfc 3fff2740  
3fff2710:  3fff656d 3fff4b18 00000099 00000010  
3fff2720:  3fff4b50 3fff656d 00000010 00000010  
3fff2730:  3fff656d 00000000 3fff4b50 4021fc2a  
3fff2740:  b1b8d11d 54aec1ab 3c0847a4 d054a820  
3fff2750:  feb74b2f 00000000 00000038 4021f348  
3fff2760:  00020016 00000014 00000010 00000010  
3fff2770:  00000033 00000016 3fff4b50 00000000  
3fff2780:  00000000 3fff4b50 3fff4b50 4021fd5e  
3fff2790:  00000000 3fff655d 3fff4b50 4021ff25  
3fff27a0:  00000000 3fff4a30 3fff4b50 4021ffac  
3fff27b0:  3fff4a00 4020a566 3fff2850 3fff173c  
3fff27c0:  3fff1740 3fff2850 3fff4a10 40207caa  
3fff27d0:  17b18268 3fff1890 4020a660 3fff1890  
3fff27e0:  3ffe8348 000001bb 3fff2850 40207f5a  
3fff27f0:  3ffe9c90 17b18268 3ffe95b0 3fff1890  
3fff2800:  3ffe8348 000001bb 3fff2850 402072d5  
3fff2810:  3ffe9c90 17b18268 3ffe9c90 17b18268  
3fff2820:  3ffe95a2 00000016 3fff185c 4020bd94  
3fff2830:  3fff1798 3fff185c 3fff185c 40207f70  
3fff2840:  3fff185c 3fff185c 3fff185c 40206d15  
3fff2850:  3ffe9698 00000000 000003e8 3fff1890  
3fff2860:  3fff0e44 3fff49d8 3fff4a10 40208c19  
3fff2870:  3ffe95c0 00000000 000003e8 3fff18a0  
3fff2880:  00000000 00000000 00000016 40101941  
3fff2890:  3fffdc20 00000000 00000001 40206f0a  
3fff28a0:  3fffdc20 00000000 3fff1889 4020a6a9  
3fff28b0:  00000000 00000000 3fff18a0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
waiting sensor calibration...
init...Joining Wifi Network
Connected to VM298111-2G

Exception (28):
epc1=0x4021b8c0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000011 depc=0x00000000

the code 

client.print("POST WHATEVER URL HTTP/1.1\n\n");
client.print("Host: mailgun.net\n");
client.print("Connection: keep-alive\n");
client.print("Content-Length: 91\n");
client.print("Cache-Control: max-age=0\n");
client.print("Authorization: Basic 64EncodedKey \n");
client.print("User-Agent: ESP8266 \n");
client.print("Content-Type: application/x-www-form-urlencoded\n");
client.print("Accept-Encoding: gzip, deflate\n");
client.print("Accept-Language: en\n\n");
client.print("from=user%40mail.com&to=user%40mail.com&text=Test+message+post&subject=Alarm%21%21%21")

Links2004, 2015-12-06:
you run out of ram.
and every client.print sends a tcp package this is not very efficient and may part of the problem.

we have also a http client library ready to use:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h#L67
then you not need to handle all the header stuff your self.

Apple456, 2015-12-06:
I try to build a String before and then perform just one client.print but I got the same result.
is there any example of the HTTPClient POST with parameters?
I could not see any ..

Links2004, 2015-12-06:
currently there is no example.
minimum example (no error handling)
```cpp
 HTTPClient http;
 http.begin("https://domain/post.php");
http.addHeader("Content-Type", "application/x-www-form-urlencoded");
http.POST("from=user%40mail.com&to=user%40mail.com&text=Test+message+post&subject=Alarm%21%21%21");
http.writeToStream(&Serial);
```

MarcusF2015, 2015-12-21:
I got the same 

Panic .....Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:404 ax_port_realloc

trying https://github.com/Lstt2005/ESP8266_I.O.Broker/tree/master/Arduino/Telegram/TelegramBot-master with release 2.0.0 of https://github.com/esp8266/Arduino and Arduino IDE 1.6.5.

How can I fix it?

Apple456, 2015-12-21:
according to links2004 it all works fine, it's some memory issues related to your/mine program, even if I measured I I had 3304 free... I'm still haven't seen a solution

Links2004, 2015-12-21:
SSL / https needs ~22KB ram so 3.3KB is fare away from it.

MarcusF2015, 2015-12-21:
The ESP8266 has got 45KB RAM. Is it a good idea to use SSL on the ESP8266, because SSL needs half of its Memory?

Links2004, 2015-12-21:
based on the application it i working just fine, 
and if you connect to a IoT server its sometime required.
I personally like it encrypt when it leaves my network.
if you compare it to an ATmega with 2KB ram you still have a lot when you use SSL.
so this depends a lot of how your application use the RAM.

itsjustvenky, 2015-12-21:
ESP has 64 KB instruction RAM and 96 KB data RAM. The 45KB what you are talking about is DRAM. @Links2004  Please correct me if I am wrong.

MarcusF2015, 2015-12-21:
I learned that RAM fragmentation causes this ax_port_realloc issue. (https://github.com/esp8266/Arduino/issues/1084)
Is the String object returned by String TelegramBOT::connectToTelegram(String command) (https://github.com/Lstt2005/ESP8266_I.O.Broker/blob/master/Arduino/Telegram/TelegramBot-master/src/ESPTelegramBot.cpp) the reason for this problem? 
Are String objects using dynamic allocations?

Links2004, 2015-12-21:
@itsjustvenky yes we are taking about the heap,
the esp has 2 RAM regions

96KiB "data RAM" are splittet:
- ETS system - 16KiB
- dram0 - 80KiB
 - all static variables and so on.
 - the rest is ```heap``` ( the amount is variable based on how many static stuff)

64KiB "instruction RAM" are splittet:
- iram1 - 32KiB (used for flash mirroring the first 32KiB of flash)
- 32KiB (dynamic flash mirroring)

@MarcusF2015 yes heap / ram fragmentation can causes this. since SSL needs a the space in heap in one piece.

itsjustvenky, 2015-12-21:
@Links2004 Thanks for the information.

ugurgudelek, 2016-01-18:
@Links2004 thanks for your all info 
however how can we fix this heap fragmentation. 
In my code im parsing html every 10 sec to get exact data out of it.

Do you have any tip to solve it?

Links2004, 2016-01-18:
fragmentation can not by solved  that easily but using a static buffer for SSL will help.
its on @igrr´s todo list see: https://github.com/igrr/axtls-8266/issues/2

apassemard, 2016-01-31:
This worked for me (at last....), POST to Ubidot. What I was missing was the "writeToStream", I had assumed that the POST function would do that automatically.
`
  http.begin("http://things.ubidots.com/api/v1.6/variables/_---change-with-your-var-id----_/values");
  http.addHeader("Content-Type", "application/json");
  http.addHeader("X-Auth-Token", "_----change-to-your-token---_");
  http.addHeader("NULL", "NULL");
  http.POST("{\"value\": 20 }");
  http.writeToStream(&Serial);
  http.end();
`

of course you'd have to change the JSON payload to fit whatever you try to send.

igrr, 2016-02-27:
While we don't have a solution for fragmentation now, we do have a workaround. axTLS will attempt to handle realloc failure by shutting down the connection. Hopefully user code can handle this in a similar way connection failure is handled, i.e. by attempting to connect again.

closed 2016-02-27 by igrr