esp8266/Arduino #1161: panic ..\core_esp8266_main.cpp:85 __yield ?? why

ntasher, 2015-12-06:
My code randomly crashes (may take hours) with this panic error (and stack trace). It's not a heap size problem, as I monitor it all the time and it doesn't change before the crash. I can't figure out the code in the __yield function:
<code>
extern "C" void __yield() {
    if (cont_can_yield(&g_cont)) {
        esp_schedule();
        esp_yield();
    }
    else {
        panic();
    }
}
</code> 
so I can't understand what's wrong and where to look for an answer. 
Help?? the ESP doesn't always recover from this crash (one out of two times it just stays in this:

 ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

until I give it hardware reset.

asetyde, 2015-12-06:
What does it do your code ?

Links2004, 2015-12-06:
i think you have 2 different bugs,
- one is the crash
- one is same same as #1017

igrr, 2015-12-06:
This `if (cont_can_yield(&g_cont))` condition is there to catch cases when yield is called *not* from Arduino context (i.e. not from `setup` or `loop`). @ntasher if you could share a gist with your code, we may be able to give some recommendation.

ntasher, 2015-12-06:
@Links2004  Makes sense and I'm kind of kicking myself for not figuring this one out myself - I'm using GPIO0 and 2 for I2C and GPIO0 is the SDA line which a slave device may pull low during a transaction, and if the system reboots at that time, it will be stuck in bootloader mode. I need to see if switching SDA and SCL between pins changes anything.
@igrr My code is quite long. It's mainly a take on the FSBrowser - serving web pages from SPIFFS and controlling some devices and LCD via I2C expanders (8754). Since my code spans 8 files I don't know how practical it will be to paste it. If you can tell me what I should look up or what debug code I should add (I can modify core_esp8266_main.cpp to add some debug code there if it can help).
Nir

igrr, 2015-12-06:
If you want to take on this yourself, make a disassembly of your sketch (`xtensa-lx106-elf-obdjump -d YourSketch.cpp.elf`) and look up addresses found in the ESP stack trace. These will give you return addresses of function calls. Try to figure out which of your functions calls into `yield`.

ntasher, 2015-12-06:
@igrr I will do that and look up the addresses. would it be the 1st or the last entry in the stack? I've just recompiled my sketch so I'll sit patiently and wait for it to crash :-)

ntasher, 2015-12-07:
@igrr... OK.. it took some time to get to this but here goes:
The call to esp_yield() is in this function:
4020a70c <_ZN16ESP8266WiFiClass10hostByNameEPKcR9IPAddress>:

The stack holds the return address to the next line after the call to esp_yield() [4020a74a] :

```assembly
4020a70c:	e0c112        	addi	a1, a1, -32
4020a70f:	51d9      	s32i.n	a13, a1, 20
4020a711:	042d      	mov.n	a2, a4
4020a713:	03dd      	mov.n	a13, a3
4020a715:	030c      	movi.n	a3, 0
4020a717:	7109      	s32i.n	a0, a1, 28
4020a719:	61c9      	s32i.n	a12, a1, 24
4020a71b:	04cd      	mov.n	a12, a4
4020a71d:	0667c5        	call0	40210d9c <_ZN9IPAddressaSEj>
4020a720:	fffa41        	l32r	a4, 4020a708 <_ZN16ESP8266WiFiClass5beginEPcS0_iPKh+0x10>
4020a723:	0d2d      	mov.n	a2, a13
4020a725:	013d      	mov.n	a3, a1
4020a727:	0c5d      	mov.n	a5, a12
4020a729:	086345        	call0	40212d60 <dns_gethostbyname>
4020a72c:	742020        	extui	a2, a2, 0, 8
4020a72f:	92cc      	bnez.n	a2, 4020a73c <_ZN16ESP8266WiFiClass10hostByNameEPKcR9IPAddress+0x30>
4020a731:	0138      	l32i.n	a3, a1, 0
4020a733:	0c2d      	mov.n	a2, a12
4020a735:	066645        	call0	40210d9c <_ZN9IPAddressaSEj>
4020a738:	000386        	j	4020a74a <_ZN16ESP8266WiFiClass10hostByNameEPKcR9IPAddress+0x3e>
4020a73b:	228000        	excw
4020a73e:	282001        	l32r	a0, 401d47c0 <_lit4_end+0xcdd60>
4020a741:	b37c31        	l32r	a3, 401f7534 <_lit4_end+0xf0ad4>
4020a744:	029237        	bne	a2, a3, 4020a74a <_ZN16ESP8266WiFiClass10hostByNameEPKcR9IPAddress+0x3e>
4020a747:	047545        	call0	4020ee9c <esp_yield>
4020a74a:	1c48      	l32i.n	a4, a12, 4
```

So, is this code part of the sketch or not? I guess it's not.
Nir

ntasher, 2015-12-07:
Another observation - the DNS call is due to NTP sync provider looking up the NTP server. While this works ok for the 1st request, if I set the sync interval to 10 seconds, the ESP crashes within 10 seconds  of acquiring WIFI. I'm wondering if the fact that RTC sync is happening in the background triggers the panic due to yield not being called from the main thread...
Nir

igrr, 2015-12-07:
This is true — all network calls should go through main thread. What are
you using to start NTP?

ntasher, 2015-12-07:
I'm using NTP library. This code is used to set it up after WIFI is connected:
```C
       ...
        NTPclient.begin("time.nist.gov", 0);
        setSyncInterval(10);
        setSyncProvider(getNTPtime);
       ...

time_t getNTPtime(void)
{
  time_t temptime;
  temptime= NTPclient.getNtpTime();
  return temptime;
  
 
}

 ```

ntasher, 2015-12-07:
OK. Figured out the cause but not the root cause of this.
My code uses Ticker. Every second I call a routine to display the time on an I2C LCD display. To get the time, this routine used ```time_t nowtime=now();``` It then printed the time on the LCD. This caused the Panic crash when NTP was called to update again.
Once I moved the ``` nowtime=now();``` to the main loop (making ```nowtime``` a global variable) the problem went away. I guess calling now() from an async process (Ticker based) messes something in the system context.
Nir

Links2004, 2015-12-07:
the NTPclient will sure use network for the update, and the network will use yield.
may you shut use the c++ <time.h> functions instead there the NTP update is handled async from the SDK in background.

```cpp
// config ntp (non c++ standard)
 configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");
```

and for getting the time:
```cpp
  time_t now = time(nullptr);
  Serial.println(ctime(&now));
```
more:
http://www.cplusplus.com/reference/ctime/time/

ntasher, 2015-12-07:
@Links2004 Can you share the complete code with the required ```INCLUDE``` since the time() function is not found.
Thanks.

Links2004, 2015-12-07:
https://github.com/esp8266/Arduino/blob/master/tests/Time/Time.ino

ntasher, 2015-12-07:
@Links2004  Thanks.. now this is going to be a noob question - I was using the Arduino Time.h library with all its nice macros for hour() and minute() and stuff.. I can't use it now since it messes with the time.h time() function. Any work around?
Thanks.

nullboundary, 2016-01-20:
@ntasher I ran into the same issue with the NTP library, and followed this thread for a solution. In answer to your question about the macros in TimeLib you can get pretty far using the tm struct in time.h

```cpp 
time_t now = time(nullptr);
struct tm * timeinfo;
timeinfo = localtime (&now);
```

This gives you some values like this:

```cpp
timeinfo->tm_mon //months since january
timeinfo->tm_mday //month day
timeinfo->tm_year //years since 1900
```
check the cpp reference:
http://www.cplusplus.com/reference/ctime/tm/

Most of the values you need to be cleaned up a bit more to be considered human readable. Check out my weather clock project for a more complete example.
https://github.com/nullboundary/weatherClock-esp8266

closed 2016-02-29 by igrr

pipi61, 2016-03-07:
Hi nullboundary!
The pixels* library for esp where found?
Thank You!

nullboundary, 2016-03-11:
Hi pipi61,

I have a fork of the pixels library that I made esp8266 compatible.
https://github.com/nullboundary/Pixels

Let me know if you have any issues with it. I have only tested it with my nodemcu 1.0 in both hw and sw spi modes.