esp8266/Arduino #1162: UDP Receive: Exception (29) after a few hundred packets... ???

vogtitec, 2015-12-06:
After a short time of successfully receiving Artnet packets, the ESP throws a fatal Error:
Exception (29):
epc1=0x4022b00f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: sys 
sp: 3ffffd90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff30:  00000008 3fff24c8 3ffedc98 4022b004  
3fffff40:  3ffedc98 4022e957 3ffedc98 3ffeb6c4  
3fffff50:  3ffeb6c4 00000265 00000000 0000002b  
3fffff60:  00000002 0000001a 4021e7d1 3ffedc98  
3fffff70:  3ffeb6b8 000000f5 3ffea248 3ffea248  
3fffff80:  00000080 3ffedc98 00000000 3fff09f0  
3fffff90:  4021e17c 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffea248 40000f49 3fffdab0 40000f49  
<<<stack<<<

Can anyone help me in interpreting or fixing the error, which seems to be in the ESP-System, not in the sketch area of the code. Many thanks for all suggestions!

igrr, 2015-12-06:
This is a null pointer being dereferenced somewhere in the code. To understand where this error occurs, you should generate disassembly using `xtensa-lx106-elf-objdump -d` and see the code which triggers the exception (at 0x4022b00f).
If the error is inside the ESP libraries, it is possible that your code (or some other code) does an out-of-bounds memory access, corrupting some pointer used by the SDK libraries.

vogtitec, 2015-12-08:
Further testing revealed, that it seems to be an issue with the ESP serial library - or the code which I use to transmit dmx data. The code for transmitting is following, it is called regularly from void loop(), which also contains a delay(10) to give time for ESP tasks:
void xmit() {
  Serial1.end();
  Serial1.begin(83333, SERIAL_8N1);
  Serial1.write(0);
  Serial1.flush();
  Serial1.end();
  Serial1.begin(250000, SERIAL_8N2);
  Serial1.write(dmxbuffer, sizeof(dmxbuffer));
  Serial1.flush();
}
The exception only occurs in conjunction with other WiFi related tasks, e.g. UDP receive or server handling. Does anyone have any suggestions?

Links2004, 2015-12-08:
the serial problem may is linked to #1166 try the latest git
https://github.com/esp8266/Arduino#using-git-version-

vogtitec, 2015-12-08:
@igrr :
Unfortunately, I'm not used enough to the SDK compiling and decompiling procedures, so I can't get help out of this...
@Links2004 : Thanks, I'll give it a try !

devyte, 2017-10-19:
No basic info, no sketch. Closing due to lack of info and age.

closed 2017-10-19 by devyte