esp8266/Arduino #1163: How to create custom SPIFF Size ?

hallard, 2015-12-06:
Hi guys

I'm trying to optimize size and the balance to use OTA and SPIFFS on ESP01 1MB Flash chip

Unfortunately my SPIFFS size once compiled/gzipped my framework size is about 132K, just after 128KB limit, next limit is 256KB SPIFFS. Of course I selected this one, but this reduce my global sketch size to approx 768KB/2 (when OTA) so about 384KB (less EEP and SDK Config).
I would like fine adjust SPIFFS used to say 160KB and/or 192KB.
Of course I know what to do this in boards.txt but what about the eagle files ? (linker script ?)
If think I need to create both files `eagle.flash.1m160.ld` and `eagle.flash.1m196.ld` but I don't really know what to put in and also if this is the only thing to do ?

Any help or documentation on this would be very helpfull.

Thanks for your help

Links2004, 2015-12-06:
yes the "ld" files are linker scripts.
you need to increase the code size (len) here:
https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.1m512.ld#L11
and move the SPIFFS by the same size you increase the code
https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.1m512.ld#L14
the changes need to match the boards.txt

hallard, 2015-12-06:
@Links2004 
Thanks, simple and efficient, I'll give it a try ;-)

By the way is there any alignment I need to respect (1,2,4,8 .. bytes ) ?

last but not least, on 512 script block is set by
`PROVIDE ( _SPIFFS_block = 0x2000 );` 
and on 128 and 256 it's 
`PROVIDE ( _SPIFFS_block = 0x1000 );` 
Shouldn't I use 0x1000 for any size between 128K and 256K ?

hallard, 2015-12-06:
Ok, just tried one 160KB SPIFFS and one 144KB SPIFFS

here are my files

modified boards.txt (just the interesting part)
```
wifinfo.menu.ESPModule.ESP07160=ESP07 (1M/160K SPIFFS)
wifinfo.menu.ESPModule.ESP07160.build.flash_size=1M
wifinfo.menu.ESPModule.ESP07160.build.flash_ld=eagle.flash.1m160.ld
wifinfo.menu.ESPModule.ESP07160.build.spiffs_start=0xD3000
wifinfo.menu.ESPModule.ESP07160.build.spiffs_end=0xFB000
wifinfo.menu.ESPModule.ESP07160.build.spiffs_blocksize=4096
wifinfo.menu.ESPModule.ESP07160.upload.maximum_size=860144

wifinfo.menu.ESPModule.ESP07144=ESP07 (1M/144K SPIFFS)
wifinfo.menu.ESPModule.ESP07144.build.flash_size=1M
wifinfo.menu.ESPModule.ESP07144.build.flash_ld=eagle.flash.1m144.ld
wifinfo.menu.ESPModule.ESP07144.build.spiffs_start=0xD7000
wifinfo.menu.ESPModule.ESP07144.build.spiffs_end=0xFB000
wifinfo.menu.ESPModule.ESP07144.build.spiffs_blocksize=4096
wifinfo.menu.ESPModule.ESP07144.upload.maximum_size=876528
```
and the 2 linker scripts
- eagle.flash.1m160.ld
```
/* Flash Split for 1M chips */
/* sketch 839KB */
/* spiffs 160KB */
/* eeprom 20KB */

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xd1ff0
}
PROVIDE ( _SPIFFS_start = 0x402D3000 );
PROVIDE ( _SPIFFS_end = 0x402FB000 );
PROVIDE ( _SPIFFS_page = 0x100 );
PROVIDE ( _SPIFFS_block = 0x1000 );
INCLUDE "../ld/eagle.app.v6.common.ld"
```

- eagle.flash.1m144.ld
```
/* Flash Split for 1M chips */
/* sketch 855KB */
/* spiffs 144KB */
/* eeprom 20KB */

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xd5ff0
}
PROVIDE ( _SPIFFS_start = 0x402D7000 );
PROVIDE ( _SPIFFS_end = 0x402FB000 );
PROVIDE ( _SPIFFS_page = 0x100 );
PROVIDE ( _SPIFFS_block = 0x1000 );
INCLUDE "../ld/eagle.app.v6.common.ld"
```

Links2004, 2015-12-07:
a smaller block size the 0x1000 (4KB) make no sense since the SPI flash block size is 4KB.
yes 0x1000 is ok.
the aliment shut always be to 4KB (internal flash block size) for data areas that are changed by software.
for code (irom0_0_seg the minimum alignment is 4 but 16 is recommended)

you changes looking good.

hallard, 2015-12-07:
@Links2004 
Thanks for your help, I need now to check with @igrr my strange [SPIFFS size](https://github.com/igrr/mkspiffs/issues/11#issuecomment-162424554) problem ;-)

closed 2015-12-09 by hallard

joaolrc, 2020-04-08:
Hello All. I am trying to resize OTA to max and do not use SPIFFS at all. In arduino IDE thats as easy as selecting an option in the toolbar but in platformio im not understanding how exactly i can do that. Can anyone help me please?