esp8266/Arduino #1164: I2C not working well

Andreotti100, 2015-12-06:
I tested a simple GY-21 temperature/humidity sensor with I2C protocol. It does not work very well. Sometimes I get a reading, so I know the ports are right. 
I used 5.1K pull-up resistors on the SDA and SCL pins 4 and 5.

Is there a way to improve that?

jeffryr, 2015-12-07:
Which si7021 library are you using, there are a few floating around.

Andreotti100, 2015-12-07:
I tried these two:

https://forum.pjrc.com/threads/29237-Si7021-Humidity-and-Temperature-Sensor-Library
and
https://github.com/mlsorensen/SI7021

mtnbrit, 2015-12-07:
typically these devices need some time to take the measurement, neither of these libs seem to include any delay between writing the 'measure' command and reading the result. I have delay(75) in my code and it works reliably.

Andreotti100, 2015-12-07:
A delay of 25 or higher does the trick, many thanks!

Andreotti100, 2015-12-07:
My code:

void SI7021::_command(byte * cmd, byte * buf ) {
    _writeReg(cmd, sizeof cmd);
   delay(25);
    _readReg(buf, sizeof buf);
}

devyte, 2017-10-19:
Closing as resolved.

closed 2017-10-19 by devyte