esp8266/Arduino #1168: WiFi.config signature differs from Arduino docs

FWeinb, 2015-12-07:
The [official Arduino docs](https://www.arduino.cc/en/Reference/WiFiConfig) state that the WiFi.config signature should look like this:
 
```
WiFi.config(ip); 
WiFi.config(ip, dns); 
WiFi.config(ip, dns, gateway); 
WiFi.config(ip, dns, gateway, subnet); 
``` 

But for the esp8266 lib is is implemented like:

```
WiFi.config(ip);
WiFi.config(ip, gateway, subnet, dns);
```
See [ESP8266WiFi.h](https://github.com/esp8266/Arduino/blob/40da463ee631133a16fc8ddde7c82d507060068b/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L95-L104)

This is a little confusing.

Links2004, 2015-12-07:
dns support where implemented later for ESP.
since every one already using 
```WiFi.config(ip, gateway, subnet);```
at this point we add it to the end, or all the existing code where broken.

but yes its can be confusing.

FWeinb, 2015-12-07:
Maybe in an major update this could be done. Just to smooth out some edges. Great work so far, really delighted with this amazing project.

marvinroger, 2015-12-16:
Btw, is the project following semver?

ohosrry, 2016-03-17:
I also found this, this troubled me for a long time.can anybody fix it?

Links2004, 2016-03-23:
there is no real "fix" since if we change now, we break all existing code on every ESP.
one of two cases will be always broken.

ohosrry, 2016-03-24:
how about a "version macro" ? like:
`#ifdefine _V1_
    //arg1,arg2...
#else
    //arg2,arg1...
#endif`

Links2004, 2016-03-24:
if you define something in the ino file the define not exists in any other cpp file.
so its not that easy.
the only way to use defines for all files are setting in the plattform.txt or over the menu (like for debug)
but that not a good way to do it for this problem.

ohosrry, 2016-03-24:
whatever,you will change the function's declaration sooner or later.
you must make it identical with arduino docs,because users will be more and more.
btw,everyone comes here is for convenience,but this problem is very inconvenient. especially to new guy.
if code dose not work,we don't know what's wrong,because there is no diagnostic trace about gateway and netmask.

Links2004, 2016-03-24:
i think that is something we can change when we go to version 3.0.0,
 since with this version will have many changes in the core and will not be full compatible in all cases anyway.

igrr, 2016-06-14:
Just had an idea — we can make transition smooth because it's easy to detect argument order.
We need to distinguish between
`WiFi.config(ip, dns, gateway, subnet);`
and
`WiFi.config(ip, gateway, subnet, dns);`
So we can check the third argument — if leading octet is not 255, it means that old argument order was used, and we can correct for that.
Also reclassifying this as a bug.

devyte, 2017-11-20:
@igrr the check for the octet to choose the argument order interpretation would be done at runtime. How can we deprecate the "bad" order? Maybe print a message to Serial?

closed 2018-01-02 by igrr