esp8266/Arduino #1169: WiFi Telnet Server for Serial Monitor use - provided

rudi48, 2015-12-07:
Hello,
I needed for the **stand alone operation** application debugging information. 
For that purpose I thought, that a **Telnet connection** would be the right choice. 
Unfortunately I did not found any example for that. 
So, I had to find out. The result is shown in the source listing on this page. 
The program was tested with success with the following **Telnet Clients**:
 * telnet (Linux NetKit 0.17) Linux Ubuntu 14.04 32Bit
 * PuTTY version 0.63, Windows 7 64Bit
 * telnet Mac OS X 10.11.1 

```C
// File: TelnetServer.ino for ESP8266 NodeMCU
// 2015-12-07 Rudolf Reuter www.rudiswiki.de/wiki9 (search for "wifi")
//
// Developed for debug purpose, use like the Arduino Serial Monitor.
//
// If the Telnet session is lost, depending on the Telnet client,
// you need to press RESET on the NodeMCU and restart the Telnet client.
/*
 *   This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
 */

#include <ESP8266WiFi.h>

// Set these to your desired softAP credentials. They are not configurable at runtime.
const char *ssid = "..........";
const char *password = "..........";

boolean debug = false;  // true = more messages
//boolean debug = true;

// LED is needed for failure signalling
const short int BUILTIN_LED2 = 16;  //GPIO16 on NodeMCU (ESP-12)

unsigned long startTime = millis();

// provide text for the WiFi status
const char *str_status[]= {
  "WL_IDLE_STATUS",
  "WL_NO_SSID_AVAIL",
  "WL_SCAN_COMPLETED",
  "WL_CONNECTED",
  "WL_CONNECT_FAILED",
  "WL_CONNECTION_LOST",
  "WL_DISCONNECTED"
};

// provide text for the WiFi mode
const char *str_mode[]= { "WIFI_OFF", "WIFI_STA", "WIFI_AP", "WIFI_AP_STA" };

//----------------------- WiFi handling
void connectWifi() {
  Serial.print("Connecting as wifi client to SSID: ");
  Serial.println(ssid);

  // use in case of mode problem
  WiFi.disconnect();
  // switch to Station mode
  if (WiFi.getMode() != WIFI_STA) {
    WiFi.mode(WIFI_STA);
  }

  WiFi.begin ( ssid, password );

  if (debug ) WiFi.printDiag(Serial);

  // ... Give ESP 10 seconds to connect to station.
  unsigned long startTime = millis();
  while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  // Check connection
  if (WiFi.status() == WL_CONNECTED) {
    Serial.print("WiFi connected; IP address: ");
    Serial.println(WiFi.localIP());
  } else {
    Serial.print("WiFi connect failed to ssid: ");
    Serial.println(ssid);
    Serial.print("WiFi password <");
    Serial.print(password);
    Serial.println(">");
    Serial.println("Check for wrong typing!");
  }
}  // connectWiFi()

void signalError() {  // loop endless with LED blinking in case of error
  while(1) {
      digitalWrite(BUILTIN_LED2, LOW);
      delay(300); // ms
      digitalWrite(BUILTIN_LED2, HIGH);
      delay(300); // ms
  }
}

// declare telnet server (do NOT put in setup())
const uint16_t aport = 23; // standard port
WiFiServer TelnetServer(aport);
WiFiClient TelnetClient;

void setup() {
  delay(1000);
  Serial.begin(115200);
  delay(1000);
  Serial.println("Sync,Sync,Sync,Sync,Sync");
  delay(500);
  Serial.println();
  // signal start
  pinMode(BUILTIN_LED2, OUTPUT);
  digitalWrite(BUILTIN_LED2, LOW);
  delay(100); // ms
  digitalWrite(BUILTIN_LED2, HIGH);
  delay(300); // ms

  Serial.print("Chip ID: 0x");
  Serial.println(ESP.getChipId(), HEX);

  Serial.println ( "Connect to Router requested" );
  connectWifi();
  if (WiFi.status() == WL_CONNECTED) {
    Serial.print("WiFi mode: ");
    Serial.println(str_mode[WiFi.getMode()]);
    Serial.print ( "Status: " );
    Serial.println (str_status[WiFi.status()]);
    // signal WiFi connect
    digitalWrite(BUILTIN_LED2, LOW);
    delay(300); // ms
    digitalWrite(BUILTIN_LED2, HIGH);      
  } else {
    Serial.println("");
    Serial.println("WiFi connect failed, push RESET button.");
    signalError();
  }

  TelnetServer.begin();
  TelnetServer.setNoDelay(true);
} // setup()

void loop() {
  if (millis() - startTime > 2000) { // run every 2000 ms
    startTime = millis();
    if (!TelnetClient) {  // otherwise it works only once
        TelnetClient = TelnetServer.available();
    }
    if (TelnetClient.connected()) {
      TelnetClient.print("Telnet Test, millis: ");
      TelnetClient.println(millis());
    } else {
      Serial.println("Please connect Telnet Client, exit with ^] and 'quit'");
    }
  }
}
```
Regards, Rudolf

drmpf, 2015-12-07:
You could also try a serial to WiFi bridge like my
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html
which has webpage config to enter your network parameters and the ip and port to listen on.
Just set port 23 for telent.

me-no-dev, 2015-12-10:
Your loop will leak on new connections
```cpp
WiFiServer TelnetServer(ota_port);
WiFiClient Telnet;

void setup(){
  TelnetServer.begin();
  TelnetServer.setNoDelay(true);
}

void loop(){
  if (TelnetServer.hasClient()){
    if (!Telnet || !Telnet.connected()){
      if(Telnet) Telnet.stop();
      Telnet = TelnetServer.available();
    } else {
      TelnetServer.available().stop();
    }
  }
  if (Telnet && Telnet.connected() && Telnet.available()){
    while(Telnet.available())
      Serial.write(Telnet.read());
  }
}
```

rudi48, 2015-12-11:
Hello me-no-dev,
Thank you very much for commenting my program.
Unfortunately I do not understand, what you mean with **leak on new connnections**.
I have tried to setup a **memoryFree()** function, as it exists in AVR Arduino, with no success.
Next step was to make your program understandable for me. The **loop()** now looks:
```C
void loop() {
  if (millis() - startTime > 2000) { // run every 2000 ms
    startTime = millis();
    // Test
    if (TelnetServer.hasClient()) {
      Serial.println("Please connect Telnet Client, exit with ^] and 'quit'");
      if (!TelnetClient || !TelnetClient.connected()) {
        if (TelnetClient) {
          TelnetClient.stop();
          Serial.println("TelnetClient Stop");
        }
        TelnetClient = TelnetServer.available();
      } else {
        TelnetServer.available().stop();
        Serial.println("TelnetServer Stop");
      }
    }
    if (TelnetClient && TelnetClient.connected() && TelnetClient.available()) {
      while(TelnetClient.available()) {  // get data from Client
        Serial.write(TelnetClient.read());
      }
    }
    if (TelnetClient.connected()) {  // send data to Client
      TelnetClient.print("Telnet Test, millis: ");
      TelnetClient.println(millis());
    }
  }
}
```
The output in **Serial Monitor** was:
`Please connect Telnet Client, exit with ^] and 'quit`
`TelnetClient Stop`
`��%�������� ��!��"��'����#`
`123 <-data from Client`

The output in the **Telnet Client window:**
`$ telnet 192.168.17.231`
`Trying 192.168.17.231...`
`Connected to 192.168.17.231`
`Escape character is '^]'.`
`Telnet Test, millis: 1216875`
`Telnet Test, millis: 1218875`
`123  <- typed with keyboard`

Questions:
1. Do you have an explanation for the special characters after **TelnetClient Stop**?
2. I could disconnect with the Telnet Client and reconnect, but I never saw
the message **TelnetServer Stop**. What is then ` TelnetServer.available().stop();` for?
3. What do you mean with **leak on new connnections**?
Regards, Rudolf

me-no-dev, 2015-12-16:
by leak I mean that new connections that come in will hang and use memory because you were not handling them.
You should not check the network every 2 seconds in the loop! that should be done as often as possible (each loop). There is no problem for you to send data every 2 seconds or so, but checks and data read should be done all the time to keep the network stack happy.
The example I gave you is almost all you need and only has the sending to clients missing.
I also believe there is a WiFiTelnetToSerial example that can show you more.

rudi48, 2015-12-16:
Hello me-no-dev,
Thank you for the explanations. 
The existing example program **WiFiTelnetToSerial.ino** is for bidirectional data exchange and multiple clients. So far so good. I looked for something simple, to just print **debug messages via network** in case the ESP8266 does not have a serial link connected. Such a simple program I did not found
As a beginner in that field, I have learned something. Now I also know, how to get the **memoryFree()** value:
```C
Serial.print("Free Heap [B]: ");
Serial.println(ESP.getFreeHeap());
```
Regards, Rudolf

rudi48, 2015-12-17:
Hello me-no-dev,
I have spend some more time to improve the program. Now it works as it should. The program listing in the first post is updated.

The output of the **special characters** can be suppressed with 
```C
serverClient.flush();  // clear input buffer, else you get strange characters
```
Regards, Rudolf

tablatronix, 2016-03-05:
very useful code thanks.

now if there is a way to redirect Serial streams or grab data from their buffers it would be nice to have serial redirect capability. Arduino ota must do this somehow.

Testato, 2016-05-02:
Thanks for the code.

If i remove the redirecting code: 
```
    while(serverClient.available()) {  // get data from Client
    Serial.write(serverClient.read());
  }
```
when i write something in the telnet window, the Telnet server go down.

Why ?
It is necessary have it also if i do not want redirect telnet to serial ?

me-no-dev, 2016-05-02:
It happens because you overflow the heap not reading the contents of the client.
The following will just read the data to nowhere and keep your sketch running.
```cpp
while(serverClient.available())
  serverClient.read();
```

pieman64, 2016-08-19:
@rudi48 @me-no-dev my Google search for ESP8266 and Telnet and have brought me here.

I have also seen http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html and https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino

Are the 3 sketches that are available, for an ESP8266 to connect to a Telnet server or for the ESP8266 to be a Telnet server? I am looking for the former. Just a quick connect, send a command and exit.

rudi48, 2016-08-19:
@pieman64, on the title line it says "Server".
The reason for that was, I found nothing like that in the examples.
Regards, Rudolf

drmpf, 2016-08-19:
> I have also seen http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html and https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino

The setup() method in both links above have the line
server.begin();
These are both setting up the ESP2866 as a server to be connected to.
The  http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html  has some extra setup code to allow you to configure the ESP8266 network ssid and pw via a web page.

tablatronix, 2016-08-19:
Isnt a telnet client trivial? Can you just use wificlient and streams? Or just use any arduino telnet client code and use the wificlient with it?

tablatronix, 2016-08-20:
like this, no idea how to implement ssh, but telnet login scripting should be easy.

```C++
/*
  Telnet client

 This sketch connects to a a telnet server (http://www.google.com)

 created 14 Sep 2010
 modified 9 Apr 2012
 by Tom Igoe

 */

#include <ESP8266WiFi.h>

const char* ssid     = "ssid";
const char* password = "password";

// Enter the IP address of the server you're connecting to:
IPAddress server(10, 0, 1, 10);
int port = 80;

// Initialize the Wifi client library
WiFiClient client;

void setup() {
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // if you get a connection, report back via serial:
  if (client.connect(server, 80)) {
    Serial.println("connected");
  }
  else {
    // if you didn't get a connection to the server:
    Serial.println("connection failed");
  }
}

void loop()
{
  // if there are incoming bytes available
  // from the server, read them and print them:
  if (client.available()) {
    char c = client.read();
    Serial.print(c);
  }

  // as long as there are bytes in the serial queue,
  // read them and send them out the socket if it's open:
  while (Serial.available() > 0) {
    char inChar = Serial.read();
    if (client.connected()) {
      client.print(inChar);
    }
    delay(0);
  }

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
    Serial.println();
    Serial.print(".");
    client.stop();
    delay(1000);
  }
}
```

pieman64, 2016-08-20:
@tablatronix thanks for the telnet client code, just changed port from 80 to 23.
At present I am just trying to telnet into a Raspberry Pi on my lan.

Serial Monitor currently gives:

```
WiFi connected
IP address: 
192.168.10.171
Connected to server
ÿýÿý ÿý#ÿý'
```
I am not getting a username or password prompt though even if it type them into Serial Monitor.
I have also tried client.print("pi\n\n");  etc but no response.

Any ideas what I am doing wrong?

me-no-dev, 2016-08-20:
what username and password? telnet is not shell like ssh ;)

tablatronix, 2016-08-20:
Not sure how prompt mechanisms are implemented probably some control characters. Ill take a look later at the spec

tablatronix, 2016-08-21:
my raspi says

`SSH-2.0-OpenSSH_6.7p1 Raspbian-5`

so not gonna happen, you would have to remove ssh from telnet and accept only a simple auth scheme or implement libssh or openssh etc, neither of which ive seen done, or else we would have ota serial, maybe it is being worked on along side the ssl crypto..

pieman64, 2016-08-21:
@tablatronix did you add Telnet to your Pi as it isn't included in Jessie (dunno about the older Wheezy)?
I added it to my Pi and surely Telnet is a simple Terminal / Console without any encryption, no?
My other tests were to the scn.org Telnet which at the login prompt just asks you to enter 'visitor' at the login prompt (followed by 5 carriage returns to get to main menu). I wasn't able to work out how to send 'visitor' etc to scn.org from the ESP.

tablatronix, 2016-08-21:
yeah basic telnet is simple tcp, but many interfaces implement a login prompt, and i am not sure how it works, i have some perl scripts to negotiate serial servers years ago, ill take a look and see what I did there, cause i do not have any to test on.

zurikus, 2016-09-24:
Thank you rudi48 for sharing your code!
It was very usefull in my project. Now I'm able to monitor if sensor is doing his job via WiFi.

priyokSaha, 2017-01-26:
How do you include ESPWifi.h into arduino

rudi48, 2017-01-26:
@priyokSaha 
https://github.com/esp8266/Arduino

devyte, 2017-10-19:
Nice example, can serve as starting point for those cases where WiFiTelnetToSerial.ino is somehow not suitable.
Closing.

closed 2017-10-19 by devyte