esp8266/Arduino #1174: Burning bootloader java exception

wmarchewka, 2015-12-08:
Hello.  I am receiving a java exception when trying to burn bootloader. Programming works fine.
Specs.  
Arduino IDE 1.6.5
esp8255 arduino 2.0.0
osx  10.11.1

Error while burning bootloader.
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:304)
	at processing.app.Editor$57.run(Editor.java:2777)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Links2004, 2015-12-08:
the ESP not have a bootloader that can be burned.
it has an internal Serial bootloader, this one can not be changed and is Integrated in every ESP that you can buy.

wmarchewka, 2015-12-08:
OK thank you.  My software would load fine but it did not seem to work if unplugged from serial.  It appears the serial monitor needed restarted.  Please close issue.

closed 2015-12-08 by wmarchewka