esp8266/Arduino #1177: HTTP OTA (ESP8266httpUpdate)

ikin77, 2015-12-08:
I'm trying to add HTTP OTA to one of my projects. First, I'm using the sample sketch, just changing SSID & password.

I  can see in my Apache access.log that the new sketch is being requested from the ESP8266, with HTTP 200 code (all OK).

The problem is that OTA is not being performed. If I active debug messages from ESP8266httpUpdate library, I can see the following at Serial:

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 265920
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 172032
[httpUpdate]  - current Sketch Size: 265916
[httpUpdate] FreeSketchSpace to low (172032) needed: 265920

Seems that new sketch it's to big. But new sketch is really simple: the same OTA example provided with the library. 

I've compiled the sketch with "Generic ESP8266 Board". If I try to compile it with "NodeMCU 1.0 (ESP-12EModule) " I can upload it with Serial port to the ESP8266, but the module can't connect to the router with that option selected.

What might be going on? 

Thanks in advance and best regards!!!

Links2004, 2015-12-08:
what flash size is on you module?
you can use:
https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino
to check.
most modules that you can buy today have more then 512KB flash.

ikin77, 2015-12-08:
Hi, Links2004. Thanks for your reply. Here's the output of the sketch:

Flash real id:   001640C8
Flash real size: 4194304

Flash ide  size: 524288
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration wrong!

I've uploaded the sketch with "512K (64K SPIFFS)", as usual. It's that ok?

Thanks again :o)

reaper7, 2015-12-08:
try with "512K no SPIFFS"

and additional check my issue:
https://github.com/esp8266/Arduino/issues/1042#issuecomment-161627252
latest version from github generate smaller images

igrr, 2015-12-08:
`Flash real size: 4194304`
You can safely choose "Generic 4M" for flash size.

ikin77, 2015-12-08:
Thanks again for the answers.

512K no SPIFFS
=============

A bit better, but not enough. Now sketch is just for OTA testing. If works, I'd insert it in my real sketch, so, sice will be higher.

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 265920
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 237568
[httpUpdate]  - current Sketch Size: 265916
[httpUpdate] FreeSketchSpace to low (237568) needed: 265920
HTTP_UPDATE_FAILD

"4M (1M SPIFFS)"
==================

No wifi connection!!!! I can't even "ping" the ESP8266. I can't see activity at my Apache HTTP acces.log.

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: -1
[httpUpdate]  - len: -1
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 2879488
[httpUpdate]  - current Sketch Size: 265916
[httpUpdate] Code is (-1)
HTTP_UPDATE_FAILD
Versión 1.0.3

========================

No way :'o(

Thanks for your help!!!!

scargill, 2015-12-08:
I am getting a similar issue. I have put the .bin file in the root of my publically accessible NAS drive and in a browser put in the address and file name (tester.bin) - it's there - it downloaded.

Now I've put my ssid and password and the same link into httpdUpdate example - and it gets the connection to the WIFI and says "HTTP_UPDATE_FAILD" - but here's the thing - it says the same whether or not there the bin file is in that directory - surely if the file is not there, it should say "HTTP_UPDATE_NO_UPDATES" - regardless of whether there is enough RAM or not (the file is 210k)  ???

ikin77, 2015-12-09:
I think my problem it's related to #1147. In fact, I understand why I can't do OTA: just because not enough free space to load new sketch. The problem is that I change to 4M SPIFFS, WiFi doesn't work at all, no connection to the AP. In #1147 says connection is too slow. For me, it's not connection at all.

Same code. Code is properly loaded (I can see Serial messages), but no WiFi connection. Just connection with 512K option.

Links2004, 2015-12-09:
@scargill enable debug to see why it fails:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L35
if no file at your webserver the server will send http code 404 (not found) this is an error (wrong url or no file uploaded to the server),
I not see how this shut be "no updates". 
you only get "no updates" when the server responses with 304 (Not Modified).
if the server response whit 200 (OK) the update will by started if there is enough free space for the download.
any problem during this process will be handled as error.

@ikin77
code: -1 mean no connection to server. 
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h#L37
you can try to erase the flash may there is garbage on the "WiFi config space" in 4MB flash mode.
https://github.com/esp8266/Arduino/blob/master/doc/eclipse/makefile.init#L32

ikin77, 2015-12-09:
@Links2004

Hi, Links2004. Thanks for your help. In fact, the problem is that in 4Mb flash mode I have no Wifi connection. I can't ping ESP8266 from my module and I can't see activity at my Apache access.log file.

I'm seeing the script  you suggest me to run to erase the flash. It's a "makefile.init" file and I don't know how to run it, the first time I see this language. How can I run it? I can't see much information on Internet about this "makefile.init" file :o/

Anyway, I've tried with two different ESP8266 modules with same result :o(

Thanks!!!

Links2004, 2015-12-09:
you only need to run the line.
the "makefile.init" is a normal make file.

scargill, 2015-12-09:
Hah - you are right Andrei - no topic :-)  - ok I have my little LEDs running a treat.  I'll update the blog - thank you for this - a great new toy to play with.

ikin77, 2015-12-10:
@Links2004 
@scargill 

Thanks for yor help again. I'm to busy today and I can't try the solution you suggest. I'll try it tomorrow and I'll let you know :)

Regards!!

ikin77, 2015-12-12:
HI. I'm still stuck with the problem, trying to find out what's wrong. Just a question: if I choose Nodemcu instead of Generic ESP8266 as board in Arduino IDE, I'm supposed to install before firmware Nodemcu with esptool.py in my ESP8266.

I haven't do it. That might be trying cause of the problem?

Links2004, 2015-12-12:
no the original Nodemcu firmware has noting to do with this project, and is not needed.
did you try to erase the flash with esptool.py?

ikin77, 2015-12-12:
No, I haven't still run the script you gave to me. But I've flashed several times the esp8266 with expressif firmware. I've also tried with two different units. I'll try to run tomorrow the erase script and I'll let you know. Thanks for your fast reply!!!!

ikin77, 2015-12-12:
Hi again. I erased the flash and now I can upload a sketch Flash size 4M (3M SPIFFS) and have wifi connection!!!!

The bad thing is that I've arrived to the next error: the unit restarts before OTA. Debuggin, I'm sure of being entering the method "ESP8266HTTPUpdate::runUpdate" and the crash arrives in "Update.writeStream(in)", line 185. Any ideas?

Links2004, 2015-12-13:
du you running the latest git version?
I have done some changes to writeStream / Update class there to fix some other problems.
if not please retry with latest git:
https://github.com/esp8266/Arduino#using-git-version-

when it make still problems please enable the debug output:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L35
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater.h#L20
and post the log here.

ikin77, 2015-12-13:
I've been doing more debug....the problem is in "UpdaterClass::_writeBuffer()", when "ESP.flashWrite" is invoked.
I'm not sure of being using latest git, I'm going to check and I'll let you know. Thanks a lot!!!!

ikin77, 2015-12-13:
Getting better....

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 267408
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 778240
[httpUpdate]  - current Sketch Size: 267400
[httpUpdate] runUpdate flash...
[httpUpdate] Update ok

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0
ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

The problem is that after reboot, the new sketch is not running (it's the same I used to test OTA). Trying to find out what's going on. Is relevant to choose as board Generic ESP8266 vs NodeMCU? Actually, I'm using just an ESP8266, not NodeMCU. I keep myself doing more test. I'll let you know.

Regards

scargill, 2015-12-13:
Here’s what I did.

 

I compiled a version of the MQTT software and tested it on an ESP12 board.

 

I ran the example HTTPUPDATE software – using latest code from GIT today on Arduino 1.6.5

 

The HTTPUPDATE loaded up – and then I got UPDATE FAILD..  So instead of using the 512K setup I used the 4MEG setup.

 

This time it waited a while – looking like it might be loading the update and then…

 

[SETUP] WAIT 3...

[SETUP] WAIT 2...

[SETUP] WAIT 1...

 

ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 

 

ets Jan  8 2013,rst cause:4, boot mode:(1,7)

 

wdt reset

 

 

Nothing – sat there DEAD. I rebooted the board and indeed it HAD uploaded the new software – which then ran perfectly..

 

But if it does not reboot properly after the update it is all a bit pointless as the point of remote updating is not to have to disconnect the board to reboot it..

 

Any ideas?

 

https://github.com/esp8266/Arduino#using-git-version-

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L35
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater.h#L20

ikin77, 2015-12-13:
I'm running the latest GIT. I'm having the problem in function UpdaterClass::_writeBuffer().

I've added some Serial.println to have extra debug info:

===========================
bool UpdaterClass::_writeBuffer(){
  Serial.println("WRITE BUFFER AAA");
  yield();
  bool result = ESP.flashEraseSector(_currentAddress/FLASH_SECTOR_SIZE);
  yield();
  Serial.println("WRITE BUFFER BBB");
  if (result) {
      Serial.println("WRITE BUFFER CCC");
      result = ESP.flashWrite(_currentAddress, (uint32_t*) _buffer, _bufferLen);
 	  Serial.println("WRITE BUFFER DDD");
  }
  yield();
Serial.println("WRITE BUFFER EEE");
  if (!result) {
	Serial.println("WRITE BUFFER FFF");
    _error = UPDATE_ERROR_WRITE;
    _currentAddress = (_startAddress + _size);
#ifdef Serial
    printError(Serial);
#endif
    return false;
  }
Serial.println("WRITE BUFFER GGG");
  _md5.add(_buffer, _bufferLen);
Serial.println("WRITE BUFFER HHH");
  _currentAddress += _bufferLen;
Serial.println("WRITE BUFFER III");
  _bufferLen = 0;
  return true;
}

===========================

The last thing I see at Serial monitor is:

WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC

Soft WDT reset

========

So, I think the error occurs in "ESP.flashWrite(_currentAddress, (uint32_t*) _buffer, _bufferLen);". I go on with my debugging....

Links2004, 2015-12-13:
the reboot problem is tracked here: #1017
and its a general reboot problem (not linked to OTA), 
and has to do with how the sketch is started the first time after serial upload.

scargill, 2015-12-13:
So..

 

If I blow the HTTPUPDATE software and pull the power just before it actually does anything… reconnect – it does INDEED go off and do the update and all is well…

 

BUT..

 

[SETUP] WAIT 1...

 

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

 

load 0x4010f000, len 1264, room 16 

tail 0

chksum 0x42

csum 0x42

@cp:0

Ld

 

So there it is – then the program starts – all is well.. but if you look at the source- somewhere it is supposed to say “HTTPUPDATE OK” – I’m no seeing that anywhere…

scargill, 2015-12-13:
Also I wonder if anyone can help..

 

So I used the HTTPUPDATE software to blow the MQTT example….

 

So I update the software – and the MQTT software start up – and that’s fine.

 

But what I’d like to do – and I imagine most people using this would – is to incorporate the HTTPUPDATE software into something existing – like for example – the MQTT example – so that one could send a remote command – for the software to update itself.

 

But the HTTPUPDATE software uses a completely different set of includes and the WIFI setup uses a different function call…

 

Is the ESP8266WiFiMulti really necessary or can one simply add these two

 

#include <ESP8266HTTPClient.h>

#include <ESP8266httpUpdate.h>

 

And this

 

t_httpUpdate_return ret = ESPhttpUpdate.update("http://whatever.domain/tester.bin");

 

To, say the MQTT normal MQTT example???

 

Pete.

Links2004, 2015-12-13:
the "HTTPUPDATE OK" will not be shown see: #1202

scargill, 2015-12-13:
Thanks. Just need to figure out how to integrate that into the MQTT or other actual program…

marvinroger, 2015-12-13:
@scargill I don't understand what is your problem. You don't know how to "mix" MQTT and OTA? A sample sketch of what you're trying to do would be:

```c++
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ESP8266httpUpdate.h>

const char* ssid = "........";
const char* password = "........";
const char* mqtt_server = "broker.mqtt-dashboard.com";

WiFiClient espClient;
PubSubClient client(espClient);

void setup() {
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
}

void setup_wifi() {
  delay(10);

  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  String message = String();
  for (int i = 0; i < length; i++) {
    char input_char = (char)payload[i];
    message += input_char;
  }

  if (message == "ota") { // When we receive ota notification from MQTT, start the update
    ESPhttpUpdate.update("http://whatever.domain/tester.bin");
  } 
}

void reconnect() {
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    if (client.connect("ESP8266Client")) {
      Serial.println("connected");
      client.subscribe("devices/esp8266/control"); // Subscribe to your MQTT topic
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      delay(5000);
    }
  }
}
void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
}
```

What's the problem?

scargill, 2015-12-13:
Thank you – your version is simpler than mine – what I could not understand was that in the MQTT example, the WIFI includes were completely different to the ones used in the OTA example.

 

But I can see from your example here that I don’t need to concern myself with these two for example (as used in the OTA example)

 

#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

ikin77, 2015-12-14:
Any idea about my restart when invoking "ESP.flashWrite(_currentAddress, (uint32_t*) _buffer, _bufferLen);"? 

OTA HTTP update worked just once for me. I get rebooted the unit everytime I try to do OTA.

Thanks again!!!!

Links2004, 2015-12-14:
@ikin77  do it crash at the first flashWrite or at some point later?

ikin77, 2015-12-14:
@Links2004 

Hi. I appreciate very much your help!! The first ESP.flashWrite causes the crash. I added some "Serial.print" in function "UpdaterClass::_writeBuffer().":

    bool UpdaterClass::_writeBuffer(){
        Serial.println("WRITE BUFFER AAA");
        yield();
        bool result = ESP.flashEraseSector(_currentAddress/FLASH_SECTOR_SIZE);
        yield();
        Serial.println("WRITE BUFFER BBB");
        if (result) {
                Serial.println("WRITE BUFFER CCC");
                result = ESP.flashWrite(_currentAddress, (uint32_t*) _buffer, _bufferLen);
                Serial.println("WRITE BUFFER DDD");
        }
        yield();
        Serial.println("WRITE BUFFER EEE");
        if (!result) {
                Serial.println("WRITE BUFFER FFF");
                _error = UPDATE_ERROR_WRITE;
                _currentAddress = (_startAddress + _size);
                #ifdef Serial
                printError(Serial);
                #endif
                return false;
                 }
                Serial.println("WRITE BUFFER GGG");
                _md5.add(_buffer, _bufferLen);
                Serial.println("WRITE BUFFER HHH");
                _currentAddress += _bufferLen;
                Serial.println("WRITE BUFFER III");
                _bufferLen = 0;
                return true;
        }

The last thing I see at Serial monitor is:

        WRITE BUFFER AAA
        WRITE BUFFER BBB
        WRITE BUFFER CCC

        Soft WDT reset

Thanks again :o)

Links2004, 2015-12-14:
can you add
```cpp
  Serial.printf("[flashWrite] _startAddress:     0x%08X (%d)\n", _startAddress, _startAddress);
  Serial.printf("[flashWrite] _size:             0x%08X (%d)\n", _buffer, _buffer);
  Serial.printf("[flashWrite] _bufferLen:   0x%08X (%d)\n", _bufferLen, _bufferLen);
```
after
```cpp
   Serial.println("WRITE BUFFER CCC");
```
may this info give some hint.

ikin77, 2015-12-14:
@Links2004 

I'll do it this afternoon and I'll let you know. Thanks again :o)

itsjustvenky, 2015-12-14:
try 

```
wdt_disable();
result = ESP.flashWrite(_currentAddress, (uint32_t*) _buffer, _bufferLen);
wdt_enable();
```

I am guessing that "flashWrite" is taking too much time. I don't know what is the "_buffer" size.

Links2004, 2015-12-14:
when flashWrite works well then there will be no WDT.
It write one flash segment (4KB) at one.

the WDT comes for a reason we need to find this reason and fixing it.
since it work the first time there must be reason why the second update failed.
simply disabling WDT is a workaround, not a solution.

itsjustvenky, 2015-12-14:
@Links2004  Thanks. Yes. Lets wait for the info you asked from @ikin77. Even I want to know what are those three things.

Do you know any other reason this can happen "Soft WDT reset". Let me know.

ikin77, 2015-12-14:
@Links2004 
@itsjustvenky 

I added the lines @Links2004 suggested. Firs time I saw the message up to "WRITE BUFFER AAA", and after that, I saw the "Soft WDT reset".

I tried again, and got the following:

WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B9000 (757760)
[flashWrite] _size:             0x3FFF5C68 (1073699944)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B9000 (757760)
[flashWrite] _size:             0x3FFF5C
Soft WDT reset

========

So, the Soft WDT reset is unpredictable :o(.

 I'll try @itsjustvenky and I'll inform you...

Links2004, 2015-12-14:
please  add ```Serial.flush()``` before the ```flashWrite```.
any may rename "_size" to "_buffer ptr" I miss the naming ;)
the serial seams to be truncated.
at the first look its strange that we have the same _startAddress twice.

ikin77, 2015-12-14:
@Links2004 
Thaks again. Here's the new output....
===================
   Iniciando OTA....
   [httpUpdate] Header read fin.
   [httpUpdate] Server header:
   [httpUpdate]  - code: 200
   [httpUpdate]  - len: 291280
   [httpUpdate] ESP8266 info:
   [httpUpdate]  - free Space: 753664
   [httpUpdate]  - current Sketch Size: 291268
   [httpUpdate] runUpdate flash...
   WRITE BUFFER AAA
   WRITE BUFFER BBB
   WRITE BUFFER CCC
   [flashWrite] _startAddress:     0x000B8000 (753664)
   [flashWrite] __buffer ptr:       0x3FFF5680 (1073698432)
   [flashWrite] _bufferLen:   0x00001000 (4096)
   WRITE BUFFER DDD
   WRITE BUFFER EEE
   WRITE BUFFER GGG
   WRITE BUFFER HHH
   WRITE BUFFER III
   WRITE BUFFER AAA
   WRITE BUFFER BBB
   WRITE BUFFER CCC
   [flashWrite] _startAddress:     0x000B8000 (753664)
   [flashWrite] __buffer ptr:       0x3FFF5680 (1073698432)
   [flashWrite] _bufferLen:   0x00001000 (4096)
   WRITE BUFFER DDD
   WRITE BUFFER EEE
   WRITE BUFFER GGG
   WRITE BUFFER HHH
   WRITE BUFFER III
   WRITE BUFFER AAA

   Exception (0):
   epc1=0x40217d44 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

   ctx: sys 
   sp: 3ffed510 end: 3fffffb0 offset: 01a0

   >>>stack>>>
   3ffed6b0:  401020b9 401020b6 1961221d 221e2c22  
   3ffed6c0:  400005e1 1b61221f 22202c22 00000000  
   3ffed6d0:  40217d44 00000033 00000016 00000000

Links2004, 2015-12-14:
strange now we have a null ptr, may be course an interrupt during the flash is blocked.
do you or a library using any interrupt?

when you make a obj dump for the address, i think we see an function called from a interrupt.

ikin77, 2015-12-14:
I'm using the following libraries:

#include <Wire.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <Timers.h>
#include <OneWire.h>
#include <DS3231.h>
#include <DallasTemperature.h>

Maybe any of them use interrupts, but I've disabled them to avoid problems:

        noInterrupts();
        t_httpUpdate_return ret = ESPhttpUpdate.update("http://192.168.1.100/placasolar.bin");
        interrupts();

The result is more or less similar:

Iniciando OTA....
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 291328
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 753664
[httpUpdate]  - current Sketch Size: 291316
[httpUpdate] runUpdate flash...
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5678 (1073698424)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER GGG
WRITE BUFFER HHH
WRITE BUFFER III
WRITE BUFFER AAA

Exception (0):
epc1=0x40217d64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3fff0c60 end: 3fffffb0 offset: 01a0

Links2004, 2015-12-14:
try to make a objdump from the elf that match to the bin file.
```xtensa-lx106-elf-objdump -S some_name.elf > some_name.dobj```

then we hopefully see where the problem is coming from.
the way you disable  Interrupts will not work, since the update itself need network.
you better try:

```cpp
noInterrupts();
result = ESP.flashWrite(_currentAddress, (uint32_t*) _buffer, _bufferLen);
interrupts();
```

note: I edit your post for better reading using Markdown code tags

igrr, 2015-12-15:
or rather, try changing `FLASH_INT_MASK` as follows:
```c++
static const int FLASH_INT_MASK = ((B10 << 8) | B11111110);
```

ikin77, 2015-12-15:
@Links2004
@igrr

Thanks to you. I won't be able today to do any testing. Tomorrow I'll try and I'll inform you about the results.

Thanks again and best regards :)

ikin77, 2015-12-16:
@Links2004 

Hi again. I tested, twice, adding your suggestion:

   noInterrupts();
   result = ESP.flashWrite(_currentAddress, (uint32_t*) _buffer, _bufferLen);
   interrupts();

First try
=================

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 291312
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 753664
[httpUpdate]  - current Sketch Size: 291300
[httpUpdate] runUpdate flash...
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5878 (1073698936)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER GGG
WRITE BUFFER HHH
WRITE BUFFER III
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5878 (1073698936)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER GGG
WRITE BUFFER HHH
WRITE BUFFER III
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5878 (1073698936)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER GGG
WRITE BUFFER HHH
WRITE BUFFER III
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5878 (1073698936)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER GGG
WRITE BUFFER HHH
WRITE BUFFER III
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5878 (1073698936)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER GGG
WRITE BUFFER HHH
WRITE BUFFER III
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5878 (1073698936)
[flashWrite] _bufferLen:   0x00001000 (4096)

 ets Jan  8 2013,rst cause:4, boot mode:(3,3)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Second try
=================

httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 291312
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 753664
[httpUpdate]  - current Sketch Size: 291300
[httpUpdate] runUpdate flash...
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5578 (1073698168)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER GGG
WRITE BUFFER HHH
WRITE BUFFER III
WRITE BUFFER AAA
WRITE BUFFER BBB
WRITE BUFFER CCC
[flashWrite] _startAddress:     0x000B8000 (753664)
[flashWrite] __buffer ptr:       0x3FFF5578 (1073698168)
[flashWrite] _bufferLen:   0x00001000 (4096)
WRITE BUFFER DDD
WRITE BUFFER EEE
WRITE BUFFER GGG
WRITE BUFFER HHH
WRITE BUFFER III
WRITE BUFFER AAA

Exception (0):
epc1=0x40217d60 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffeeea0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffef040:  401020b9 401020b6 3ffebdb0 00000001  

=====================

No way :o(

Thanks anyway!!! :o)

ikin77, 2015-12-16:
@igrr 

Hi, Igrr. With your suggestion the result is:

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 291312
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 753664
[httpUpdate]  - current Sketch Size: 291300
[httpUpdate] runUpdate flash...
WRITE BUFFER AAA

 ets Jan  8 2013,rst cause:4, boot mode:(3,3)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

ikin77, 2015-12-17:
Hi, any ideas? The lastests GIT updates might be related to this issue?

Thans to all of you :o)

Links2004, 2015-12-17:
the latest changes in Update are related to wrong flash config in IDE vs real flash.
since you verified your config with
https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino
that shut not be the problem.

ikin77, 2015-12-19:
@Links2004 Thanks for your help. I'm waiting to arrive some others ESP8266....I'll try with them. Maybe is a problem of my units :o/

ikin77, 2015-12-24:
Hi again. I think all the problems were mine, I'm sorry. The power suply for ESP8266 was not enough and that was causing my unexpected reboots.

Now things are better, but OTA is not still working for me:

HTTP_UPDATE_FAILD
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 309952
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 2834432
[httpUpdate]  - current Sketch Size: 309944
[httpUpdate] runUpdate flash...
[httpUpdate] Update.writeStream failed! (ERROR[6]: Stream Read Timeout)
[httpUpdate] Update failed
HTTP_UPDATE_FAILD

//////

It looks like the http server it's taking too long to reply. Actually, it's an Apache HTTP Server running in the same network, so, speed is more than enough.

Any ideas??

Thanks again to all of you :o)

Links2004, 2015-12-24:
du you run the latest git?
i have done some changes there, for the timeout problem (#1157 and #1191).

ikin77, 2015-12-24:
Hi, @Links2004. I'm using last git, I downloaded it this morning. I'm seeing posts #1157 and #1191. I'm going to try to Try increase more the stream timeout in writeStream in Updater.cpp.

I'll let you know

ikin77, 2015-12-24:
Some news....in GIT I can see that in Updater.cpp is not present the suggesion of #1157: to add 
"data.setTimeout(10000)" at the beginning of "UpdaterClass::writeStream". I did it and I don't get any more any timeout. Know I can see a different error:

    [httpUpdate] Update.end failed! (ERROR[10]: Magic byte is wrong, not 0xE9)

I'm trying to find out what it means.

ikin77, 2015-12-26:
Any idea about the magic bit message?

Can you check if extra timeout for serial input at http ota update has been committed to latest git?

Thank!!!

Links2004, 2015-12-26:
"Magic byte is wrong, not 0xE9" means the verification of the data written to flash has failed,
the first byte of the image is always 0xE9, but it is not?

most common is the problem that there is not enough power, 
try to add a 100nF and a 10uF or bigger capacitor to the 3V3 line of the ESP.

The ```setTimeout(10000)``` is not in the latest git.
will add it later.

ikin77, 2015-12-26:
Hi, Makus. It's strange because most of times I get the magic byte error message, but every so often update is performed. You can see two attempts in Serial:

Links2004, 2015-12-26:
```ets Jan 8 2013,rst cause:4, boot mode:(1,7)```
the reboot problem is tracked here: #1017
and its a general reboot problem (not linked to OTA), 
and has to do with how the sketch is started the first time after serial upload.

will do some tests with the link tomorrow.

Links2004, 2016-02-29:
the server send the data in chunks, 
the problem will not happen with 2.1.0 since the http update "downgrade to HTTP 1.0" which not supports http chunking.
@ikin77 if there still problems with 2.1.0 feel free to open a new issue for it.

closed 2016-02-29 by Links2004