esp8266/Arduino #1179: Bug in snprintf

exbarboss, 2015-12-09:
Hi,

Was working with ESP8266 and found issue with `snprintf`.

Here is the sample:
```Arduino
char _converted[33];
float value = 10;
int size = sizeof(_converted)-1;
  
snprintf(_converted, size, "Value is %.2f", value);
Serial.print("Core test: "); Serial.println(_converted);
```

Expected result was: `Core test: Value is 10.00`
Actual result is: `Core test: Value is %.2f`

igrr, 2015-12-09:
Please see https://github.com/esp8266/Arduino/issues/73.

exbarboss, 2015-12-09:
Thanks for fast reply! Sorry for duplicating the issue was searching by `snprintf`. Will the support of float be added in future?

igrr, 2015-12-13:
We are actually using vsprintf which is present in the ESP8266 ROM code. Since the code is in ROM, there is no way to extend it with FP format support. 
Rolling our own implementation of sprintf (or taking one from newlib) is possible, but is pretty low on my list.

tofrnr, 2017-06-30:
char* str[100];
// not possible for AVR and nodeMCU but possible for Arduino ARM core
sprintf(s, "floats: %4.2f %+.0e %E \n", 3.1416, 3.1416, 3.1416); 

not possible in Arduino IDE 1.8.2 feat. esp8266 board manager
when is it automatically available via Arduino IDE?

igrr, 2017-06-30:
2.4.0 is in release candidate stage now (2.4.0-rc1). To test it, uninstall the existing version using boards manager, then use the boards manager URL provided in the release notes: https://github.com/esp8266/Arduino/releases/tag/2.4.0-rc1 (see "Obtaining 2.4.0-rc1 pre-release" section near the bottom of the page).

tofrnr, 2017-06-30:
thank you, but then I'll wait for it to come with the Arduino IDE board manager.
Do you have a release date?

closed 2018-01-02 by igrr