esp8266/Arduino #1180: error

ndkeprom, 2015-12-09:
Arduino: 1.6.5 (Windows XP), Board: "WeMos D1, 80 MHz, Serial, 921600, 4M (3M SPIFFS)"

Using library Arduino_ESP8266_libs_master in folder: D:\arduino1.0.4\libraries\Arduino_ESP8266_libs_master (legacy)

C:\Documents and Settings\Admin\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Documents and Settings\Admin\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Documents and Settings\Admin\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266 -IC:\Documents and Settings\Admin\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\d1 -ID:\arduino1.0.4\libraries\Arduino_ESP8266_libs_master C:\DOCUME~1\Admin\LOCALS~1\Temp\build1890523624863991975.tmp\WiFi_Connection.cpp -o C:\DOCUME~1\Admin\LOCALS~1\Temp\build1890523624863991975.tmp\WiFi_Connection.cpp.o 

In file included from WiFi_Connection.ino:7:0:
D:\arduino1.0.4\libraries\Arduino_ESP8266_libs_master/ESP8266_TCP.h:16:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
Multiple libraries were found for "ESP8266_TCP.h"

 Used: D:\arduino1.0.4\libraries\Arduino_ESP8266_libs_master

 Not used: D:\arduino1.0.4\libraries\Arduino-ESP8266_libs-master

Error compiling.

igrr, 2015-12-09:
Apparently you are trying to compile a sketch which uses this library:
https://github.com/INEXTH/Arduino-ESP8266_libs
This library was written for AVRs to use the ESP8266 as a serial peripheral.
For some reason you are trying to use this library on the ESP8266 itself. This won't work.

closed 2015-12-09 by igrr

ndkeprom, 2015-12-09:
Thankyou