esp8266/Arduino #1183: Compiler error at FTP Client program, help needed

rudi48, 2015-12-09:
Hi, I like to adapt http://playground.arduino.cc/Code/FTP to ESP8266.
Most part of the adapted program WiFi_FTP_Client.ino works, but there is a **compiler error** left,
which I can not resolve.
Most part of the program works, except the **file data transfer** to the **FTP server**.
If I declare `const char clientBuf[64] = "aa";`
instead of `byte clientBuf[64];`, it works, but that is not what I need.
If I use the library `Ethernet.h` instead of `ESP8266WiFi.h` it compiles OK, **but no data transfer**.
By the way, if I compile the original program **FTP.ino** with `Ethernet.h`, I get no compiler error.
Please see the error messages (Arduino 1.6.5 and 1.6.6):
```C
In file included from /Volumes/DAT/Users/rudi/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32:0,

                 from /Volumes/DAT/Users/rudi/Documents/Arduino/wifi_FTP_client/wifi_FTP_client.ino:10:
/Volumes/DAT/Users/rudi/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h: In instantiation of 'size_t WiFiClient::write(T&, size_t) [with T = unsigned char [64]; size_t = unsigned int]':

/Volumes/DAT/Users/rudi/Documents/Arduino/wifi_FTP_client/wifi_FTP_client.ino:285:34:   required from here

/Volumes/DAT/Users/rudi/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:119:36: error: request for member 'available' in 'source', which is of non-class type 'unsigned char [64]'

     size_t left = source.available();
                                    ^
/Volumes/DAT/Users/rudi/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:123:5: error: request for member 'read' in 'source', which is of non-class type 'unsigned char [64]'
     source.read(buffer.get(), will_send);
     ^
```
Who can help me to solve that puzzle please?
Regards, Rudolf

The program with the error, the **error lines are marked**:
```C
/*
   FTP passive client for IDE v1.0.1 and w5100/w5200
   http://playground.arduino.cc/Code/FTP
   Modified 6 June 2015 by SurferTim

   You can pass flash-memory based strings to Serial.print() by wrapping them with F().

   2015-12-09 Rudolf Reuter, adapted to ESP8266 NodeMCU
*/
#include <ESP8266WiFi.h>
// #include <Ethernet.h>
#include <FS.h>

// comment out next line to write to SD from FTP server
#define FTPWRITE

// Set these to your desired softAP credentials. They are not configurable at runtime.
const char *ssid = "FRITZ-7170R";
const char *password = ".............";

boolean debug = false;  // true = more messages
//boolean debug = true;

// LED is needed for failure signalling
const short int BUILTIN_LED2 = 16;  //GPIO16 on NodeMCU (ESP-12)

unsigned long startTime = millis();

// provide text for the WiFi status
const char *str_status[]= {
  "WL_IDLE_STATUS",
  "WL_NO_SSID_AVAIL",
  "WL_SCAN_COMPLETED",
  "WL_CONNECTED",
  "WL_CONNECT_FAILED",
  "WL_CONNECTION_LOST",
  "WL_DISCONNECTED"
};

// provide text for the WiFi mode
[... 342 lines cut ...]

void loop() {
  byte inChar;

  if (Serial.available() > 0) {
    inChar = Serial.read();
  }

  if (inChar == 'f') {
    if (doFTP()) Serial.println(F("FTP OK"));
    else Serial.println(F("FTP FAIL"));
  }

  if (inChar == 'r') {
    String fileNameDir;
    Dir dir = SPIFFS.openDir("/");
    while (dir.next()) {
      fileNameDir = dir.fileName();
      size_t fileSize = dir.fileSize();
      Serial.printf("FS File: %s, size: %s\n", fileNameDir.c_str(), formatBytes(fileSize).c_str());
    }
  }
  delay(10);
}
```

Links2004, 2015-12-10:
try to use it like this 
```cpp
dclient.write((const uint8_t *) &clientBuf[0], 64);
```

for a faster upload you shut increase the 64 to 1460.

rudi48, 2015-12-10:
Hello Markus,
Thank you very much, now the compiler error is gone :-)
I gnawed already 3 days at that problem.
After the patch the program works as expected.
I could upload my first file (35 KB) to the FTP server. Jubel!

One question of curiosity, how do found out, what to do?
Is there a tutorial describing that?
For me it is still a mystery.
Regards, Rudolf

Links2004, 2015-12-10:
I read the error message, then I take a look to the code.
then compare the prototype of "write"
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClient.h#L50
```cpp
size_t write(const uint8_t *buf, size_t size);
```
with what you try to give in.

```uint8_t [64]``` vs ```const uint8_t *```.
by this i know there only a typecast is needed (uint8_t is compatible to uint8_t). 
Its basic C / C++ type conversion.

rudi48, 2015-12-10:
Hello Markus,
Thank you very much for the explanation. I have learned something :-)

Unfortunately I looked at line 115 of WiFiClient.h, which was mentioned in the error message:
inline size_t WiFiClient::write(T& source, size_t unitSize) {

The increased buffer size (1460) works also.

Regards, Rudolf

rudi48, 2015-12-10:
Hello Markus,
Does it make sense to place this FTP program to the ESP8266WiFi examples?
Regards, Rudolf

igrr, 2015-12-10:
It would be very useful for others to package this example as a library.

rudi48, 2015-12-10:
Hello Ivan,
Who can make such a library?
I am more or less a copy&paste programmer ;-)
My final source code of the example file is documented on
http://www.rudiswiki.de/wiki9/WiFiFTPServer

Regards, Rudolf

rudi48, 2015-12-10:
I have added an upload / download flag to the program:
http://www.rudiswiki.de/wiki9/WiFiFTPServer

Regards, Rudolf

danbicks, 2016-02-14:
Rudi, must say awesome work. I have tried example with my server it starts all good asks for user
then password which is sent, then just sits there. Any ideas anyone?

Here is serial log:

WiFi connected; IP address: 192.168.0.11
WiFi mode: WIFI_STA
Status: WL_CONNECTED
Ready. Press d, u or r
SPIFFS opened
Command connected
220---------- Welcome to Pure-FTPd [privsep] [TLS] ----------
220-You are user number 8 of 1000 allowed.
220-Local time is now 08:55. Server port: 21.
220-This is a private system - No anonymous login
220-IPv6 connections are also welcome on this server.
220 You will be disconnected after 15 minutes of inactivity.
Return DB Ok..
Send USER
331 User SESP@dbtec.org OK. Password required
Return DB Ok..
Send PASSWORD

Thanks Dans

danbicks, 2016-02-14:
PS my server is a just host one, account works fine tested with FTP client.

Dans

danbicks, 2016-02-15:
Rudi Update.

All working, I removed from the client.print lines the F pointer to store string in ram. All works perfect without it on ESP :)

Superb work, finally can FTP files to a server brilliant work. Would be nice to wrap this lot up in a library.

Dans

rudi48, 2016-02-15:
Hello Dans,
Thank you very much for finding the F pointer problem. However, from the specification:
http://arduino.esp8266.com/versions/1.6.5-947-g39819f0/doc/reference.html
the Program Memory feature "F" should work.
So, I am not sure that this was the cause of your problem.
Regards, Rudolf

danbicks, 2016-02-15:
Hi Rudi,

Really strange I am using 1.6.5 with latest staging version and your example sketch.

First of all would sit there after sending password and wait 15 mins for FTP server disconnection. I added to your ercv() routine a ten second time-out and tried again, timed out after every attempt so I knew it was just waiting for response from ftp server. Just as a try I removed the F pointer from password and username, tried again and it proceeded further. I then removed all of them from the doFtp routine and hey presto works like a treat.

Maybe Igrr could shed some light on this. I am very happy, have tested it today solidly and works every time.

A big thanks again for all your hard work.

Dans

rudi48, 2016-02-15:
Hello Dans,
I believe in what you tested.
I have first tested with Arduino version 1.6.6 and then 1.6.7, but not very intensive.
That may be the difference.
It is very mysterious, but on the other hand the Program Memory feature "F" came from the Atmel Mega chips, where RAM memory is very limited. Fortunately with the ESP8266 it is not so needed.
Regards, Rudolf

danbicks, 2016-02-19:
Hi Rudi,

I have had great success with this and am in the process of wrapping it all up in a library with additional functionality added. 

Can you give me a list of all the credits to include within my library. Version history amendments etc.

Big thanks again

Dans

rudi48, 2016-02-21:
Hello Dans,
The credit goes to SurferTim, please see the link on top of my listing:
http://playground.arduino.cc/Code/FTP
It was my pleasure, to do something useful for the ESP8266 community.
Thank you very much for providing the program as a library.
Regards, Rudolf

GeneM87, 2016-05-13:
Hey rudi48,

I am trying to do something similar but keep running into roadblocks. I am trying to just simply write a "hello" onto a text file on an ftp server. The Arduino with the ESP8266 chip is communicating just fine with the FTP server through the Serial Monitor AT commands, but I am having trouble applying your code. I don't have an SD card reader / writer. Would you be able to help me with developing a code that simply writes "hello" onto a text file on an ftp server? Once I can do that, I can figure out how to transmit the data from multiple sensors onto the ftp server, and access it remotely from another location, which is my main goal.

Any assistance would be greatly appreciated.

Thank you!

-Gene

rudi48, 2016-05-13:
Hello Gene,
I have build end of 2015 a temperature measurement with an ESP8266 Module, with WiFi connection
to a Raspberry Pi as an Web Server. Please have a look at 
http://www.rudiswiki.de/wiki9/WiFiLoggerOneWire
This may be an example for your project.
Also http://thingspeak.com/ could be used.
Regards, rudi48

Clemzo, 2016-06-09:
Hi,
Thank you very much for this lirary
I use esp8266FTPServer-master on my ESP-12 and it's work fine.
I need to stop or close ftp server whithout clsoe the wifi connection, how can'i do ?
Thank you.

akashboghani, 2016-06-28:
Hi, 
how can I do this while the ESP8266 is connected to the Arduino?

I need the ESP/Arduino setup to act as a client and upload a text file periodically to the ftp server.

BinarySurfer, 2016-07-08:
Hi akashboghani,

Did you solve the problem? 

I assume you have the ESP8266 module wired to an Arduino board?

I connect successfully to my FTP server & I am able to manipulate the directories etc 

I am also able to get my Dataport but I have no idea how to connect to it.

The ESP8266wifi.h seems to only allow one client connection at a time........I must be missing something very simple!

Thanks

istabraqmahmood, 2016-11-02:
the code is not working, the SPIFFS open fail !
any help, please ?

rudi48, 2016-11-03:
@stabraqmahmood 
Some important informations are missing:
1. What ESP8266 module is used?
2. What Arduino version is used?
3. What is the error message?

istabraqmahmood, 2016-11-04:
@rudi48 
the serial monitor show:
WiFi connected; IP address: 192.168.0.183
WiFi mode: WIFI_STA
Status: WL_CONNECTED
................................
SPIFFS open fail
________________________________
i`m using nodemcu v 1.0 (esp12-e) on arduino IDE v 1.6.12
please help 
thanks

rudi48, 2016-11-04:
@stabraqmahmood
It may be that your **SPIFFS** file system is not yet formatted, please see for details:
[arduino-esp8266fs-plugin](https://github.com/esp8266/arduino-esp8266fs-plugin)
you may also have a look at my wiki:
[NodeMCU](http://www.rudiswiki.de/wiki9/WiFiCar-NodeMCU#NodeMCU)

MajedRayan, 2016-11-13:
Hi rudi48
great work 
but can I use SD card instead of SPIFFS 
I replace FS.h with SD.h 
`
#include <ESP8266WiFi.h>
#include <SPI.h>
#include <SD.h>
// comment out next line to write to SD from FTP server
#define FTPWRITE

// Set these to your desired softAP credentials. They are not configurable at runtime.
const char *ssid = "SSID";
const char *password = "PASSWORD";

//boolean debug = false;  // true = more messages
boolean debug = true;
//start using SD
//static bool hasSD = false;

// LED is needed for failure signalling
const short int BUILTIN_LED2 = 16;  //GPIO16 on NodeMCU (ESP-12)

unsigned long startTime = millis();

// provide text for the WiFi status
const char *str_status[]= {
  "WL_IDLE_STATUS",
  "WL_NO_SSID_AVAIL",
  "WL_SCAN_COMPLETED",
  "WL_CONNECTED",
  "WL_CONNECT_FAILED",
  "WL_CONNECTION_LOST",
  "WL_DISCONNECTED"
};

// provide text for the WiFi mode
const char *str_mode[]= { "WIFI_OFF", "WIFI_STA", "WIFI_AP", "WIFI_AP_STA" };

// change to your server
IPAddress server(185,27,134,11);

WiFiClient client;
WiFiClient dclient;

char outBuf[128];
char outCount;

// change fileName to your file (8.3 format!)
String fileName = "test.txt";
String  path = "/test.txt";
// SD file handle 
File myFile;

void signalError() {  // loop endless with LED blinking in case of error
  while(1) {
      digitalWrite(BUILTIN_LED2, LOW);
      delay(300); // ms
      digitalWrite(BUILTIN_LED2, HIGH);
      delay(100); // ms
  }
}

//format bytes

String formatBytes(size_t bytes) {
  if (bytes < 1024) {
    return String(bytes) + "B";
  } else if (bytes < (1024 * 1024)) {
    return String(bytes / 1024.0) + "KB";
[... 328 lines cut ...]
    upload = false;
    if (doFTP(upload)) Serial.println(F("FTP OK"));
    else Serial.println(F("FTP FAIL"));
  }
  
  if (inChar == 'u') {
    upload = true;
    if (doFTP(upload)) Serial.println(F("FTP OK"));
    else Serial.println(F("FTP FAIL"));
  }  

  if (inChar == 'r') { 
    readSD();  
  }
  delay(10);  // give time to the WiFi handling in the background
}

`

I can't compile the code it give me error 
`
C:\Users\Black\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SD\src/SD.h:68:5: error: request for member 'read' in 'src', which is of non-class type 'char'

     src.read(obuf, leftLen);
`

cow77, 2016-11-14:
If in doFTP you change:
    while (dclient.connected()) {
      while (dclient.available()) {
        char c = dclient.read();
        myFile.write(c);
        if (debug) Serial.write(c);
      }
    }

to 

    while (dclient.connected()) {
      while (dclient.available()) {
        byte c = dclient.read();
        myFile.write(c);
        if (debug) Serial.write(c);
      }
    }

(char changed to byte), at least it compiles error-free.

Stangl1, 2017-07-15:
Hallo Rudi,
schönes Projekt, läuft bei mir soweit.
Nun habe ich die Sache auf meinen Webserver (ESP8266) versucht, da bekomme ich keine Verbindung.
Läuft die ganze Sache nur als Client?
Möchte von meinem Server (ESP8266) Daten auf einen WebSpace (WWW) speichern.
Wie gesagt als Client schaffe ich das mit einem Testprogamm.
Ist sowas grundsätzlich möglich?
Danke.

Stangl1, 2017-07-15:
Hi,
hier der Code der super funktioniert, aber eben nur als CLIENT.

/*
   This file modifies SurferTim's FTP code (http://playground.arduino.cc/Code/FTP)
   to work with the esp8266 chip.This only contains the functionality to upload
   a file to your server but additional functionality could easily be added by
   modifying the code starting at line 202. Functionality would be implemented using
   raw FTP commands. A list of such commands can be found at: http://www.nsftools.com/tips/RawFTP.htm

   https://github.com/esp8266/Arduino/issues/1183
   http://www.esp8266.com/viewtopic.php?f=29&t=13956

   The HTML files are dummys, they are created by my ESP8266 WiFi Server.
   This is just do test FTP-Upload and see if the external Web-Server (apache) works.
*/

//#define FORMAT                                       // To format the SPIFFS if needed
#define DEBUG                                        // Use this for DEBUG-Mode

#include <ESP8266WiFi.h>
#include <FS.h>

const char* VERSION = "Easy-Pool VX.17.07.15";

// Toggle LED_BUILTIN without delay
unsigned long LedDelta_ms = 750;                       // distance to next toggle time 750 ms
unsigned long LedNext_ms;                              // next toggle time

//Change these variables to meet your needs
const char* ssid = "3370";
const char *wpwd = "xxx";

// FTP stuff
const char* host = "v45858.1blu.de";                   // External Server, try out :-)
const char* userName = "xxx";
const char* password = "xxx";

// FTP upload Interval without delay
unsigned long ftpDelta_ms = 60000 * 2;                 // distance to next Upload
unsigned long ftpNext_ms;                              // next Upload-Time

// File Operation
String ftpFile = "";
char ftpPath[20] = "/var/www";                         // Path to the Apache-Webserver on my external Web-Space

// DO NOT CHANGE THESE
char outBuf[128];
char outCount;

extern "C" {                                           // needed to avoid link error on ram check
#include "user_interface.h"
}

#define LED_PIN LED_BUILTIN

IPAddress ip(192, 168, 1, 115);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);

WiFiClient client;
WiFiClient dclient;

void setup() {
  pinMode(LED_PIN, OUTPUT);

#ifdef DEBUG                                             // If DEBUG is defined you will see something on your Serial-Screen, this is not to save Memmory
  // Serial Setup
  Serial.begin(19200);
#endif
[... 408 lines cut ...]
      outBuf[outCount] = thisByte;
      outCount++;
      outBuf[outCount] = 0;
    }
  }
  if (respCode >= '4')  {
    efail();
    return 0;
  }
  return 1;
  yield();
}

void efail() {
  byte thisByte = 0;
  client.println(F("QUIT"));
  while (!client.available()) delay(1);
  while (client.available())  {
    thisByte = client.read();
    Serial.write(thisByte);
  }
  client.stop();
  Serial.println(F("Command disconnected"));
  yield();
}

Stangl1, 2017-07-15:
Hier der Code, funktionier leider nur als Client.

/*
   This file modifies SurferTim's FTP code (http://playground.arduino.cc/Code/FTP)
   to work with the esp8266 chip.This only contains the functionality to upload
   a file to your server but additional functionality could easily be added by
   modifying the code starting at line 202. Functionality would be implemented using
   raw FTP commands. A list of such commands can be found at: http://www.nsftools.com/tips/RawFTP.htm

   https://github.com/esp8266/Arduino/issues/1183
   http://www.esp8266.com/viewtopic.php?f=29&t=13956

   The HTML files are dummys, they are created by my ESP8266 WiFi Server.
   This is just do test FTP-Upload and see if the external Web-Server (apache) works.
*/

//#define FORMAT                                       // To format the SPIFFS if needed
#define DEBUG                                        // Use this for DEBUG-Mode

#include <ESP8266WiFi.h>
#include <FS.h>

const char* VERSION = "Easy-Pool VX.17.07.15";

// Toggle LED_BUILTIN without delay
unsigned long LedDelta_ms = 750;                       // distance to next toggle time 750 ms
unsigned long LedNext_ms;                              // next toggle time

//Change these variables to meet your needs
const char* ssid = "3370";
const char *wpwd = "xxx";

// FTP stuff
const char* host = "v45858.1blu.de";                   // External Server, try out :-)
const char* userName = "xxx";
const char* password = "xxx";

// FTP upload Interval without delay
unsigned long ftpDelta_ms = 60000 * 2;                 // distance to next Upload
unsigned long ftpNext_ms;                              // next Upload-Time

// File Operation
String ftpFile = "";
char ftpPath[20] = "/var/www";                         // Path to the Apache-Webserver on my external Web-Space

// DO NOT CHANGE THESE
char outBuf[128];
char outCount;

extern "C" {                                           // needed to avoid link error on ram check
#include "user_interface.h"
}

#define LED_PIN LED_BUILTIN

IPAddress ip(192, 168, 1, 115);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);

WiFiClient client;
WiFiClient dclient;

void setup() {
  pinMode(LED_PIN, OUTPUT);

#ifdef DEBUG                                             // If DEBUG is defined you will see something on your Serial-Screen, this is not to save Memmory
  // Serial Setup
  Serial.begin(19200);
#endif

[... 407 lines cut ...]
      outBuf[outCount] = thisByte;
      outCount++;
      outBuf[outCount] = 0;
    }
  }
  if (respCode >= '4')  {
    efail();
    return 0;
  }
  return 1;
  yield();
}

void efail() {
  byte thisByte = 0;
  client.println(F("QUIT"));
  while (!client.available()) delay(1);
  while (client.available())  {
    thisByte = client.read();
    Serial.write(thisByte);
  }
  client.stop();
  Serial.println(F("Command disconnected"));
  yield();
}

rudi48, 2017-07-16:
@Stangl1
This is a FTP client only, because a FTP server is already contained in the ESP8266 examples. Just the FTP client was missing.

Stangl1, 2017-07-16:
Hallo Rudi,

erst mal danke für die Antwort, hoffe du spricht deutsch, mein Englisch ist nicht so tolle.

FTP Client ist klar, dachte mir schon das das verwirrt.

Das Problem ist wenn mein ESP, sprich WiFI, also die Anwendung als Client läuft ist alles gut.

Der ESP soll jedoch als Webserver laufen, nicht der FTP.

Läuft der ESP als Webserver bekomme ich per FTP jedoch keine Daten raus, nur wenn der ESP als Client läuft.

Danke für mögliche Hilfe.

Stangl1, 2017-07-16:
Hallo zusammen,
hier die Lösung, einfach Server stoppen und wieder starten, manchmal ist es wirklich einfach, man muss nur drauf kommen.

Hier das Ergebnis:
v45858.1blu.de

  if (millis() >= ftpNextMeas_ms) {                        // FTP Upload to v45858.1blu.de
    ftpNextMeas_ms = millis() + ftpMeasDelta_ms;
    Serial.println(F("FTP-Upload started..."));
    
    // Stop the Server
    server.stop();
    
    ftpFile = "/index.html";
    FTPUpload(1);

    ftpFile = "/dayly.html";
    FTPUpload(2);

    ftpFile = "/monthly.html";
    FTPUpload(3);
    
    // Start the Server
    server.begin();
  }

rudi48, 2017-07-16:
@Stangl1 
Congratulations that you found a solution. I was reading yout question.

me-no-dev, 2017-07-16:
Абсолютно съм несъгласен да се поства на език различен от Английски ;) Има си Гугъл Преводач за тези неща ;)

devyte, 2017-10-19:
@rudi48 this is pretty cool. Was it ever turned into a library?

rudi48, 2017-10-19:
No, I do not have the expertise to do so.

TomElectronic, 2018-02-28:
Hello rudi or everybody who can help.
I am writing source code for ESP8266 with Kinetis MCU.
I don't use Arduino library, I am writing all low level code.

Wifi is working fine. I added FTP client functionality, it works fine (I can connect to Filezilla Server) but I can't send "data" of my ASCII file.
I understand with wireshark and your code that after STOR command, I must read new port number and send the content of my file to this port.
But I don't know how to change this port because I am already connected at port 21 (with AT+CIPSTART=\"TCP\",\"192.168.0.1\",21\r\n"). 
And visibly, when ESP8266 is connected to one port, he refused to connect to another port without disconnect before.

I supposed that I would create 2 FTP clients (as your code with "client" and "dclient") but it don't work.
I tried :
AT+CIPMUX=1\r\n;   // multiple connections allowed
OK
AT+CIPSTART=0,\"TCP\",\"192.128.0.1\",21\r\n   // create Client0 and connect to server on port 21
Link type ERROR

Do you have an idea to help me, pls ?
Regards,
Thomas.

devyte, 2018-02-28:
@TomElectronic you're in the wrong  place, you're talking about the AT firmware, but this is the Arduino firmware.

TomElectronic, 2018-02-28:
Yes, you're right, I'm sorry...

But this is the only code that I found who use 2 client with ESP8266.
And I don't understand how to do that at AT command level.

I just wrote a message on espressif forum that I found after that post.
I hope someone read this and know AT programming for this module so he can give me advices...

Thanks,
Thomas.

Skm1982, 2018-03-10:
Hello everybody. Is there a way to realize a secure ftp connection with esp8266 and this client? I mean FTPS one. I used WiFiClientSecure to open control socket and plain WiFiClient for data - it works fine. Unfortunately client can not make secure connection on data. And as far as I understand it is not 'lack of memory' problem - I tried to open secure control connections on two independent servers, worked too. Is it possible to make data connection secured too? Thanks.

topnotchrally, 2018-06-30:
I have this code working on my ESP8255. It will connect and transfer a file to my Mac but will not do so to my Raspberry Pi, with ProFTPD server. I can connect/transfer to the RPi using Filezilla.

Can someone please help. I wish to FTP from the ESP to the RPi.
Here is what I see on the ESP

WiFi connected; IP address: 192.168.2.100
WiFi mode: WIFI_STA
Status: WL_CONNECTED
Ready. Press d, u or r
SPIFFS opened
Command connected
220 ProFTPD 1.3.5 Server (Debian) [::ffff:192.168.2.148]
Send USER
331 Password required for pi
Send PASSWORD
230 User pi logged in
Send SYST
215 UNIX Type: L8
Send Type I
200 Type set to I
Send PASV
227 Entering Passive Mode (192,168,2,148,176,83).
Data port: 45139
Data connection failed
FTP FAIL

Thanks.

NightSkySK, 2019-06-24:
> I have had great success with this and am in the process of wrapping it all up in a library with additional functionality added.

Dans, Have you converted this into Library? It would help us a lot!

devyte, 2019-12-18:
Reducing 2.7.0 scope, pushing back

dplasa, 2020-05-26:
> > I have had great success with this and am in the process of wrapping it all up in a library with additional functionality added.
> 
> Dans, Have you converted this into Library? It would help us a lot!

I have worked on that code base a little bit and put together a working library that features
* LittleFS (including directory support) / SPIFFS support
* active and passive mode FTP
* supports ftp FEAT, SIZE, MSLD, MDTM commands
* works nicely with ftp command or FileZilla

Maybe you want to give this a try and check it out. See  [https://github.com/dplasa/esp8266FTPServer](https://github.com/dplasa/esp8266FTPServer)

danbicks, 2020-05-27:
Hi,

This has been a long time ago since I tinkered with this. I have attached
the class module I created for this, allows upload and download to FTP
server wrapped up in a class module.

Hope this helps.

Regards

Dans

/*
  Wrapper for FTP functionality using ESP8266
  Created by Danbicks.
  Additonal functions added for getting setting filename / paths
  Debug option for additional serial logging.

  Credits: Surfer Tim for Ino Arduino sketch Implementation 2015-12-09 
  Updates by : Rudolf Reuter
  Library development and additional functions: Danbicks 24/02/2016
*/

#ifndef DBFTP_h
#define DBFTP_h

#include "Arduino.h"
#include <WiFiClient.h>
#include "FS.h"

class DBFTP
{

   public:
      void SetFilename(String VarFile);
      void SetPathName(String VarPath);
      void SetFtpServer(String VarServer);
      void SetFtpPort(int VarPort);
      void SetFtpUser(String VarUser);
      String GetFilename();
      String GetPathName();
      String GetFtpServer();
      int GetFtpPort();
      String GetFtpUser();
      void SetDebug(bool VarDebug);
      byte doFTP(boolean upload); 
      String formatBytes(size_t bytes); 

   private:
      // private functions and alias
      WiFiClient client;
      WiFiClient dclient;
      byte eRcv();
      void efail(); 
      //String formatBytes(size_t bytes); 

      // Private Variables etc
      
      int _pin;
      boolean debug = true;  // true = more messages 
      char outBuf[128];  // was 128
      char outCount;
      // change fileName to your file (8.3 format!)
      String fileName = "WT06.txt";
      String  path = "/WT06.json";
      String FTP_Server = "ftp.dbtec.org";
      int FTP_Port = 21;
      String FTP_User = "ESP1@dbtec.org";
      String FTP_Pwd = "Electr0nics_1234";
      File fh;  // SPIFFS file handle
[... 291 lines cut ...]
     if (debug) Serial.write(thisByte); // MODDED DEBUG ADDITION
  }

  client.stop();
  if (debug) Serial.println("Command disconnected");
  fh.close();
  if (debug) Serial.println("Spiffs file closed");
}  // efail

//--------------------------------------------------------------------

String DBFTP::formatBytes(size_t bytes) 
{
  if (bytes < 1024) {
    return String(bytes) + "B";
  } else if (bytes < (1024 * 1024)) {
    return String(bytes / 1024.0) + "KB";
  } else if (bytes < (1024 * 1024 * 1024)) {
    return String(bytes / 1024.0 / 1024.0) + "MB";
  } else {
    return String(bytes / 1024.0 / 1024.0 / 1024.0) + "GB";
  }
}

#endif

dplasa, 2020-05-27:
Hi Dans,
thanks for sharing! 
Now that I've read your code, I notice my sloppy reading of this thread... wile this is all about an ftp **client**, my library provides a ftp **server** ...
:)

danbicks, 2020-05-28:
Hi Dan, no Problems. Nice idea to have FTP server running on ESP. I like
it. Probably only good though for small file sharing between a network of
ESP devices. Maybe a good idea to host FTP on a raspberry pi server for
small scale IOT stuff, Another option, would also work well alongside node
RED.
Great work buddy, nice library.
Dans

dplasa, 2020-05-28:
Hey Dans, thanks!
Well, speed is about 180kB/s FROM and 40kB/s TO the esp8266. That is not wowing but I actually just use it to put configuration files on my projects ;-) 

After reading your code, that I realized, I could include a client with almost no sweat.... So I took the liberty, to include a ftp client based on your code as well.

See [https://github.com/dplasa/FTPClientServer](https://github.com/dplasa/FTPClientServer)
Have fun!

devyte, 2020-05-29:
Nice work. Now that there's both a client and server, I suggest the following:
- Add a comment in your repo Readme stating the core version against which you tested
- Be more explicit about credits, i. e. mention danbicks explicitly in addition to linking to this thread
- Add some kind of CI or job or test to check building against the core. There are some changes coming, and some could be breaking. It would be good to have some automated test to check at least that thw code compiles. 

I have some comments about code cleanup and good practices, but we can discuss that in your repo.
To leave the comment here, I suggest factoring out the WiFiClient/Server so that they can be provided as constructor arguments. This is a precursor to supporting encryption, aka FTPS. The setup for CA etc would be done outside before passing the objects in, see examples.

Finally, I'll add a link to your repo and close this.

closed 2020-05-29 by devyte

JeffZuckerberg, 2022-01-02:
> I have worked on that code base a little bit and put together a working library that features
>
> * LittleFS (including directory support) / SPIFFS support
> * active and passive mode FTP
> * supports ftp [...]

I dplasa,

i've tried to use your library, but i got this error every time and i can't transfert  a file to FTP server :  Transfer failed after 1 ms, code: 4294967295, descr=Local file error. 
Even with the example you provide, i got the same error. 

Could you help me to manage it please ? I've attached a screenshot of the error message.

Best regards,
[image]

dplasa, 2022-01-03:
> error dplasa

Hi,

I'm not sure what happens there. Are you using the provided sample as
intended? (That is: serial monitor, use G or P with a valid filename
directly after the G/P, e.g. Ptest.txt or Gtest.txt ) ?

Is the name of the file a valid one on either file system (local or remote)?

PR #7336 (2020-05-29): Add FTP client/server lib link
Fixes #1183