esp8266/Arduino #1187: Can't connect to ESP8266 AP?

JonasWestAlro, 2015-12-10:
I am trying out this project: https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/example-sketch-ap-web-server on an ESP-07 Module.

The project opens an access point and a responses to a couple of HTTP requests. My iPhone6 connects fine to the ESP8266 access point, and everything works. Problem is that on both my android phone and my PC can detect the network, but fails to connect to it. My iPhone5 seems to connect to the network, but the server doesn't responed to any requests. 

My settings is attached in the picture.

[image]

drmpf, 2015-12-10:
try this example, 
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html
creates and access point, serves a webpage and processes the response
works for me

JonasWestAlro, 2015-12-10:
I've tried to downgrade to version 1.6.5 of the ESP8266 Arduino library - and I am now running the official example under File->Examples->ESP8266WiFi->WiFiAccessPoint.ino

Still same result. The iPhone6 can sometimes come through, but non of the other devices can.

Links2004, 2015-12-10:
have you tried using a other channel then "1"?
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L87

JonasWestAlro, 2015-12-11:
THAT WORKED! Thank you! Is it the Wifi frequency channel it sets?

Links2004, 2015-12-12:
yes, its the WiFi channel.

closed 2015-12-12 by Links2004

ZenHoop, 2016-08-03:
I have the same issue, changing the channel only seemed to help temporarily. broadcom 802.11 b/g/n on 64bit WIN 10. I phone connects flawlessly everytime. PC seems random completely resetting the all network settings from the command prompt, forgetting the network, Disable and enable adapter. Change adapter power settings, nothing seems to do the trick. Connection is hit or miss.

amishm766, 2018-07-26:
How to change channel