esp8266/Arduino #1193: ticker exception

bedenko, 2015-12-10:
I have a ticker function **test** called every 20ms.
program crashes when i call **test.detach();** with following output:
```
Exception (9):
epc1=0x40101102 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000027 depc=0x00000000

ctx: cont
sp: 3ffeff10 end: 3fff0100 offset: 01a0

>>>stack>>>
3fff00b0:  3ffe8540 3ffef0d8 3ffef0ac 4020c8b8
3fff00c0:  4020987b 3ffeed38 3ffef0ac 4020c924
3fff00d0:  4020b2c1 00000000 3ffef0d0 402071c5
3fff00e0:  3fffdc20 00000000 3ffef0d0 4020b2e9
3fff00f0:  00000000 00000000 3ffef0e0 40100114
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

If i remove **test.detach()**, the program runs forever (no crash).

igrr, 2015-12-13:
Could you please share sketch code to reproduce? Thanks.

bedenko, 2015-12-13:
Oh, sorry Igrr, i just realized, that i was writing outside my array, and with that, caused the crash.
I fixed it, and now sketch works as it should.

Sorry for false alarm.

closed 2015-12-20 by igrr