esp8266/Arduino #1194: Version 2.0 breaks implicit function declaration

tigoe, 2015-12-11:
With the latest version, implicit function declaration no longer works, making it incompatible with many existing Arduino sketches. 

Implicit function definition is a common pattern in many Arduino sketches on the net, and was an intentional choice in the original Arduino API.  Unfortunately, it no longer works in the ESP8266 Arduino port. 

In previous versions, this worked:

```
void setup() {
}

void loop() {
  myFunction();
}

void myFunction() {
}
```

Now it produces the error "myFunction was not declared in this scope".

Here's [an example](
https://github.com/tigoe/ESP8266Examples/blob/master/ESPScanNetworks/ESPScanNetworks.ino) that worked with the previous version but fails now, for testing.

igrr, 2015-12-11:
Please see https://github.com/esp8266/Arduino/issues/1066. This seems to be caused by the new arduino-builder getting confused by some c++11 stuff found in standard library headers (memory, functional).

Protoneer, 2015-12-11:
Have you tried to move the function up before the setup and loop functions?

>
> Please see #1066. This seems to be caused by the new arduino-builder
getting confused by some c++11 stuff found in standard library headers
(memory, functional).

OzQube, 2015-12-11:
I was getting the same error, and yes, moving the function up does resolve the error for me.

mangelajo, 2015-12-11:
Yeah, it's also unable to resolve functions across different .ino files in a project. @igrr, what's the workaround, how does it get confused?

igrr, 2015-12-12:
@mangelajo see some notes in https://github.com/arduino/arduino-builder/issues/68
Workaround is possible for the ESP8266WiFi library, which pulls `<vector>`. It may be replaced by a linked list.
But for the ESP8266WebServer library, i don't see a workaround. We use function objects there heavily.

marvinroger, 2015-12-13:
I think this should not even be an option, as a lot of us are using `std::bind` (well, I am). so you can't really get rid of `<functional>` (`std::function`). We have to wait for the arduino-builder fix.

devyte, 2017-10-19:
I believe this is caused by the Arduino ctags (v5.8). Not much can be done in this repo.
Closing.

closed 2017-10-19 by devyte