esp8266/Arduino #1196: DNS_SD_Arduino_OTA No response from device

dariocdj, 2015-12-11:
Hi guys, 
i'm using arduino 1.6.5 with staging esp core 1.6.5-1160-gef26c5f
Windows XP 
Firewall disabled, antivirus disabled
Python 2.7 in path
Sketch : DNS_SD_Arduino_OTA without any mod (only ssid and pass)
Esp-01 and Esp-12 tested.

Arduino IDE found port when i select Uploading Using OTA

When i try to open serial monitor it ask me for a password 

If i open a shell i can ping regularry esp on 192.168.0.211

if i open a telnet client and try to connect to 192.168.0.211 port 8266 it seems to open connection (don't refuse it)

When i try to upload  sketch: 
python.exe C:\Documents and Settings\Cdj.SERVER\Dati applicazioni\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1160-gef26c5f/tools/espota.py -i 192.168.0.211 -p 8266 -f c:\temp\build6064416799227852722.tmp/DNS_SD_Arduino_OTA.cpp.bin 

16:04:39 [ERROR]: No response from device

Thanks for helping
Dario

dariocdj, 2015-12-11:
Well.. this is the mac address :
18:fe:34:a5:db:e5 esp8266
10:bf:48:7e:ea:24 pc with arduino ide

Attached you can find wireshark log 

Thanks for helping !
Dario 

[image]

[log.pdf](https://github.com/esp8266/Arduino/files/59737/log.pdf)

krzychb, 2015-12-11:
Sketch uses 342,558 bytes (32%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 43,246 bytes (52%) of dynamic memory, leaving 38,674 bytes for local variables. Maximum is 81,920 bytes.
python.exe C:\Users\Krzysztof\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1160-gef26c5f/tools/espota.py -i 192.168.1.11 -p 8266 -f C:\Users\KRZYSZ~1\AppData\Local\Temp\build8549792214206597213.tmp/DNS_SD_Arduino_OTA.cpp.bin 
Uploading.............................................................................................................................................................................................................................................

dariocdj, 2015-12-12:
No way... tried on Win 7 64bit with esp 1.6.5 stable core and esp 1.6.5 staging core but same result : 

01:41:21 [ERROR]: No response from device

What version of phyton are you using? I'm using 2.7.10. What kind of esp ? How is it connected ? I don't have any ideas where to bang my head ! 
Thanks
Dario

dariocdj, 2015-12-12:
Also tried to erase eeprom first....

krzychb, 2015-12-12:
Hi Dario,

I can’t believe how close you are to get this working and how simple should be the reason so it is not working :smile:

Here is my python:
```
Python 2.7.10 (default, May 23 2015, 09:44:00) [MSC v.1500 64 bit (AMD64)] on win32
```
I am currently using NodeMCU boards, namely LoLin and Amica as well as ESP8266-EVB by Olimex.

[image]

All are working without any issues with DNS_SD_Arduino_OTA.ino and esp8266/Arduino release 1.6.5-1160-gef26c5f (of Sep 30, 2015).

I have been testing OTA back and forth when writing first three chapters of: https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md and was always getting repeatable results and stable operation of OTA. This is assuming I use compatible s/w on each step.

I am wondering if you see anything on serial monitor when OTA fails. If successful, the output on serial monitor looks like below:
```
Arduino OTA Test
Sketch size: 346664
Free size: 700416
IP address: 192.168.1.12
Update Start: ip:192.168.1.10, port:48266, size:346672
Update Success: 6113
Rebooting...

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0
ld

Arduino OTA Test
Sketch size: 346664
Free size: 700416
IP address: 192.168.1.12

```

If there are no clues on serial monitor and you are not moving any way forward I would put it aside for a while and try other way of doing OTA: https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#web-browser. If this fails as well maybe the problem is with your Wi-Fi network / router.

Good luck and I am looking forward to seeing how you handle this issue :+1:

Krzysztof

dariocdj, 2015-12-12:
Thank you for prompt answer, i 've forget something..... Serial monitor prompt me a password !!!

krzychb, 2015-12-12:
Yes - you mentioned it in your first post :smile: 

I assume you selected network port to do OTA.
Then you try to open Serial Monitor in Arduino.
This will not work - it is not implemented for ESP.

You need an external serial monitor.
If you do not have one I recommend http://www.compuphase.com/software_termite.htm

Krzysztof

[image]

dariocdj, 2015-12-12:
Ah ok perfect, dunno if I can try this w.e. But I want to find 10min to go in my lab and debug this!!! Thank you for helping

dariocdj, 2015-12-14:
Ok Krzysztof, with an external serial monitor this error appear : 

Arduino OTA Test
Sketch size: 346664
Free size: 90112
IP address: 192.168.0.211
Update Start: ip:192.168.0.2, port:48266, size:346672
Update Begin Error
....
thanks
Dario

igrr, 2015-12-14:
`Free size: 90112`
That's way too small to fit 346672 bytes.
If you are selecting "Generic ESP8266 module" in Board menu, you need to select larger flash size (i.e. 4M).
Of course, your board must have more than 512k flash in order for this to work...

dariocdj, 2015-12-14:
I test it immediately......thanks for prompt answer

dariocdj, 2015-12-14:
i can't believe it !!!!!! IT WORKS !!! Thanks a lot to everybody !!!!

closed 2015-12-14 by igrr

dariocdj, 2015-12-14:
Ok guys, now esp-12 work fine, but esp-01 no....

flash size is correct... i've a Manufacturer: c8 Device: 4013 ESP-01 so 4mbit....

this is serial response : 

Arduino OTA Test
Sketch size: 346664
Free size: 700416
IP address: 192.168.0.226
Update Start: ip:192.168.0.2, port:48266, size:346672

Exception (9):
epc1=0x40233a30 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000015 depc=0x00000000

ctx: sys
sp: 3ffffda0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff40:  40217e6a 3ffec4a4 00000000 40233b3c
3fffff50:  00000009 fffffffe 3ffebe10 00000000
3fffff60:  00000002 0000001a 40218815 60000600
3fffff70:  4021215d 3ffebe10 3ffec1ac 4021215d
3fffff80:  402121a2 3fffdab0 00000000 3fffdcb0
3fffff90:  3ffebe38 3fffdc20 3ffebd6c 40201795
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
   nlrnnn
b|bl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn
bbl`nlrnnn

Wtf ? 
thanks
Dario

igrr, 2015-12-14:
4 Mbit is 512 kbyte. You are telling the compiler that you have 4Mbyte flash, and the application crashes because flash is actually smaller.

dariocdj, 2015-12-14:
uhmmmmmmm so esp-12 is 32mbit ?

igrr, 2015-12-14:
Yes most of them have 32 Mbit flash.

Our "flash size" menu is in megabytes, but flash manufacturers usually list sizes in megabits. Perhaps this leads to some confusion.

dariocdj, 2015-12-14:
yes !  So i can't use OTA for esp-01 with 512k because ds_sample is bigger ? my esp-01 is a 512k(64spiffs) or 1MB(512k spiffs) ? sorry for misunderstanding....i've a Manufacturer: c8 Device: 4013 ESP-01 .....
Thanks

itsjustvenky, 2015-12-14:
The new ESP-01 (dark gray/black) are having 1MB (megabytes or 4 MegaBits). Old ESP-01 (blue) are just 512KB (half MB). 

From arduino IDE, there are four 1MB option's are available to choose from.

dariocdj, 2015-12-14:
Ok, last thing... what mean spiffs size (ok SPI Flash File System)... if i have an esp-12 and i program it with 4mb(1mb spiffs) would mean that i've 1mbyte of free space for sketch ? spiffs depend from hardware or not ? or can i program esp-12 with 4mb(3mb spiffs) without problems ? thanks

krzychb, 2015-12-14:
Hi Dario,

I told you that you were very close to resolution and the issue was simple :smile:

SPIFFS size is the flash space reserved for your files. If you reserve 1MB on 4MB flash you will have 3MB for your sketch, or 1.5MB if you are using OTA. 

Please check out flash memory layout -https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#flash-layout

There is also very desriptive picture how new sketch is copied around during OTA process between exisitng sketch and SPIFFS -https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#update-process---memory-view

Krzysztof

Links2004, 2015-12-14:
a sketch can max have 1MB, running code over 1MB is not possible.

itsjustvenky, 2015-12-14:
ESP can only map/load first 1MB from flash.. is that right @Links2004. thanks.

Links2004, 2015-12-14:
@itsjustvenky yes, the rest can be used for storing the OTA update, SPIFFS, eeprom and so one.
but for code execution we can only use the first 1MB.

itsjustvenky, 2015-12-14:
@Links2004 one more question : How the EEPROM space is allocated ? is it default with a size or it will allocated when we use in sketch "EEPROM.begin(...)". Let me know thanks.

Links2004, 2015-12-14:
in flash the eeprom is stored after the SPIFFS.
https://github.com/esp8266/Arduino/blob/master/libraries/EEPROM/EEPROM.cpp#L120
when you call begin the flash will copied to ram for easier usage. 
on commit it will be written to flash.

i think we get a bit of topic here. 
for questions you can use the chat https://gitter.im/esp8266/Arduino
or the forum http://www.esp8266.com/viewforum.php?f=25

tablatronix, 2016-01-24:
Stupid question , i am having ota issues , but my basic ota ino does not give me this debug info like sketch size etc, how to i get this for debugging? Did i not enable something.

And yes the space stuff needs to be made simpler for basic users, too much reading to understand what size and chips can support ota... If it can be handled by the ide that would be nice, and i even knnow what mbits are and its still confusing.

tablatronix, 2016-01-24:
Ok I, I didn't realize the debug flags and memory settings had to be flashed, the board menus are very confusing as to what are runtime settings and what are flashed device settings.

I had to switch to esp8266 generic to even get the debug options to show and then change them and  and change my esp 12e from 4m/3m to 4m/1m and flash serial first to get ota to work.

Hope it helps. sigh

suezz, 2016-03-16:
THANKS for OTA!!!!!!
unbelievable...:))

tablatronix, 2017-01-10:
I was having the same problem enabled debugging

```
Start
Progress: 0%:err -8
Connect Failed
Error[2]: Connect Failed
premature end: res:0, pos:0/251328
Error[4]: End Failed
ERROR[0]: No Error
```

Turns out to be port blocked, and esp could not get back to computer.
I do not know where you find actual information about how OTA works or where the documentation is, but from what I gather ide sends a UDP packet to ota port 8266 on esp, esp then fetches update from a dynamic port running in python host. No idea how to make this work through firewalls. Or where one find the docs for this mechanism or how to override listening port or use static.

igrr, 2017-01-23:
@tablatronix I don't think this part is documented, but if you run `espota.py` from the command line it will give you a help with a bunch of options. One of them is `-P host_port`, which allows setting source port on the host side. See https://github.com/esp8266/Arduino/blob/master/tools/espota.py.

Overriding this port from the IDE hasn't been implemented as the IDE doesn't make it easy to add user-configurable parameters without an exponential blowup of the number of lines in boards.txt. If you find a way to do this though, PR will be welcome.

tablatronix, 2017-01-24:
I saw that but could not figure out how to pass it from ide or platformio, can espota load an ini file or do a local override from a env var? Some external override at least for now?

HobbytronicsPK, 2017-06-30:
Hi, 
I ran into a similar kind of problem and found the solution. In my case error was as following;

`16:05:36 [ERROR]: No Answer
16:05:36 [ERROR]: No Answer`

OTA works fine if I start Arduino IDE, choose NodeMCU 1.0 from boards and select network port from port menu. Problem arises if I open Serial monitor on Arduino IDE with network port selected. It shows password screen. Right after this step, Arduino IDE starts giving the problem as mentioned above.

Choosing Serial Port from Port menu, restarting Arduino IDE and re-selecting network port resolves the problem in my case. I am using [NodeMCU development](https://hobbytronics.com.pk/product/nodemcu/) board made by amica.

tablatronix, 2017-06-30:
sounds like an arduino IDE bug, like it locks the port up, I would report it to them. Serial monitor does not work with OTA on esp also.