esp8266/Arduino #1199: Boot loader failsafe mechanism

mangelajo, 2015-12-11:
I'm theorising here, so there are chances I got something wrong. As I understood, if the boot loader copy process fails (device switched off during boot loader copy process) there's no recovery mechanism, because the copy orders are in RTC memory, and not in flash.

So, an idea here.

Why don't we save the end of the flash to set the boot loader command for upgrade?, then when an upgrade has properly finished, we erase that area.

Thoughts?

igrr, 2015-12-13:
Or rather, write a few more "finalizing" bytes to that area, indicating that the command has been processed. This would avoid having to erase that sector twice per update.

mangelajo, 2015-12-21:
@igrr, right, that sounds even better.

devyte, 2018-01-15:
While this is (probably) not a breaking change, I consider it non-trivial, so pushing back milestone.

davisonja, 2020-09-03:
This is #6538

d-a-v, 2021-03-31:
duplicate #6538

closed 2021-03-31 by d-a-v