esp8266/Arduino #1202: Not message after update OTA

Opticcenter, 2015-12-12:
Sketch code 
```
/**
 * httpUpdate.ino
 *
 *  Created on: 27.11.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

const char* ssid = "*******";
const char* password = "************";

void setup() {

    USE_SERIAL.begin(115200);
    //USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }
    WiFiMulti.addAP(ssid, password);
}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        t_httpUpdate_return ret = ESPhttpUpdate.update("http://192.168.1.39/OTA/httpUpdate.cpp.bin");
        //t_httpUpdate_return  ret = ESPhttpUpdate.update("https://server/file.bin");

        //USE_SERIAL.println(ret);

        USE_SERIAL.println("Finish Update");
        USE_SERIAL.println(ret);

        switch(ret) {
            case HTTP_UPDATE_FAILED:
                USE_SERIAL.println("HTTP_UPDATE_FAILED");
                break;

            case HTTP_UPDATE_NO_UPDATES:
                USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
                USE_SERIAL.println("HTTP_UPDATE_OK");
                break;
        }
    }
}
```

output console  serial port 

```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0
ld
```

Why not message  'Update finish' ?

Links2004, 2015-12-12:
if the update was success the module will be reboot to the new firmware.
I plan to add a option to allow to change this behavior.

Opticcenter, 2015-12-12:
How to receive messages about the status of the update?
Update OK
Not Update
Update Failed 
?

Links2004, 2015-12-12:
its only reboots when the update is ok,
if there is no update or it failed you will get the return value.

yeralin, 2015-12-12:
@Links2004 Little bit off topic. But can you take a look at this issue please? https://github.com/esp8266/Arduino/issues/1156 I've been waiting for response for 7 days already. Thank you

marvinroger, 2015-12-13:
@Links2004 yeah, adding an option to not reboot automatically would be great, I need to run some code (EEPROM-related) before rebooting. Furthermore, it is easy to do, as the `handleUpdate` already have a `reboot` parameter.

Links2004, 2015-12-13:
yes the handleUpdate reboot change where the preparation for this ;)

marvinroger, 2015-12-13:
But where to put the parameter? The `update` function is already quite-heavy in terms of parameters. After `https`?

Links2004, 2015-12-13:
i need add it on the end or it will break all existing code.
and yes it will be the ~7 parameter.

marvinroger, 2015-12-14:
Great! We will need to update the docs / examples too. Thanks btw ;)

baruch, 2016-03-25:
I added a pull request #1817  to add a callback when an update actually starts, i.e. after it was verified that a new file exists and the server says to update the code and just before downloading begins. All other stages except for the pre-reboot can be done easily by printing before and after the function call and during the update all tcp and udp connections besides the one for the update are suspended anyway so progress in the middle cannot be sent.

If restart is not done automatically you can then also send a notification after the function completed and you go on to restart yourself.