esp8266/Arduino #1203: Error in time.c using Visual Micro in VStudio 2015

gmag11, 2015-12-12:
I am getting compile errors when using Time library, that implies the use of time.c. Errors are due to missing file types definition.
```
"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/include" -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I"C:\Users\Martin_G\Documents\Arduino\CCMonitor" -I"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266" -I"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\variants\generic" -I"C:\Users\Martin_G\Documents\Arduino\libraries\ArduinoJson\src" -I"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src" -I"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src" -I"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\Ticker" -I"C:\Program Files (x86)\Arduino\libraries" -I"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries" -I"D:\C\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\Extensions\sti13djo.jxu\Micro Platforms\default\debuggers" -I"C:\Users\Martin_G\Documents\Arduino\libraries" "C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\time.c" -o "C:\Users\Martin_G\AppData\Local\V.Micro\Arduino\Builds\CCMonitor\generic\time.c.o"
time.c:27:3: error: unknown type name 'time_t
:time_t      tv_sec
time.c:28:3: error: unknown type name 'suseconds_t
:suseconds_t tv_usec
time.c:32:40: warning: 'struct tm' declared inside parameter list [enabled by default]
:extern char* sntp_asctime(const struct tm *t)
time.c:32:40: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]
time.c:33:15: error: unknown type name 'time_t
:extern struct tm* sntp_localtime(const time_t *clock)
time.c:40:1: error: unknown type name 'time_t
:static time_t s_bootTime = 0
time.c:In function 'ensureBootTimeIsSet
time.c:47:9: error: unknown type name 'time_t
:time_t now = sntp_get_current_timestamp()
time.c:At top level
time.c:78:19: error: unknown type name 'clockid_t
:int clock_gettime(clockid_t unused, struct timespec *tp)
time.c:86:1: error: unknown type name 'time_t
:time_t mktime(struct tm *t)
time.c:In function 'mktime
time.c:90:44: error: dereferencing pointer to incomplete type
:return DIFF1900TO1970 + system_mktime(t->tm_year, t->tm_mon + START_MONTH, t->tm_mday, t->tm_hour, t->tm_min, t->tm_sec)
time.c:90:56: error: dereferencing pointer to incomplete type
:return DIFF1900TO1970 + system_mktime(t->tm_year, t->tm_mon + START_MONTH, t->tm_mday, t->tm_hour, t->tm_min, t->tm_sec)
time.c:90:81: error: dereferencing pointer to incomplete type
:return DIFF1900TO1970 + system_mktime(t->tm_year, t->tm_mon + START_MONTH, t->tm_mday, t->tm_hour, t->tm_min, t->tm_sec)
time.c:90:93: error: dereferencing pointer to incomplete type
:return DIFF1900TO1970 + system_mktime(t->tm_year, t->tm_mon + START_MONTH, t->tm_mday, t->tm_hour, t->tm_min, t->tm_sec)
time.c:90:105: error: dereferencing pointer to incomplete type
:return DIFF1900TO1970 + system_mktime(t->tm_year, t->tm_mon + START_MONTH, t->tm_mday, t->tm_hour, t->tm_min, t->tm_sec)
time.c:90:116: error: dereferencing pointer to incomplete type
:return DIFF1900TO1970 + system_mktime(t->tm_year, t->tm_mon + START_MONTH, t->tm_mday, t->tm_hour, t->tm_min, t->tm_sec)
time.c:At top level
time.c:93:1: error: unknown type name 'time_t
:time_t time(time_t * t)
time.c:93:13: error: unknown type name 'time_t
:time_t time(time_t * t)
time.c:In function 'asctime
time.c:105:5: warning: passing argument 1 of 'sntp_asctime' from incompatible pointer type [enabled by default]
:return sntp_asctime(t)
time.c:expected 'const struct tm *' but argument is of type 'const struct tm *
:extern char* sntp_asctime(const struct tm *t)
time.c:At top level
time.c:108:8: error: unknown type name 'time_t
:struct tm* localtime(const time_t *clock)
time.c:113:1: error: unknown type name 'time_t
:char* ctime(const time_t *t)
Error compiling core
```
I've not been able to find how to solve this, any clue?

Regards,
Germán

Links2004, 2015-12-12:
rename the h file. 
we have a time.h in the core, and the lib has one that gives conflicts, rename the one of the lib and change the include too, and it will work.

gmag11, 2015-12-12:
I should have guessed it. Thank you.
It was not simply rename the lib as other libs did refer time.h too... but now it compiles.

closed 2015-12-13 by Links2004