esp8266/Arduino #1204: esp_iot_sdk_v1.5.0 plans?

nemeier, 2015-12-12:
Any plans to upgrade to a newer esp SDK?
I'm looking for the mesh feature :)

asetyde, 2015-12-12:
It's better to have a stable , without bugs ... Go up when base has some critical spot it's not good idea ,

Also I hope to arrive on sdk 1.5 with introduce some correction but it's better a methodical approach .

nemeier, 2015-12-12:
Is there a timeframe before the project upgrade to ESP SDK 1.5, existing IP mesh is not what I'm looking fore.
Are there any mac (addresses) to mac protocols in the ESP SDK 1.3?

igrr, 2015-12-12:
@nemeier 1.5 update will be part of the next staging release, due next week.

nemeier, 2015-12-12:
Then I'll just wait. No reason to fool around and trying to replace it myself :) I'm just going waiting for access to mesh, event just the SDK rutines, will be fine with me :) Thanks.

I'll Just close the thread.

closed 2015-12-12 by nemeier