esp8266/Arduino #1207: Cannot flash code to ESP8266

darkmarc86, 2015-12-13:
Hi guys, I'm having an issue that I was hoping to get some help with.

I can communicate with the ESP8266 via AT commands with no problem, but I cannot flash any code onto it. I have tried two different ESP8266 devices with the same result. One of them is a ESP8266-01 and the other a ESP8266-07. I believe I am following the correct procedure in placing the device into bootloader mode; connect GPIO0 to ground and then reset it by temporarily connecting RST to ground.
When I try to flash my LED blink code, I get these messages from Arduino IDE:

Warning: platform.txt from core 'ESP8266 Modules' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Sketch uses 198,840 bytes (45%) of program storage space. Maximum is 434,160 bytes.
Global variables use 33,034 bytes (40%) of dynamic memory, leaving 48,886 bytes for local variables. Maximum is 81,920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed

The settings that I am using within the Arduino IDE are the default ones for ESP8266:
Board: Generic ESP8266 Module
Flash Mode: DIO
Flash frequency: 40MHz
Upload using: Serial
CPU Frequency: 80MHz
Flash size: 512K (64 SPIFFS)
Reset Method: ck
Upload speed: 115,200
Port: COM2
Programmer: AVRISP mkII

I am using an Arduino Mega2560 as the USB-to-serial converter. I have verified that USB-to-serial converter works because I am able to successfully communicate with the ESP2866 with AT commands and even connect to my wifi network with no issues.
My hardware connections are: 
Arduino Mega 2560 to ESP8266 ESP-01:
—————————
GND — GND
TX — RX via a voltage divider that steps own 5V to 3.3V
RX — TX
ESP8266 connections:
———————–
VCC - +3.3V
GND - GND
CH_PD - +3.3V via a 10kOhm resistor
GPI0 - GND
RST - GND

Somebody mentioned trying to connect TX to TX and RX to RX which I have tried. The warnings at the start of the code compilation worry me a bit. Perhaps there is some software problem with the Ardino IDE? Within the ‘Boards manager’ I have installed ‘esp8266 by ESP8266 community version 2.0.0’.
If anybody could provide any assistance I would be very grateful. I’ve spent about four days trying to get these chips to work with quite a bit of frustration. Thanks in advance.

penfold42, 2015-12-13:
You should have Pull-up resistors on reset and gpio0 as well.

Have you seen the schematics in the documentation ?

darkmarc86, 2015-12-13:
Hi Penfold. I just tried it again with the additional pull-up resistors and it didn't change anything. The fact that I'm successfully communicating over serial with AT commands suggests to me that my wiring is correct. 

The schematics I'm following are these ones:
http://esp8266.github.io/Arduino/versions/2.0.0/doc/boards.html#nodemcu-0-9

penfold42, 2015-12-13:
Are you running a sketch on the arduino as a serial gateway/repeater or abusing it for the USB to serial chip ?

darkmarc86, 2015-12-13:
I'm running the Arduino purely as a USB-to-serial converter, but I'm using the Arduino software (IDE) to transfer my 'LED blink' sketch over to the ESP8266. The 'LED blink' code is based on this page:

http://www.arduinesp.com/blink-manual-flash

danielmawhirter, 2015-12-13:
There are two ways I can think of to use an arduino as usb-serial for this purpose. One is to read bytes form one serial port and output them on another. This introduced too much delay to work for me, but YMMV. The other (which the wording suggests you are trying) is to bypass the arduino chip and just use the TX and RX pins which connect to the serial converter. I would expect this to work under the AT command condition where no bootloader commands are in play, but to fail under the programming condition because the arduino will be in its bootloader as well and talking over serial.
TL;DR You might want to get a USB-UART board to program ESPs.

darkmarc86, 2015-12-13:
Hi Daniel. When I am flashing the ESP8266, within the Arduino IDE I have set the board manager to 'Generic ESP' so my understanding is that the code doesn't get flashed onto the Arduino chip, but rather bypasses the Arduino and goes to the ESP chip.

danielmawhirter, 2015-12-14:
The "board" setting is used by the compiler and then the serial communications open to the COM serial port. That port is connected to both the arduino and the ESP (assuming the wiring is right). The signaling pins on the USB-serial chip on the arduino are permanently connected to the arduino chip. This means that when the serial communication opens, both the arduino and ESP will receive on their rx pins and both will try to reply on their tx pins. The arduino's communication cannot be suppressed, so this causes the failed sync. This will not be a feasible method to flash the ESP.

darkmarc86, 2015-12-14:
Thanks Daniel! That explanation makes sense to me. I've just ordered a FT232RL USB-to-serial converter off ebay. Hoping it arrives before Christmas!

penfold42, 2015-12-14:
If you keep just the atmega2560 in reset you should be able to avoid that conflict.
Check the schematic of the arduino mega - it might have pull-ups on rx/tx to 5v which the ESP won't like

darkmarc86, 2015-12-14:
I found on the Arduino website for the mega2560 it says:

Reset pin: Bring this line LOW to reset the microcontroller. Typically used to add a reset button to shields which block the one on the board.

So you're saying that I should permanently drive this pin to ground while programming the software onto the ESP8266?

darkmarc86, 2015-12-14:
Nope, tried grounding the RESET pin on the Arduino while flashing the ESP but I still get the errors.

danielmawhirter, 2015-12-14:
I'll offer the software version if you want to try it. Connect the ESP to one of the other hardware serial ports on the Mega. Load the following sketch on the mega:
```
void setup() {
  Serial.begin(11520); //tools->upload speed
  Serial1.begin(115200); //connected to ESP
}
void loop() {
  if(Serial.available() > 0) { //computer sent something
    Serial1.write(Serial.read()); //pass along to ESP
  }
  if(Serial1.available() > 0) { //ESP sent something
    Serial.write(Serial1.read()); //pass along to computer
  }
}
```
It forwards bytes between the endpoints. SoftwareSerial was too slow for me, but since the mega has multiple hardware serial, maybe it has a chance.

itsjustvenky, 2015-12-15:
Here is an example (not exactly but related). Uploading sketch to Pro Mini using Uno board.

http://www.instructables.com/id/Uploading-sketch-to-Arduino-Pro-Mini-using-Arduino/?ALLSTEPS

lnxbil, 2016-01-26:
I run into the same problem and it took my some time to understand how to put it into firmware mode:

0. Wire correctly and make sure that AT command stuff works, so communication is OK
1. Open serial monitor, 74880 baud
2. GPIO0 to ground
3. reset to ground
4. reset to VCC
5. serial monitor should show ``` ets Jan  8 2013,rst cause:1, boot mode:(1,6)```
6. unplug GPIO0
7. serial monitor should still show that line from 6, nothing more - if not, return to 3
8. now your device is in the correct mode

and flashing onto should work - at least it did the trick for me.

riskorja, 2016-02-25:
Hi
with new FTDI driver message "NON GENUINE DEVICE FOUND!"
I was able to revert to the CDM v2.08.30 Driver for the next connect/disconnect.

To build  "Arduino Mega2560 as the USB-to-serial converter" 
The first step was to upload simple code found on Arduino forum to Arduino Mega2560

```
void setup ()
{
  
  Serial.begin (115200); 
 
  Serial1.begin (115200);
 
} // setup

void loop ()
{
  
  while ( Serial1.available () )
  { 
    Serial.write ( Serial1.read () );   
  }
   
  while ( Serial.available () )         
  {
    Serial1.write ( Serial.read () );  
  }
  
} // loop
```

My hardware connections are: 
Arduino Mega 2560 to ESP8266 ESP-12E:
—————————
GND — GND
TX1 — RX via a voltage divider that steps own 5V to 3.3V
RX1 — TX
After that I was able to Serial Monitor on USBx

To flash ESP-12E  
I changed board type to 
"Node MCU 1.0 (ESP-12E...)"; Serial Upload ; Arduino USBx port
and connected ESP-12E to Arduino Mega2560 with folowing
———————–
VCC - +3.3V
GND - GND
CH_PD - +3.3V
GPI0 - GND 
RST - RESET
GPIO15 - GND

All other GPIO disconnected (DHT22 on GPIO2 did not help :)

This should work until new CH340G makes his journey.

closed 2016-02-29 by igrr

w3hacker, 2016-10-16:
@darkmarc86 hahaha.We need a usb to ttl.

dobro7, 2017-06-25:
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

I am getting this after I followed lnxbil advice. I am trying to upload a sketch from Arduino IDE -> FTD1272 ->ESP8266 01. I was successful getting it to take the simple Blink example, but now I can't get it to upload.

Any ideas? What information is needed from me?

elnaz-t, 2019-01-29:
Hello every one
I had the same   error espcom_upload_ failed   with my esp01 when i was programming it by arduino uno and I searched  very much,  every one said something,  even one said that there is not a certain way,  
BUT I FOUND IT NOW!!! :)
When i was reading above comments i saw
danielmawhirter  comment  that said RX  and TX   in arduino and esp  are working togather and i thoutgh may be i should first upload   Bareminimum   code on arduino and then program my esp01,  i do that and  it was correct! 
 I think that this is the solusion for this error,  please try it.
Connections :
                Esp                    Arduino
                 Vcc                   Vcc  3. 3v
              Gnd                      Gnd
              CH-PD                  Vcc  3. 3v
               GPIO0                  Gnd
                Rx                        Rx»0
                Tx                        Tx»1

Thanks ; )