esp8266/Arduino #1209: mqtt lib causes panic ..\core_esp8266_main.cpp:85 __yield

Suxsem, 2015-12-13:
This problem is related to: https://github.com/esp8266/Arduino/issues/1161

Hi all. I tried pubsubclient library (multiple forks of it) but none of them was reliable.
I finally found this awesome mqtt library (https://github.com/i-n-g-o/esp-mqtt-arduino) based on (https://github.com/tuanpmt/esp_mqtt).

The problem is that the library relies on "system_os_post" and other mechanisms that fires yield() outside of the setup and loop functions(), causing my esp8266 to reboot.

Now, i just tried to comment the panic() call inside C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp at line 85.

NOW THE LIBRARY WORKS GOOD!

So the question is: it's dangerous to remove the panic() call inside core_esp8266_main.cpp?
Why it's there?

asetyde, 2015-12-13:
Panic causes reboot of esp ?

Suxsem, 2015-12-13:
I don't remember if it reboots or just freezes. Anyway when it panics i get the stack trace and the arduino sketch stops.
Removing panic() from core_esp8266_main.cpp at line 85 fixed the problem but i'm worried about the consequences... do you know why the panic() it's there?

asetyde, 2015-12-13:
No , but it's two month that I ve panic -> freeze issue , my device blocks after 3h or 2 days or random and not reboot .. 

It's terrible bug .. I use MySQL and dht library but from oracle some people say to me that Arduino on tcp when there are packets error device can block

Links2004, 2015-12-13:
the panic function will dump the stack and rebootes then, it will not freeze the ESP.

@Suxsem when you remove the panic, for yield it may work in some cases but its there for a reason.
if you remove it you can run functions from the run task or from IRQ, and this can crash any think at any time.
Its the same problem that you have when you designe a program for multi core systems.

chaeplin, 2015-12-13:
I had a lot of reboot with knolleary's library.
I had commted out Serial.begin(), no reboot again.

Suxsem, 2015-12-13:
knolleary's library doesn't implement qos 2 and other features. https://github.com/i-n-g-o/esp-mqtt-arduino it's way more reliable (i'm sending messages at 10Hz without problems).

But it needs panic() to be commented in core_esp8266_main.cpp and I don't know if it's safe to do

DKfraDK, 2015-12-13:
I do realize that this is not a solution to the problem, however I have been using https://github.com/Imroy/pubsubclient for quite a while now, and it is very stable!

Suxsem, 2015-12-13:
I already tried it... It's not stable if you need to receive more then 10 messages in a second

Suxsem, 2015-12-13:
@Links2004 I missed your comment! 
I'm understanding that yield must be fired from the main thread, but why panic instead of just skip the actual yield?

Links2004, 2015-12-13:
when a yield is inside the code to 99% also other calles are there that work across different task.
all this task sends a message back to the "arduino" task (where setup and loop runs).
when you use the functions i other tasks the events go to the arduino task and the task is doing what ever and now gets an unexpected event, at this point its only luck what will be the result.
since the arduino task can be at any position of the code.

igrr, 2015-12-13:
@Suxsem 
The reason to have an assertion there is to make programming errors more obvious, so that they cause guaranteed failure instead of hard-to-catch undefined behaviour.

For instance, someone might be calling a function `foo` from a library `Bar`. This function, in turn, calls into a blocking API, like `Serial.println`. You may try calling `foo` from a non-Arduino context, and that might cause undefined behavior which will cause sporadic exceptions or even race conditions.
 Making the program fail hard and fast gives an obvious clue that program is doing something wrong.

In your particular case, the issue is either that the library itself calls blocking functions from a wrong task, or you call some blocking functions from publish/subscribe callbacks. In the first case, library has to be fixed. In the second case, you need to move blocking calls outside of your publish/subscribe callbacks into Arduino `setup`/`loop`.

closed 2015-12-13 by igrr

Suxsem, 2015-12-13:
I agree with you, now it's more clear, thanks! Can be closed