esp8266/Arduino #1215: IDE doesn't find Network ports (Win)

thegimliboy, 2015-12-14:
I uploaded BasicOTA to my ESP and everything worked perfectly... On my Mac (it did showed the Network ports in IDE)... But with my Win10 PC (tried bonjour 2 and 3) AND my Win7 laptop (again bonjour 2 and 3) it doesn't list/find any Network ports... But a mDNS Browser (Win) does list my ESP... I don't know what to do... (I also tried starting Arduino IDE with admin permissions...)
I know it *should* be working at least on Win7 because of this screenshot: https://github.com/esp8266/Arduino/raw/master/doc/ota_updates/ota-ide-ota-upload-configuration.png

Heeeelp... (Sorry for my bad english)

krzychb, 2015-12-14:
Hi thegimliboy - to load BasicOTA using OTA you need  Arduino-PR-4107-BUILD-421. Is it what you have installed on your Win10 PC?
Krzysztof

thegimliboy, 2015-12-14:
Hi,
Is 1.6.5-r5 a vaild answer? It's the same version than on my Mac and on my
Windows 7 Laptop... Otherwise I don't know which kind of version I got...

krzychb, 2015-12-14:
To make specific OTA examples work you need specific version of Arduino IDE and board manager listed under https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#application-example

In you case, If you are have Arduino release 1.6.5-r5 then you need to use 1.6.5-947-g39819f0 (of July 23, 2015) or 1.6.5-1160-gef26c5f (of Sep 30, 2015) version of platform package together with DNS_SD_Arduino_OTA.ino example.

On the other hand, to make BasicOTA.ino example work, you need to follow specific version of s/w and steps listed under https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#arduinoota

Good luck!

Krzysztof

thegimliboy, 2015-12-15:
Thank you very much :)

So basicly it was just taking this IDE (https://github.com/esp8266/Arduino/pull/984#issuecomment-155905800) and installing the lastest Libary from the Git... Thank you :)

thegimliboy

closed 2015-12-15 by thegimliboy

krzychb, 2015-12-16:
Thank you for update :smile:
I am really glad you made it work!

If you like this functionality, give thumbs up to our campaign to get it merged into Arduino IDE - https://github.com/arduino/Arduino/pull/4107#issuecomment-165015782
:+1:

thegimliboy, 2015-12-16:
Yey :D They finally merged it :smile:

OTA is just awesome...