esp8266/Arduino #1216: espcomm_open failed

thoskins7, 2015-12-15:
Objective is to use ESP8266 to send data collected by Arduino Uno. In Arduino's Serial monitor I am able to send AT Commands with no issues for weeks, but everytime I try using Serial commands from the IDE I get this communication error with the ESP. My ability to send the AT commands flawlessly in the serial monitor would indicate to me that I have the necessary 3.3v power, proper pull-ups, pins, etc in place.

Arduino is connected to computer via USB. ESP01 is connected to Arduino via either pins 0,1 or 2,3.  Can anyone help with suggestions? Am I trying to implement this in the wrong way? Can I not use the Arduino to gather data and then use the ESP01 to send the data out to either a TCP or UDP server?

itsjustvenky, 2015-12-15:
See this example project : http://electronut.in/IoT-temp-sensor/

drmpf, 2015-12-15:
You can also check out my Cheap WiFi shield for Arduino,  connects via pins 0,1  sends/receives serial via wifi.  http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html

closed 2016-02-29 by igrr