esp8266/Arduino #1217: Exception with HTTPClient and some web servers

vlast3k, 2015-12-15:
I noticed a strange exception thrown when using HTTPClient to invoke some servers (thingspeak talkback api).
If i call it in the loop() method - after 1-2 calls an exception is thrown and the esp rebooted. If i call it from setup() there is no such issue. Also when calling other servers (tried ping.eu) it works in any situation.
It does not look like an issue with the heap, as it is enough.

The source of the example is here : https://gist.github.com/vlast3k/7332803c2508c0d4db55
I am using the latest GIT version and a nodemcu 0.9.
I have not yet tried it with using a plain socket call - this is the next thing to do

The exception is:
```
Exception (9):
epc1=0x401012fb epc2=0x00000000 epc3=0x00000000 excvaddr=0x40203c4d depc=0x00000000

ctx: cont 
sp: 3fff0ea0 end: 3fff10b0 offset: 01a0

>>>stack>>>
3fff1040:  00000000 3fff2398 0000000f 00000000  
3fff1050:  40106cac 3fff1100 00000011 00000000  
3fff1060:  00000000 000000c8 ffffffff 402055d0  
3fff1070:  00000000 00000000 3fff1100 40202234  
3fff1080:  40203c59 00000000 00000000 00000000  
3fff1090:  00000000 00000000 3fff0080 40203c84  
3fff10a0:  00000000 00000000 3fff0090 40100114  
<<<stack<<<
```

Links2004, 2015-12-15:
the null ptr comes from the fact that you define the return of ```getTalkback()``` as String but you never return anything, when the ESP starts to cleaning its ram there is a null ptr and then you get the Exception.

vlast3k, 2015-12-15:
Aaaahhh, so stupid of me...
Sorry!
Too many time spent programming in Java, and seems like i have to get used to watch closely  what happens

closed 2015-12-15 by vlast3k