esp8266/Arduino #1221: Change ESP8266 MAC address

ikin77, 2015-12-15:
Hi. I started using ESP8266 through AT commands that were sent from an Arduino. With Espressif firmware it's possible to change MAC address (SoftAP & Client) (AT+CIPSTAMAC AT+CIPAPMAC).

I can't find a way to do the same thing with ESP8266/Arduino libraries. Is that possible?

Thanks in advance and best regards,

  Francisco

torntrousers, 2015-12-15:
I asked about that the other day on [Gitter 12/04 16:29](https://gitter.im/esp8266/Arduino/archives/2015/12/04) and was given:
```
void initVariant() {
  uint8_t mac[] = {0x77, 0x01, 0x02, 0x03, 0x04, 0x05};
  wifi_set_macaddr(STATION_IF, &mac[0]);
}
```

ikin77, 2015-12-15:
@torntrousers 
Thanks!!! Did you try it? That it works??

ikin77, 2015-12-17:
I close the issue. The solution works :)

closed 2015-12-17 by ikin77

gojimmypi, 2016-11-24:
I was not able to change my MAC address, at least not the value returned by WiFi.macAddress.

Curiously I could not: `#include <user_interface.h>
`
instead I needed:

```
extern "C" {
#include "user_interface.h"
}
```

```

String myMacAddressString() {
	String res;
	char myMac[18] = { 0 };
	for (int i = 0; i < 6; i++) {
		sprintf(&myMac[i * 3], "%02X:", WiFi.macAddress()[i]); // format each number as 2 digit hex, followed by colon.  "12:"
	}
	res = (String)myMac;
	res.toLowerCase();
	return res.substring(0,res.length() - 1); // minus 1 in order to not return the trailing ":"
}

	Serial.print("my MAC = ");
	Serial.println(myMacAddressString());

	Serial.println("Changing MAC...");
	uint8_t  mac[] = { 0x77, 0x01, 0x02, 0x03, 0x04, 0x05 };
	wifi_set_macaddr(STATION_IF, &mac[0]);
	Serial.println("MAC Change Done!");

	Serial.print("my MAC = ");
	Serial.println(myMacAddressString());
```

(my MAC is the same both before and after the call to wifi_set_macaddr)

See also: WiFi.macAddress;  `#include <ESP8266WiFi.h>`

```
	Serial.print("MAC Address:");
	Serial.printf("%02x", WiFi.macAddress()[0]);
	Serial.print(":");
	Serial.printf("%02x", WiFi.macAddress()[1]);
	Serial.print(":");
	Serial.printf("%02x", WiFi.macAddress()[2]);
	Serial.print(":");
	Serial.printf("%02x", WiFi.macAddress()[3]);
	Serial.print(":");
	Serial.printf("%02x", WiFi.macAddress()[4]);
	Serial.print(":");
	Serial.printf("%02x", WiFi.macAddress()[5]);
	Serial.println();

``` 

ESP8266WiFi Built-In by Ivan Grokotlov, Version 1.0.0.0 installed in Arduino IDE, but compiled with Visual Studio 2015 with Visual Micro.