esp8266/Arduino #1225: unable to POST

Apple456, 2015-12-15:
I am trying to post on mailGun or PushBullet to add notifications on my project but all the time i invoke those functions the ESP8266-01 resets, I believe is a memory issue but I'm not able to see what is wrong with this code, can anyone help?
thanks

this is for Push Bullet 
void pushNotification(){
  Serial.println("- pushNotification");
  HTTPClient client;
  client.begin("api.pushbullet.com", 443 ,"/v2/pushes", true, "2C BC 06 10 0A E0 6E B0 9E 60 E5 96 BA 72 C5 63 93 23 54 B3");
  client.addHeader("Content-Type", "application/json");
  client.addHeader("Access-Token","XXXXXXXXXXXXXXXXXXXXXXX");
  client.POST("{'body':'Alarm triggered','title':'Alarm!','type':'note'}");
  client.writeToStream(&Serial);
  String payload = client.getString();
  Serial.println("push notification payload ");
  Serial.println(payload);

  //TODO manage errors
  client.end();
  server.send(200, "text/html",payload); // server is ESP8266WebServer server(80);
}
and this for mailGun

void handleAlarmMail(){
  Serial.print("- handleAlarmMail");
  HTTPClient client;
  client.begin("api.mailgun.net", 443 ,"/v3/mg.bluemanifest.com/messages", true, "39 B5 14 B8 3D 88 28 86 28 F3 5A F8 22 8B 48 DD 02 EC 62 EC");
  client.addHeader("Content-Type", "application/x-www-form-urlencoded");
  client.addHeader("Authorization","Basic XXXXXXXXXXXXXXXXXXXX");
  client.POST("from=email@email.com&to=email@email.com&text=Something text in here not too long &subject=Alarm%21%21%21");
  client.writeToStream(&Serial);
  String payload = client.getString();
  Serial.println("payload ");
  Serial.println(payload);

  //TODO manage errors
  client.end();
  server.send(200, "text/html",payload); // server is ESP8266WebServer server(80);
}

Links2004, 2015-12-15:
you mix the usage of ```client.getString();``` and ``` client.writeToStream(&Serial);``` that will not work.
a other problem can be the free ram. SSL / HTTPS needs many ram.

Apple456, 2015-12-16:
how should I get the response after using writeToStream?

Links2004, 2015-12-16:
you can not use both.
```writeToStream``` will read all data from server and write it to serail in this case.
```getString``` try to get all data again, but there is no data left, that is problematical.
you can use ```getString``` or ```writeToStream``` but not both at the same time.
since you do ```Serial.println(payload);``` I recommend to remove ```writeToStream``` in your case.

note: ```getString```  is desigend for short response (need to fit in ram)

Apple456, 2015-12-16:
Thank you for your quick answer, I try to comment out first getString() then writeToStrem so using one or the other but i still have the same issue! the response itself isn't that big... 

- pushNotification

Panic .../Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp:404 ax_port_realloc

ctx: cont 
sp: 3fff23b0 end: 3fff2900 offset: 01b0

>>>stack>>>
3fff2560:  40106cc6 3fff4e40 00000000 3fff680d  
3fff2570:  36363636 3fff62f8 3fff6304 4020811d  
3fff2580:  3fff680d 00000000 3fff4e40 4021f9b2  
3fff2590:  3fff680d 00000000 3fff4e40 40220d92  
3fff25a0:  00000001 00000001 3fff6800 4021ef46  
3fff25b0:  0a3c5f4e 46eec10b be8b8d75 000000e2  
3fff25c0:  00000010 3fff680d 3fff4eec 3fff2600  
3fff25d0:  3fff680d 3fff4d48 000000b9 00000010  
3fff25e0:  3fff4e40 3fff680d 00000010 00000010  
3fff25f0:  3fff680d 00000000 3fff4e40 402207e2  
3fff2600:  d88dba7a 1e5b234c 0156ccd9 acf2089a  
3fff2610:  6f405262 00000000 00000038 4021ff00  
3fff2620:  00020016 00000014 00000010 00000010  
3fff2630:  00000033 00000016 3fff4e40 00000000  
3fff2640:  00000000 3fff4e40 3fff4e40 40220916  
3fff2650:  00000000 3fff67fd 3fff4e40 40220add  
3fff2660:  00000000 3fff4b58 3fff4e40 40220b64  
3fff2670:  3fff4838 4020b0b2 3fff4af0 3fff177c  
3fff2680:  3fff1780 3fff4af0 3fff4848 40207ed6  
3fff2690:  41f70e48 3fff4888 4020b1ac 3fff4888  
3fff26a0:  00000000 000001bb 3fff4af0 40208186  
3fff26b0:  3ffe9cc0 41f70e48 3fff273c 00000001  
3fff26c0:  40106cac 000001bb 3fff4af0 402074fd  
3fff26d0:  3ffe9cc0 41f70e48 3ffe9cc0 41f70e48  
3fff26e0:  3fff4a70 00000001 3fff27c4 4020d2fe  
3fff26f0:  3fff4ae0 00000000 3fff2790 4020819c  
3fff2700:  40106cac 3fff27fc 3fff2790 402087f8  
3fff2710:  00000001 00000039 3fff2790 40208afb  
3fff2720:  3fff4878 40106c9a 3fff27f0 00000039  
3fff2730:  3fff4b10 0000000f 00000004 4020d2a6  
3fff2740:  3ffe974f 00000039 3fff27f0 4020d2fe  
3fff2750:  0000000a 00000000 3fff27f0 00000001  
3fff2760:  00000001 00000001 3fff18a4 40208bbc  
3fff2770:  3ffe928e 4021c8a0 3fff2790 40208bd5  
3fff2780:  00000001 00000001 3fff18a4 40206bee  
3fff2790:  3fff4af0 3fff4af0 3fff4a70 0000001f  
3fff27a0:  00000012 000001bb 3fff4868 0000000f  
3fff27b0:  0000000a 00000001 3fff4aa0 0000003f  
3fff27c0:  0000003b 3fff4bd0 0000005f 00000050  
3fff27d0:  3fff4a40 0000001f 00000011 00000000  
3fff27e0:  00000000 00000000 ffffffff 4020d200  
3fff27f0:  3fff4888 0000003f 00000039 3fff4848  
3fff2800:  0000000f 0000000c 3fff2840 4020d32e  
3fff2810:  40106cac 3fff4758 3fff4758 4020c0a6  
3fff2820:  0000000f 3fffc6fc 3fff4758 402093fe  
3fff2830:  40209408 3fff4758 3fff2860 4020943a  
3fff2840:  3fff4848 0000000f 00000005 4020d454  
3fff2850:  40209408 3fff4758 3fff0e40 4020959d  
3fff2860:  3fff4828 0000000f 00000005 402077b3  
3fff2870:  3fff0e7c 401011f3 3fff18e0 00000001  
3fff2880:  00000000 0000000f 4020b1ac 3fff18e0  
3fff2890:  402012e6 00000001 00000001 3fff18d8  
3fff28a0:  3fffdc20 000003e3 3fff0e40 40209698  
3fff28b0:  3ffe9580 00000000 000003e8 0000a325  
3fff28c0:  3fff0e60 3fff4a08 00000016 40101941  
3fff28d0:  4020b1cd 00000000 3fff18d1 40207039  
3fff28e0:  3fffdc20 00000000 3fff18d1 4020b1f5  
3fff28f0:  00000000 00000000 3fff18e0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

this is the response from mailGun for example...

{
  "id": "<20151216185254.2434.44842@mg.domain.com>",
  "message": "Queued. Thank you."
}

Links2004, 2015-12-16:
your heap/ram is to less ```ax_port_realloc``` is a clear indicator. 
SSL needs 22KB heap.
check if you can optimize the heap usage of you sketch.
```ESP.getFreeHeap();``` will show you the free heap.

igrr, 2016-02-29:
2.1.0 includes a workaround for this issue.

closed 2016-02-29 by igrr

rpiontik, 2016-06-23:
Hi! 

I had the same problem. But it is not leak problem. I had ~26kb free RAM. Server was sending empty body that generated the error. If server sended any data, I did not have the problem.

Roman.