esp8266/Arduino #1228: Max HTTP data size is only 256bytes?

JonasWestAlro, 2015-12-16:
Hey all, 

I am sending a HTTP POST request to my ESP-module. The POST request has its header and then 1024 bytes of data afterwards. Problem is that I only receive 256 of the data bytes in the application? ie. client.available() stops being >0 when I've received 256 bytes of post data (not the header - that I always receive whole).

Is there some kind of data limitation in the arduino port? (When using AT commands the buffers are upto 2048bytes).

Links2004, 2015-12-16:
the only limitation is the free ram to store the data.
but without log and code its hard to say where the problem is.

JonasWestAlro, 2015-12-17:
Sorry, I found a newbie mistake of my own. How can I delete this issue? :)

closed 2015-12-17 by igrr