esp8266/Arduino #1233: Device blocked on ERROR_PACKET

asetyde, 2015-12-17:
I use mysql library , i see a block of device when i reboot server mysql and device is connected .. system freezer .. on library i not see particular trace of error .. i use stable 1.6.5 , with 2.0.0 many problem ... 
How can i do ??

[image]

about library functions :  

[image]

[image]

[image]

asetyde, 2015-12-22:
@igrr @Links2004 Can you help me ? my code it blocks randomly , how tools can i use to debug ? serial not help

serial trace :

+++++++++++++++++

Opening port
Port open
EEPROM READ
System Error 
7
EEPROM  INIT  -wait wdt message and please turn off power and turn on

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
Port open
.......................EEPROM  WRITE 
ok
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
EEPROM READ
ok check integrity 
System start...loading wifi connection
..................
Start
 - SYSn : forecastq
 - SYSnapp : thermostat
 - SYSgroup : base
 - SYSv : 1.1
 - STRUCT SIZE : 352
 - FREE HEAP : 21696
 - CPU : 80
 - Wifi ip : 192.168.1.107
 - Wifi Mac : 18:fe:34:fa:3a:97
--------------------------------------
Start iamaliveSQLforecast!
Connected to server version 5.6.25-4
Disconnected.
End iamaliveSQLforecast!
--------------------------------------
--------------------------------------
Start putalldataSQLforecast!
Connected to server version 5.6.25-4
Disconnected.
end putalldataSQLforecast!
--------------------------------------
--------------------------------------
Start putdataSQLsensorstatistics!
Connected to server version 5.6.25-4
Disconnected.
End putdataSQLsensorstatistics!
--------------------------------------
21024
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
6.74
6
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
21216
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
6.34
6
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
**************************
Temperature : 
21.80
Humidity : 
34.80
HexIndex : 
24.81
**************************
--------------------------------------
Start getalldataSQLforecast!
Connected to server version 5.6.25-4
Haschange: 0
Needreset :0
None done
Disconnected.
end getalldataSQLforecast!
--------------------------------------
21272
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
6.16
6
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
21272
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
6.55
6
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
**************************
Temperature : 
21.80
Humidity : 
35.20
HexIndex : 
24.82
**************************
21272
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
6.83
6
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
21496
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
6.26
6
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
21496
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
5.31
5
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
**************************
Temperature : 
21.80
Humidity : 
35.20
HexIndex : 
24.82
**************************
21496
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
5.41
5
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
21496
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
5.43
5
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
**************************
Temperature : 
21.80
Humidity : 
35.10
HexIndex : 
24.82
**************************
21496
~~~~~~~~~~~~~~~~~~~~~~~~~
Impedance real read : 
6.85
6
Light conversion LUMEN : 
60
~~~~~~~~~~~~~~~~~~~~~~~~~
--------------------------------------
Start getalldataSQLforecast!

code ti freezes

++++++++++++

asetyde, 2015-12-22:
I modified some line to work .. but something not work .. random freeze of esp

igrr, 2015-12-22:
I've taken a look at Mysql library.
There is a line there which is a good candidate for this issue (mysql.cpp:611, `Connector::read_packet`):
```c++
  // Read packet header
  for (int i = 0; i < 4; i++) {
#if defined WIFI
    while (!client.available());
#endif
    local[i] = client.read();
  }
```
If something happens to the connection (i.e. it drops due to a network issue), this `while (!client.available());` will be executed endlessly. There are a few other similar loops in the code as well.

I suggest to use Stream::readBytes function instead, because it knows how to handle timeouts. Either do the necessary changes yourself, or contact library author and suggest the fix. 
Note that `wait_for_client` method does have timeout handling (similar to Stream::readBytes).

asetyde, 2015-12-22:
Many many thx Igrr I'm waiting library author because i not ve knowledge to improve or correct it :   

i exclude wifi control :  

void Connector::read_packet() {
  byte local[4];

  if (buffer != NULL)
    free(buffer);

//#if not defined WIFI
  // Wait for client (the server) to send data
  wait_for_client();
//#endif

  int avail_bytes = wait_for_client();
  while (avail_bytes < 4) {
    avail_bytes = wait_for_client();
  }

  // Read packet header
  for (int i = 0; i < 4; i++) {
#if defined WIFI
    //while (!client.available());
#endif
	  local[i] = client.read();
  }

  // Get packet length
  packet_len = local[0];
  packet_len += (local[1] << 8);
  packet_len += ((uint32_t)local[2] << 16);
#if not defined WIFI
  // We must wait for slow arriving packets for Ethernet shields only.
  avail_bytes = wait_for_client();
  while (avail_bytes < packet_len) {
    avail_bytes = wait_for_client();
  }
#endif
  // Check for valid packet.
  if (packet_len < 0) {
    print_message(PACKET_ERROR, true);
    packet_len = 0;
  }
  buffer = (byte *)malloc(packet_len+4);
  if (buffer == NULL) {
    print_message(MEMORY_ERROR, true);
    return;
  }
  for (int i = 0; i < 4; i++)
    buffer[i] = local[i];

  for (int i = 4; i < packet_len+4; i++) {
//#if defined WIFI
//    while (!client.available());
//#endif
    buffer[i] = client.read();
  }
}

asetyde, 2015-12-22:
but on wait_for_client i'dont know i can do or to do : 

int Connector::wait_for_client() {
  int num = 0;
  int timeout = 0;
  do {
    num = client.available();
    timeout++;
    if (num < MIN_BYTES_NETWORK and timeout < MAX_TIMEOUT) {
      delay(100);  // adjust for network latency
    }
  } while (num < MIN_BYTES_NETWORK and timeout < MAX_TIMEOUT);
  return num;
}

asetyde, 2015-12-22:
also this define it's arbitrary decided 

#define MAX_CONNECT_ATTEMPTS 3
#define MAX_TIMEOUT          5
#define MIN_BYTES_NETWORK    4

or come from ... ?

asetyde, 2015-12-22:
@igrr  we have new rewritten function but block it persist 

boolean Connector::wait_for_data() {
	bool data_available = client.available();
	int milliseconds = 0;

	// Check to see if data is available right away, don't wait.
	if (data_available) {
		return true;
	}
	// No data, so we wait.
	while ((!data_available) && (milliseconds < DATA_TIMEOUT)) {
		data_available = client.available();
		if (data_available) {
			return true;
		}
		else {
			delay(WAIT_INTERVAL);
			milliseconds += WAIT_INTERVAL;
		}
	}
	print_message(WIFI_TIMEOUT, true);
	return false;
}
#endif

/**
* read_packet - Read a packet from the server and store it in the buffer
*
* This method reads the bytes sent by the server as a packet. All packets
* have a packet header defined as follows.
*
* Bytes                 Name
* -----                 ----
* 3                     Packet Length
* 1                     Packet Number
*
* Thus, the length of the packet (not including the packet header) can
* be found by reading the first 4 bytes from the server then reading
* N bytes for the packet payload.
*/
void Connector::read_packet() {
	byte local[4];

	if (buffer != NULL)
		free(buffer);

#if not defined WIFI
	// Wait for client (the server) to send data
	wait_for_client();
#endif

	int avail_bytes = wait_for_client();
	while (avail_bytes < 4) {
		avail_bytes = wait_for_client();
	}

	// Read packet header
	for (int i = 0; i < 4; i++) {
#if defined WIFI
		// Wait for client. Abort if no data after TIMEOUT_DATA milliseconds
		if (!wait_for_data()) {
			return;
		}
#endif
		local[i] = client.read();
	}

	// Get packet length
	packet_len = local[0];
	packet_len += (local[1] << 8);
	packet_len += ((uint32_t)local[2] << 16);
#if not defined WIFI
	// We must wait for slow arriving packets for Ethernet shields only.
	avail_bytes = wait_for_client();
	while (avail_bytes < packet_len) {
		avail_bytes = wait_for_client();
	}
#endif
	// Check for valid packet.
	if (packet_len < 0) {
		print_message(PACKET_ERROR, true);
		packet_len = 0;
	}
	buffer = (byte *)malloc(packet_len + 4);
	if (buffer == NULL) {
		print_message(MEMORY_ERROR, true);
		return;
	}
	for (int i = 0; i < 4; i++)
		buffer[i] = local[i];

	for (int i = 4; i < packet_len + 4; i++) {
#if defined WIFI
		// Wait for client. Abort if no data after TIMEOUT_DATA milliseconds
		if (!wait_for_data()) {
			return;
		}
#endif
		buffer[i] = client.read();
	}
}

asetyde, 2015-12-23:
@igrr  my code it work on stable 1.6.5 but not on 2.0.0 stable .. some freeze random . we wait for stable 2.1.0 or staging ..i see many correction ... i not use git

closed 2016-10-28 by asetyde