esp8266/Arduino #1234: SoftwareSerial buffer > 255 corrupts data

supersjimmie, 2015-12-17:
I have a serial device that sends roughly 1200 bytes every 10 sec.
During waiting for that, other code needs to keep running, so I need to buffer it all and process it in one call. I then check every 15 sec to fetch all buffered data from the serial buffer.

Example with just a delay to fill the buffer, based on the SoftwareSerial example file:
#include <SoftwareSerial.h>

SoftwareSerial swSer(14, 12, true, 250);

void setup() {
  Serial.begin(115200);
  swSer.begin(115200);
  Serial.println("\nSoftware serial test started");

  for (char ch = ' '; ch <= 'z'; ch++) {
    swSer.write(ch);
  }
  swSer.println("");

}

void loop() {
  delay(15000);
  while (swSer.available() > 0) {
    Serial.write(swSer.read());
  }
  while (Serial.available() > 0) {
    swSer.write(Serial.read());
  }
}

As long as I use SoftwareSerial swSer(14, 12, true, 250);  all goes fine but I lose a lot of data.
(yes, I use inverted = true. that works fine, thank you for that!)
0-1:24.2.1(151217110000W)(00700.778*m3)
0-1:24.4.0(1)
!12C9

When going to a buffer of 260 or higher (need about 1200), data gets corrupted.
¡48224.2.1(151217110000W)(00700.778*m3)
0-1:24.4.0(1)
H&T¦Hø
      E72

This is not just because of bad connection. I have tested many times with higher and lower.
Without the delay, nearly no corruption at all. So my data transfers are good.

me-no-dev, 2015-12-17:
Software Serial libraries are not good for the ESP. They are made to be used on real-time MCUs and because of the network, the sketch is not exactly real-time.
For your luck though, I just happened to write a couple of libraries yesterday for @celsworth that use a timer to read/write the data. There are two libraries, one had both TX and RX and the other one is just RX that is meant to complement Serial1 which can only send. That is the best possible scenario and it gives you full duplex Serial with just about any baudrate you need (below 230400 bps)
Here is the pull request that is pending for Merge: https://github.com/esp8266/Arduino/pull/1232

Use:
```cpp
#include "ESPSoftRx.h"
#define RX_PIN 4 //any pin you want but 16

void begin(){
  Serial.begin(115200);
  Serial1.begin(115200);
  SerialRx.begin(115200);
}

void loop(){
  // Serial -> Serial1
  while(Serial.available()) Serial1.write(Serial.read());
  // SerialRx -> Serial
  while(SerialRx.available()) Serial.write(SerialRx.read());
}
```

me-no-dev, 2015-12-17:
oh that and the buffers are smaller than what you want ;)
you need to read the serials more often and buffer the large data yourself, then parse it when you want.

supersjimmie, 2015-12-17:
I don't think my problem has to do with real-time issues. I'm also vey convenient with the possibility to use new the INVERTED option in plerup's library.

It just seems that the buffer cannot exceed 256 bytes and/or something gets corrupted when I use a larger buffer size.

But is this the right place to put an issue with the SoftwareSerial library of 'plerup' anyway?

me-no-dev, 2015-12-17:
I would not really consider this a bug because in order for your particular sketch to work, the buffer that SoftwareSerial or any other Stream uses needs to be larger than what you expect as data, because you are not polling the buffer while waiting and there is nowhere for the new bytes to be put.
Depending on the particular buffer implementation that might cause overwrite of previous unread data (overflow), might not put the bytes in at all or any other unhandled situation.

If you read the serials often and put their contents in your own buffer that can hold the amount you expect, you should be set. If it still corrupts, then its related to what I explained and deleted.

supersjimmie, 2015-12-17:
I did not call it a bug, I only said that data gets corrupted when I set the buffer to >255 bytes.
If there is a way to make that work different I'll be happy, if not than I just know.

The sending device sends a whole message of 600-1200 bytes at once with 115k2 bps, without my code knowing when exactly. 

Because there are other things to do that take some time, I cannot just check the serial and also process everything else within every 7ms (which is about the time to send the data), I need a buffer for the entire message.

me-no-dev, 2015-12-17:
timing at such hight speeds is really critical. I have not yet spent time on INVERTED or stop bits or parity because it's unrelated to the way the Soft Serial works. I know how much optimization I needed to get stable at 115200 and if his code misses the stop bit somewhere, it will not trigger correctly on the next start bit if it follows right after (multiple bytes transmitted) and that error will carry on until there is enough time between transmitted bytes to read the current error and start waiting for new start bit.
My library suffered from the same problem until I put it all in RAM and optimized every line (maybe should use the pin interrupt exclusively) now it runs pretty stable even at 230400 and I can optimize some more.

supersjimmie, 2015-12-17:
But how does timing relate to just setting the buffer to 250 or 260 bytes then?
And about the inverted, I have the same issue when I use hardware inverting. (and also not with a 250b buffer and it occurs at 260b buffer).

Based on the 250-260 difference, I suspect that is has something to do with some uint8 or so that overflows while it is able to setup with more.

me-no-dev, 2015-12-17:
yeah, I thought it might be that and read through the library code but did not find any uint8_t that can be the reason. Maybe I missed something?

me-no-dev, 2015-12-17:
https://github.com/plerup/espsoftwareserial/blob/master/SoftwareSerial.cpp#L154
should be 
```cpp
if (next != m_outPos) {
```

supersjimmie, 2015-12-17:
Ah well that does not solve my primary issue, but it does solve the corruption.
Now even when I setup a large buffer, it still only buffers 255 max.

me-no-dev, 2015-12-17:
so you do SoftwareSerial swSer(14, 12, true, 1200); and you still can not read more than 255?
what does Serial.printf("Available: %d\n", swSer.available()); yield?

supersjimmie, 2015-12-17:
My mistake, I forgot to change the buffer again.
When I receive one message, it says: Available: 654  (my max 1250 is including possible garbage, partial double message etc)

The data is still corrupted so it seems the change from m_inPos to m_outPos did not solve that.

But when I let Notepad++ count all characters after that, it's 667.

closed 2016-03-03 by supersjimmie