esp8266/Arduino #1237: OTA update fails at every 2nd attend: no response from server

NL-Splinter, 2015-12-17:
Hi all,
ESP8266 OTA is working fine for me the first time I power on the ESP12E: I can upload a sketch using the Arduino IDE, ESP reboots and starts the (new) sketch. But then any second attempt to OTA update a sketch consistently gives an IDE error: No response from device. This is all using the same sketch (initial start, 1st update and 2nd update). It seems ESP.restart() doesnt do something a hard (power) restart does do. Everytime I power on/off the ESP, its back to square 1: first OTA fine, second gives error. Is this a known issue? Thanks and regards, Jan Hein

me-no-dev, 2015-12-28:
yes. it's something in the bootloader that reboots to the wrong mode after serial flash. If you hit the reset after you upload the sketch through the serial, then OTA will not fail

devyte, 2017-10-19:
Closing as known issue (no real solution known at this time).

closed 2017-10-19 by devyte