esp8266/Arduino #1238: get.py AttributeError

breemts, 2015-12-17:
Hi,
I tried to install it by pulling from git. When I run python get.py I get:
```
Platform: i686-mingw32
Downloading esptool-0.4.6-win32.zip
Traceback (most recent call last):
  File "get.py", line 109, in <module>
    get_tool(tool)
  File "get.py", line 71, in get_tool
    urllib.urlretrieve(url, local_path, report_progress)
AttributeError: module 'urllib' has no attribute 'urlretrieve'
```
My path is: somepath\Arduino\hardware\esp8266com\esp8266\tools I think this is correct.
Is there a problem with mingw? Should I use something else like cygwin?
Thanks

igrr, 2015-12-17:
Does it work if you run it from Windows command prompt (without MinGW)?

breemts, 2015-12-17:
This is the output I get from the Windows command. 
Steps I did:
1. Clone Git into ..\Arduino\hardware\esp8266\esp8266 folder
2. Open cmd as Admin
3. cd intp ...\esp8266\tools
4. python get.py
No idea why in the error message is `Platform: i686-mingw32`

tablatronix, 2015-12-17:
Same problem here
python 351 32bit installer
maybe a path issue, or missing lib ?

tablatronix, 2015-12-17:
I installed 2.7 and it seemed to work, I noticed fix_urllib.py contains `urlretreive` so maybe it was deprecated.

still didnt fix my compile error tho...
`1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++": file does not exist`
:disappointed:

breemts, 2015-12-17:
Thanks for the hint. Haven't seen that there is a difference between python 3.5 and 2.7. 
Works fine with python 2.7. 
Now I can use this with an older Arduino Studio but with autocomplete. If someone is interested: [link](http://iot.criativasoft.com.br/arduino/)

closed 2015-12-17 by breemts

HugoSH, 2017-12-04:
I have Python 2.7.5 under Windows 7 (64)  and have the same error as above. 
Platform: i686-mingw32
Downloading esptool-0.4.12-win32.zip
Traceback (most recent call last):
  File "get.py", line 116, in <module>
    get_tool(tool)
  File "get.py", line 75, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "C:\Python27\lib\urllib.py", line 94, in urlretrieve
    return _urlopener.retrieve(url, filename, reporthook, data)
  File "C:\Python27\lib\urllib.py", line 240, in retrieve
    fp = self.open(url, data)
  File "C:\Python27\lib\urllib.py", line 208, in open
    return getattr(self, name)(url)
  File "C:\Python27\lib\urllib.py", line 437, in open_https
    h.endheaders(data)
  File "C:\Python27\lib\httplib.py", line 969, in endheaders
    self._send_output(message_body)
  File "C:\Python27\lib\httplib.py", line 829, in _send_output
    self.send(msg)
  File "C:\Python27\lib\httplib.py", line 791, in send
    self.connect()
  File "C:\Python27\lib\httplib.py", line 1176, in connect
    self.sock = ssl.wrap_socket(sock, self.key_file, self.cert_file)
  File "C:\Python27\lib\ssl.py", line 387, in wrap_socket
    ciphers=ciphers)
  File "C:\Python27\lib\ssl.py", line 143, in __init__
    self.do_handshake()
  File "C:\Python27\lib\ssl.py", line 305, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [Errno 1] _ssl.c:504: error:140770FC:SSL routines:
SSL23_GET_SERVER_HELLO:unknown protocol