esp8266/Arduino #1241: "/bin/xtensa-lx106-elf-g++": file does not exist  -using staging

tablatronix, 2015-12-17:
exec: "/bin/xtensa-lx106-elf-g++": file does not exist

why is it using relative pathing , i have no idea where it is looking, tried running filemon, and didn't find it.

```
Arduino: 1.6.6 (Windows 7), Board: "Adafruit HUZZAH ESP8266, 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

C:\bin\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\bin\Arduino\hardware" -hardware "C:\Users\USERNAME\AppData\Local\Arduino15\packages" -tools "C:\bin\Arduino\tools-builder" -tools "C:\bin\Arduino\hardware\tools\avr" -tools "C:\Users\USERNAME\AppData\Local\Arduino15\packages" -built-in-libraries "C:\bin\Arduino\libraries" -libraries "C:\Users\USERNAME\Documents\projects\arduino\sketches\libraries" -fqbn=esp8266:esp8266:huzzah:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10606 -build-path "C:\Users\USERNAME\AppData\Local\Temp\build5b7afb42f1d9d91f581a9217a8885356.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\USERNAME\Documents\projects\arduino\sketches\AutoConnect\AutoConnect.ino"
C:\bin\Arduino\arduino-builder -compile -logger=machine -hardware "C:\bin\Arduino\hardware" -hardware "C:\Users\USERNAME\AppData\Local\Arduino15\packages" -tools "C:\bin\Arduino\tools-builder" -tools "C:\bin\Arduino\hardware\tools\avr" -tools "C:\Users\USERNAME\AppData\Local\Arduino15\packages" -built-in-libraries "C:\bin\Arduino\libraries" -libraries "C:\Users\USERNAME\Documents\projects\arduino\sketches\libraries" -fqbn=esp8266:esp8266:huzzah:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10606 -build-path "C:\Users\USERNAME\AppData\Local\Temp\build5b7afb42f1d9d91f581a9217a8885356.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\USERNAME\Documents\projects\arduino\sketches\AutoConnect\AutoConnect.ino"
"/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\USERNAME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc2\cores\esp8266" "-IC:\Users\USERNAME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc2\variants\adafruit" "C:\Users\USERNAME\AppData\Local\Temp\build5b7afb42f1d9d91f581a9217a8885356.tmp\sketch\AutoConnect.ino.cpp"
exec: "/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling.
```

closed 2015-12-17 by tablatronix

tablatronix, 2016-03-31:
for reference
I installed stable into esp8266com/
oops

pabloevaristo, 2017-01-20:
@tablatronix , did you solve that error??

tablatronix, 2017-01-20:
Yeah I installed stable into the path for dev, i guess its hardcoded somewhere to esp8266 and not esp8266com/

Sheep666, 2017-09-17:
@tablatronix  Hello,I also got this issue.
Can u tell how to solve it?
Appreciated!

jostidetosti, 2017-09-24:
@tablatronix @Sheep666  Same here.

KevinLod, 2018-05-04:
Hello all, 
Im taking the following message from the IDE ... When Im trying to compile a code related to the ESP8266 wifi module

Arduino: 1.8.5 (Windows 8.1), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\mohamed  zakaria\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=512K0,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -vid-pid=0X2A03_0X0043 -ide-version=10805 -build-path C:\Users\MOHAME~1\AppData\Local\Temp\arduino_build_963388 -warnings=none -build-cache C:\Users\MOHAME~1\AppData\Local\Temp\arduino_cache_859932 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.esptool.path=C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12 -verbose C:\Users\mohamed  zakaria\Desktop\sketch_may04a\sketch_may04a\sketch_may04a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\mohamed  zakaria\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=512K0,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -vid-pid=0X2A03_0X0043 -ide-version=10805 -build-path C:\Users\MOHAME~1\AppData\Local\Temp\arduino_build_963388 -warnings=none -build-cache C:\Users\MOHAME~1\AppData\Local\Temp\arduino_cache_859932 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.esptool.path=C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12 -verbose C:\Users\mohamed  zakaria\Desktop\sketch_may04a\sketch_may04a\sketch_may04a.ino
Using board 'generic' from platform in folder: C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Using core 'esp8266' from platform in folder: C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Detecting libraries used...
"C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\MOHAME~1\AppData\Local\Temp\arduino_build_963388/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC"  -DESP8266 "-IC:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\generic" "C:\Users\MOHAME~1\AppData\Local\Temp\arduino_build_963388\sketch\sketch_may04a.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\mohamed  zakaria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\MOHAME~1\AppData\Local\Temp\arduino_build_963388/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=8 [... cut ...] ants\generic" "C:\Users\MOHAME~1\AppData\Local\Temp\arduino_build_963388\sketch\sketch_may04a.ino.cpp" -o "C:\Users\MOHAME~1\AppData\Local\Temp\arduino_build_963388\preproc\ctags_target_for_gcc_minus_e.cpp"
exec: "C:\\Users\\mohamed zakaria\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling for board Generic ESP8266 Module.

I've suffered so much trying to solve this problem ,,, advice ??

samshum2, 2020-01-27:
Any body can help?

samshum2, 2020-01-27:
exec: "/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling.

d-a-v, 2020-01-27:
```
cd tools
./get.py
```

Artemis-Wynk, 2020-01-29:
@samshum2  I'm getting the exact same error message on OSX with the latest IDE and ESP-12E. What's the solution?

samshum2, 2020-02-07:
ok thanks.

RAKINANWAR, 2021-03-07:
> 
> 
> exec: "/bin/xtensa-lx106-elf-g++": file does not exist
> Error compiling.

i am also having same issues......can you help me please

Abi-sha, 2021-03-13:
> Yeah I had accidently installed stable into the path for dev, i guess its hardcoded somewhere to esp8266 and not esp8266com/

can u explain more

BassemKattan, 2021-05-01:
SOLVED!

Backup your arduino folder so you don't lose your code.
Uninstall Arduino
Download Arduino IDE version 1.8.4 (Haven't tested higher versions yet...Need to test one by one to figure out where they broke it)
Installing with Boards Manager

Starting with 1.6.4, Arduino allows installation of third-party platform packages using Boards Manager. We have packages available for Windows, Mac OS, and Linux (32 and 64 bit). You can do this from the website by selecting previous versions...

Start Arduino and open Preferences window.
Enter http://arduino.esp8266.com/stable/package_esp8266com_index.json into Additional Board Manager URLs field. You can add multiple URLs, separating them with commas.
Open Boards Manager from Tools > Board menu and install esp8266 platform (and don't forget to select your ESP8266 board from Tools > Board menu after installation).

At this point i was able to compile example projects successfully

svenjie, 2021-05-13:
> SOLVED!
>
> Backup your arduino folder so you don't lose your code.
> Uninstall Arduino
> Download Arduino IDE version 1.8.4 (Haven't tested higher versions yet...Need to test one by one to figure out [...]

Did exactly as above per BassemKattan and still getting 
`exec: "/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling for board Adafruit Feather HUZZAH ESP8266.`

SUAul, 2021-05-26:
I have the same problem that @svenjie help :(

mmasell, 2022-04-24:
I have the same problem and I already installed Arduino version 1.8.19, now what?

aferlazzo, 2022-09-03:
> I have the same problem and I already installed Arduino version 1.8.19, now what?

Me too. Any solution?