esp8266/Arduino #1243: Wemos d1 mini pin mapping

danielmawhirter, 2015-12-18:
(using 2.0.0 package manager version)
The Wemos D1 mini's pins do not map to arduino pin numbers accurately. Here is the mapping as I have found it so far (using pinmode(), digitalWrite()'s, and a mega for analogRead()'s):
normal status: `d0=low, d1=high, d2=low, d3=high, d4=high, d5=high, d6=high, d7=high, d8=low`
After running the test with "10" pin d1 appears to have become normally low. Not a clue why, but I have eliminated those "d1 low" results below.
arduino pin -> result/connection
>0 -> d3
1 -> nc
2 -> d4 = led
3 -> nc
4 -> d2
5 -> d1
6 -> watchdog(4) reset
7 -> watchdog(4) reset
8 -> watchdog(4) reset
9 -> watchdog(4) reset
10 -> nc
11 -> watchdog(4) reset
12 -> d7 low, d6 sporadic
13 -> d0 high, d7 sporadic
14 -> d0 high, d6 low, d7 low, d5 sporadic
15 -> d0 high, d8 sporadic
16 -> d0
17 -> nc
18 -> nc
19 -> nc
20 -> stack trace and pin-based(2) reset
21 -> stack trace and watchdog(4) reset
22 -> stack trace and watchdog(4) reset
23 -> stack trace and watchdog(4) reset
24 -> nc
25 -> nc
26 -> nc
27 -> nc
28 -> nc
29 -> nc

igrr, 2015-12-18:
Wemos D1 mini pin mapping is available here:
https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h#L49-L61
You should be able to use `D0` – `D10`, `D13` identifiers in your sketch.

danielmawhirter, 2015-12-18:
So it is, I was mistakenly trying the lowercase variants and fell back to raw numbers. Thank you for your help!

closed 2015-12-18 by danielmawhirter

wemos, 2015-12-20:
@danielmawhirter 
more:
http://www.wemos.cc/wiki/doku.php?id=en:d1_mini#pin

danielmawhirter, 2015-12-20:
Thanks, glad to see more documentation and shields coming up!

wemos, 2015-12-20:
thank you for your supporting.

Bushii7, 2017-09-01:
please check this forum .. i need some solution :( urgent 
i have wemos esp8266 nd audiocodec MIKRO E.
https://communities.intel.com/message/495988#495988

suculent, 2017-09-01:
Sorry, mate. You’re heavily off-topic.

However, quick Google search for WM8731 library does it:

https://github.com/hughpyle/machinesalem-arduino-libs/tree/master/WM8731

Bushii7, 2017-09-01:
question was actually for WEMOS 
need an audio codec library compatible  with wemos D1 mini...