esp8266/Arduino #1246: significant increase flash usage after update sdk to 1.5

reaper7, 2015-12-18:
I noticed, that after changes sdk to 1.5
(examples compiled on esp8266 Arduino github version for day 17.12.2015 VS  18.12.2015 with new 1.5 SDK)
flash usage increase ~18k but reducing RAM is only ~1.3-1.4k

Examples from core libs:
ArduinoOTA->BasicOTA (old/new) from 222186 to 240868 (RAM 34348 to 33072) 
ESP8266WebServer->HelloServer (old/new) from 228758 to 247050 (RAM 37712 to 36304) 

Other examples:
WifiManager->AutoConnect (old/new) from 235826 to 254188 (RAM 37724 to 36344) 

My example (uart camera + WifiManager + ArduinoOTA)
(old/new) from 245966 to 264604 (RAM 41136 to 39784) 

This increase of flash usage is related to sdk or maybe some compilator options?

Yesterday I was happy that all my sketches fits to 512k ESP version via OTA but with new SDK it is no longer possible :(

igrr, 2015-12-18:
I tried to calculate symbols size difference and got the following:
https://gist.github.com/igrr/28fa97e8ca56f8621e80
Note the net decrease in total symbol size.
Here's the script which i used:
https://gist.github.com/igrr/503c58dc537cb470f336
Symbol list files were obtained using this command:
```
xtensa-lx106-elf-nm -f posix sketch.ino.elf > /tmp/sym-1.x
```
I used HTTPSRequest sketch to do the test.

Will look into this again tomorrow.

igrr, 2015-12-20:
Okay, so my test missed the fact that only global symbols are listed in `nm` output. Local (static) symbols are not visible in the final executable but they still add up to sketch size.
I'm not sure what we can do to reduce final code size in this case.

reaper7, 2015-12-20:
really bad news :(
I tested 1.3/1.41/1.5 SDK's on CHERTS DevKit -> "blinky" example
and results are unfortunately similar:
1.3:
```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000350  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       000000cc  3ffe8350  3ffe8350  00000430  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .bss          00006928  3ffe8420  3ffe8420  00000500  2**4
                  ALLOC
  3 .text         00006226  40100000  40100000  000004fc  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .irom0.text   000295bc  40240000  40240000  00006730  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8350|     848
    rodata|           ReadOnly Data (RAM)|    3FFE8350|    3FFE841C|     204
       bss|      Uninitialized Data (RAM)|    3FFE8420|    3FFEED48|   26920
      text|          Uncached Code (IRAM)|    40100000|    40106226|   25126
irom0_text|             Cached Code (SPI)|    40240000|    402695BC|  169404
------------------------------------------------------------------------------
Entry Point : 40100004 call_user_start()
Total Used RAM : 27972
Free RAM : 53948
Free IRam : 7642 or 24026 if 48k IRam
------------------------------------------------------------------------------
```
1.41:
```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000374  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       000000f0  3ffe8380  3ffe8380  00000460  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .bss          00006ad8  3ffe8470  3ffe8470  00000550  2**4
                  ALLOC
  3 .text         000067de  40100000  40100000  00000550  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .irom0.text   0002cbb4  40240000  40240000  00006d30  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8374|     884
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFE8470|     240
       bss|      Uninitialized Data (RAM)|    3FFE8470|    3FFEEF48|   27352
      text|          Uncached Code (IRAM)|    40100000|    401067DE|   26590
irom0_text|             Cached Code (SPI)|    40240000|    4026CBB4|  183220
------------------------------------------------------------------------------
Entry Point : 40100004 call_user_start()
Total Used RAM : 28476
Free RAM : 53444
Free IRam : 6178 or 22562 if 48k IRam
------------------------------------------------------------------------------
```
1.5:
```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000374  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00000100  3ffe8378  3ffe8378  00000458  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .bss          00006230  3ffe8478  3ffe8478  00000558  2**4
                  ALLOC
  3 .text         00006732  40100000  40100000  00000558  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .irom0.text   0002d494  40240000  40240000  00006c90  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8374|     884
    rodata|           ReadOnly Data (RAM)|    3FFE8378|    3FFE8478|     256
       bss|      Uninitialized Data (RAM)|    3FFE8478|    3FFEE6A8|   25136
      text|          Uncached Code (IRAM)|    40100000|    40106732|   26418
irom0_text|             Cached Code (SPI)|    40240000|    4026D494|  185492
------------------------------------------------------------------------------
Entry Point : 40100004 call_user_start()
Total Used RAM : 26276
Free RAM : 55644
Free IRam : 6350 or 22734 if 48k IRam
------------------------------------------------------------------------------
```
so IROM0 is bigger ~16k (1.5 vs 1.3)

closed 2016-02-29 by igrr