esp8266/Arduino #1247: Arduino 1.6.7 + stable 2.0.0 - not working combination yet

mozgy, 2015-12-18:
Official 1.6.7 arduino is out and installing stable 2.0.0 on top of that, does not produce working environment ->

Board generic (platform esp8266, package esp8266) is unknown
Error compiling.

mozgy, 2015-12-18:
Git copy works with warning ->
Board esp8266com:esp8266:wifinfo doesn't define a 'build.board' preference. Auto-set to: ESP8266_WIFINFO

akmandala, 2015-12-28:
Hi, I would like to run latest ESP8266 Arduino with Arduino 1.6.7, and finish installing Git Version into Hardware directory. But can you help me how to tell Arduino 1.6.7 to pick up the Git Version? In my case restarting Arduino does not automatically included on board manager.

mtnbrit, 2016-01-13:
+1

I have cloned the git version into ~/Library/Arduino15/packages/esp8266/hardware/esp8266com/esp8266

but as @akmandala I don't see anything but the Arduino AVR boards under Tools->Board

akmandala, 2016-01-13:
In my case, my mistake as I have two Arduino installation (one binary copy
and one in Program Files). As soon as I clone to the right copy then it was
ok. Also check Board Manager.

closed 2016-02-27 by igrr