esp8266/Arduino #1250: ESP8226HTTPUpdate

adrionics, 2015-12-18:
After having some updates corrupted when the server returned an unexpected response on a server error, I made some changes to the update function in ESP8226HTTPUpdate.cpp so the error response was not written to the ESP and treated like an update.

I added a boolean parameter to the handleUpdate function in ESP8266HTTPUpdate called "requireMD5" with a default value of false.
Then, in the handleUpdate function inside ESP8266HTTPUpdate.cpp itself,
```
    if(http->hasHeader("x-MD5")) {
        DEBUG_HTTP_UPDATE("[httpUpdate]  - MD5: %s\n", http->header("x-MD5").c_str());
    } else if (requireMD5) {
        DEBUG_HTTP_UPDATE("[httpUpdate]  - MD5 not found, aborting...\n");
        http->end();
        return HTTP_UPDATE_FAILED;
    }

``` 
You may consider adding something like this to the main code base!

mtnbrit, 2016-01-13:
Hello, is there a how-to in the works for creating the MD5 hash and using it to verify the bin file integrity when doing ESP8266HTTPUpdate? Can I assume there would be additions to the advanced updater server-side php script?

adrionics, 2016-01-13:
This is VBScriot I used on the server: http://stackoverflow.com/a/31453654/1022455

Links2004, 2016-01-13:
take a look her:
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#server-request-handling
there is a sample php script

mtnbrit, 2016-01-14:
Thanks, didnt realize the script already supported md5, cool!

I am getting an error when attempting httpUpdate, as follows. I added debug lines to show the content of buf[0] etc, as you can see it appears tcp->peekBytes is not returning the head of the file properly. I have confirmed the file does in fact start with 0xE9.

im running 1.6.7 with the git version.

    :wcs ra 43Receiving OTA from /v2/ota.php
    :wr
    :sent 101
    :ww
    :ref 1
    :wr
    :sent 334
    :ww
    :rn 1460
    [httpUpdate] Header read fin.
    [httpUpdate] Server header:
    [httpUpdate]  - code: 200
    [httpUpdate]  - len: 343104
    [httpUpdate]  - MD5: c4ee9d3ebff1d75df4145265cf349624
    [httpUpdate] ESP8266 info:
    [httpUpdate]  - free Space: 700416
    [httpUpdate]  - current Sketch Size: 344520
    :abort
    :ur 1
    :del
    :rch 1460, 1460
    :rch 2920, 1460
    :rch 4380, 1460
    [httpUpdate] runUpdate flash...
    :pd 4, 5840, 283
    :rpi 1177, 4
    [httpUpdate]buf[0]: 3C
    [httpUpdate]buf[1]: 62
    [httpUpdate]buf[2]: 72
    [httpUpdate]buf[3]: 20
    [httpUpdate] magic header not starts with 0xE9
    :ur 1
    :close
    :del
    HTTP_UPDATE_FAILED Error (-106): Verify bin header failed

Links2004, 2016-01-14:
```3C 62 72 20``` is the same as ```<br```
i am sure html is not what you want to get, check the php/apache error log, there is something going wrong.

mtnbrit, 2016-01-14:
Thanks, the php was returning an error about the version header not being specified. Its working now.

closed 2016-02-27 by igrr