esp8266/Arduino #1251: Board d1 (platform esp8266, package esp8266) is unknown

webwarrior, 2015-12-19:
Installed via the Board manager on Arduino 1.6.7

Can't compile any of the samples.

I used this for the board manager to get the wemos D1 board working:
http://www.wemos.cc/versions/77428ba/package_esp8266com_index.json

Am I better off with:
http://arduino.esp8266.com/stable/package_esp8266com_index.json ?

As the Wemos D1 wifi board esentially has NodeMCU thingy on it...

What should I do?

igrr, 2015-12-20:
Are you getting this error when you are using boards manager package from wemos.cc or from arduino.esp8266.com?

webwarrior, 2015-12-20:
Hi,

I get it when I install the board manager using the link provided by wemos.cc

Does that make sense?

igrr, 2015-12-20:
If you get an error with a link provided by wemos.cc, then perhaps you should raise an issue with them?

That said, we do have "Wemos D1" board option in the latest 2.0.0 stable version (`http://arduino.esp8266.com/stable/package_esp8266com_index.json`). Compiling sketches with this board selected in "Tools > Board" menu seems to work.

webwarrior, 2015-12-20:
I just installed the v 2.0.0 stable version you mentioned on my Mac and that seemed to work and compile.

Don't have the board here with me to upload it to test it.

Where do I take up the issue with wemos.cc?

It would probably help them if I can help them solve bugs like these.

Why do they not provide your link, rather than theirs, as you do have the D1 and D1 Mini boards in your list?

Thanks for the help!

Shaun O'Reilly

closed 2015-12-20 by webwarrior

igrr, 2015-12-20:
The first line on http://www.wemos.cc/wiki/doku.php?id=en:arduino_get_started says:

    The WeMos D1 & D1 mini is join to esp8266/Arduino now

I guess this is supposed to mean that these boards are supported in our project now. Perhaps @wemos can clarify this further.

webwarrior, 2015-12-20:
I will take it up with them, as their link installs esp8266/1.6.5-20151115-77428ba/libraries/esp8266:

This will help other noobs like me in the future to get the correct libraries.

You are probably the most helpful person in the entire Arduino space, I have seen your dedication in the other posts.

Well done!

webwarrior, 2015-12-20:
I suggested the change on the wiki page.

Hope that will help future visitors.

wemos, 2015-12-20:
thank you @webwarrior @igrr

wemos, 2015-12-20:
@igrr
I delete that word in my wiki.
sorry, my poor english.
I write that words in my wiki, just mean you can find D1 boards in esp8266/arduino

webwarrior, 2015-12-20:
you are welcome.

I have the "NodeMCU ESP8266 WiFi V1.0 Rev 2.0 Dev Board Lua" , as well as the "Based ESP8266 WeMos D1 WiFi UNO for Arduino Compatible Development Board". Both stopped allowing me to update them via the USB cable after I tested one of my sketches on both, think it had an overload bug to to a big json string.

I can now update the NodeMCU ESP8266 board luckily when I press the flash button, but not otherwise.

The D1 board does not have that feature, and I am unable to update it via the usb cable now.
Is there other software that can help me out of that situation?

webwarrior, 2015-12-20:
@wemos , I thought that sentence was useful for newcomers.

But if you do not need it there, it does not matter. Your english is not too bad!

wemos, 2015-12-20:
do you setup usb-uart driver  from my website

webwarrior, 2015-12-20:
Will do, thanks.

webwarrior, 2015-12-20:
@wemos, The problem just got bigger:
error: cannot access /dev/cu.usbmodem1421 - might be a brick now?

Where do I continue this thread?

esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 921600
	setting port from /dev/tty.usbserial to /dev/cu.usbmodem1421
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/wq/1r7wy_ms37lfgwf5d_vdz3pr0000gn/T/build3b90184a28127009d57a138389f76d18.tmp/Blink.ino.bin success
opening port /dev/cu.usbmodem1421 at 921600
error: cannot access /dev/cu.usbmodem1421

error: espcomm_open failed

wemos, 2015-12-20:
do you use mac osx?

webwarrior, 2015-12-20:
yes

webwarrior, 2015-12-20:
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 921600
	setting port from /dev/tty.usbserial to /dev/cu.usbmodem1421
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /Users/shaunoreilly/Dropbox/Arduino_Dev/nodemcu_float_0.9.6-dev_20150704.bin success
opening port /dev/cu.usbmodem1421 at 921600
error: cannot access /dev/cu.usbmodem1421

error: espcomm_open failed

webwarrior, 2015-12-20:
I can see the lights on the board - the "on" led and "SCK"

webwarrior, 2015-12-20:
Windows does pick up the usb port.
Will try it on there

wemos, 2015-12-20:
Can you find any port in Arduino IDE [Tools]->[port]?

wemos, 2015-12-20:
And what is your mac osx version?

webwarrior, 2015-12-20:
10.11.2

webwarrior, 2015-12-20:
On Windows I get this:

D:\Downloads\Arduino>esptool.exe  -vv -cd nodemcu -cb 921600 -cp COM4 -ca 0x00000 -cf nodemcu_float_0.9.6-dev_20150704.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 921600
        setting port from COM1 to COM4
        setting address from 0x00000000 to 0x00000000
        espcomm_upload_file
        stat nodemcu_float_0.9.6-dev_20150704.bin success
error: Failed to open COM4
error: espcomm_open failed

It definitely does not want to communicate over the USB port anymore.
Should the SCK LED be permanently on?

wemos, 2015-12-20:
where d1 board do you buy from?

webwarrior, 2015-12-20:
http://www.ebay.com.au/itm/181904401257?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

wemos, 2015-12-20:
sorry, @webwarrior, it may be not produce from wemos.cc
this my board:
[image]

wemos, 2015-12-20:
It may be some hardware error with CH340G IC or ESP-12F module

wemos, 2015-12-20:
Our board use new logo ESP-12F module from 2015-10
[image]

webwarrior, 2015-12-20:
It does not have that logo, so, I assume it is not yours....

webwarrior, 2015-12-20:
I will take it up with the ebay seller, thx.

wemos, 2015-12-20:
^_^

webwarrior, 2015-12-20:
He should not be using the wemos name in his ebay listing, as he can get in trouble for that.

wemos, 2015-12-20:
Thank you for your supporting.

webwarrior, 2015-12-20:
It is the least I can do for the IOT community....

webwarrior, 2015-12-20:
Is this another fake?

http://store.cutedigi.com/esp8266-based-arduino-uno-like-wifi-board-d1/

Shaun

wemos, 2015-12-20:
http://www.aliexpress.com/store/1331105

webwarrior, 2015-12-20:
Awesome! thx

webwarrior, 2015-12-21:
@wemos,  Did you ever make boards that did not have your logo at the back?

[image]

webwarrior, 2015-12-21:
[image]

wemos, 2015-12-21:
It's not my board.

webwarrior, 2015-12-21:
Cool, I have reported it with Ebay...

otuk, 2015-12-22:
I have the same problems, board looks the same.  Some times it uploads sketches sometimes not.  I also got it in ebay.

webwarrior, 2015-12-22:
@otuk, I reported the listing with eBay, and I believe I am going to get a refund for the fake board.
If you do the same, then we can nail the fake producing criminals....

It is extremely frustration when it does that... I just wish there was a way to set one pin high / low, to force a flash to memory...., like on the NodeMCU board with the "flash" button...

mikecrean, 2015-12-28:
Hi I have what looks like a genuine WeMos.cc board and can compile the Hello World example but when i try a down load I get this error.

Sketch uses 220,852 bytes (21%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 31,438 bytes (38%) of dynamic memory, leaving 50,482 bytes for local variables. Maximum is 81,920 bytes.
java.io.IOException: Cannot run program "C:\Arduino\arduino-1.6.6\hardware\esp8266com\esp8266/tools/esptool/esptool.exe": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:113)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:157)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1178)
	at processing.app.Sketch.exportApplet(Sketch.java:1152)
	at processing.app.Sketch.exportApplet(Sketch.java:1124)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2452)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 9 more

Any one with any input please.

webwarrior, 2015-12-28:
Sounds like the Port is busy, or the incorrect port is selected...

yd6njk, 2016-05-08:
[image]

Please help me, Im Using WeMos D1 Board like in picture, I have install : http://arduino.esp8266.com/stable/package_esp8266com_index.json 
In board manager I choose " WeMos D1 R2 & Mini " and get error

Build options changed, rebuilding all
In file included from c:\documents and settings\admin\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:63:0,

from C:\Documents and Settings\admin\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\admin\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\allocator.h:46:75: fatal error: bits/c++allocator.h: No such file or directory

 #include <bits/c++allocator.h> // Define the base class to std::allocator.

                                                                           ^

compilation terminated.

exit status 1
Error compiling for board WeMos D1 R2 & mini.

Please help me what must I do ? Thks

yd6njk, 2016-05-08:
Im no see WeMos D1 board at Board Manager, Just see " WeMos D1 R2 & Mini " and WeMos D1 (Retired). I have try twice but still get error code:
----------------------------------------------------------------------------------------------------------------------------------
Build options changed, rebuilding all
In file included from c:\documents and settings\admin\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:63:0,

from C:\Documents and Settings\admin\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\admin\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\allocator.h:46:75: fatal error: bits/c++allocator.h: No such file or directory

#include // Define the base class to std::allocator.

compilation terminated.

exit status 1
Error compiling for board WeMos D1 R2 & mini.

webwarrior, 2016-05-08:
Hi,

This is not made by WeMos, and is a dodgy version...

Mine got bricked and I could not reset it...

I filed a complain against Paypal and said the guy was selling fake boards
and got my money back.

I also went to all the ebay seller's listing of the board, and flagged them
as fake.

Where did you buy yours from?

Octowell, 2016-05-14:
Hi guys, recently i also got the same issue.

_**Board d1_mini (platform esp8266, package esp8266) is unknown**_

_**Error compiling for board WeMos D1 R2 & mini.**_

unfortunatelly, mine got to work fine before.
here is my scenario

i got 2 laptop, which is running on Win 7 ultimate 32bit and Win 7 pro 64bit
Both using Arduino IDE 1.6.9
Both using the same method to get started on arduino IDE. 
CH340G driver already installed and work fine in both laptop.
i can compile and upload basic sketch to it using the laptop with Win 7 32bit.
i just found out that my Wemos D1 R2 didn't seems to work on the laptop with Win 7 64bit OS.

not surer if the OS is the main problem.
but i will update if there is any significant improvement on solving this issue.

Cheers
Well

[image]
[image]

Octowell, 2016-05-14:
**SOLVED**

as i thought before, it is not the OS problem.
but i realized that my User folder naming which caused the problem.
as in Linux, a folder is suggested to be named without any spaces

Refering to this
[http://askubuntu.com/questions/398400/how-do-i-navigate-to-folders-with-spaces-in-their-names-i-get-no-such-file-or](url)

[image]

on the laptop i suspected before, i create a new User named "Other" to try if it work
[image]

and it works like a charm
[image]

trying to add double quotes and backslash to my current User folder doesn't work
so the simplest way to solve this is to create a new user with name that doesn't contain any spaces.

try it and good luck

Cheers
Well

PaulRB, 2016-08-14:
Hi. The fix above does not work for me. My sketchbook location is /home/paul/sketchbook and there are no spaces.

I use genuine D1 Mini which I bought from the Wemos shop on AliExpress.

I also get the error message
`Board d1_mini (platform esp8266, package esp8266) is unknown

Error compiling for board WeMos D1 R2 & mini.`
I am using Arduino IDE 1.6.10 on Ubuntu 14.0.4 LTS.
I have downloaded the boards package from http://arduino.esp8266.com/stable/package_esp8266com_index.json (v 2.3.0)

I was also unable to use IDE 1.6.7 for the same reason.

I still have IDE 1.6.5 installed and that will compile and upload to the Mini correctly. However, a sketch I am currently developing has problems and it has been suggested that I upgrade to the latest IDE.

krzychb, 2016-08-26:
Hi @PaulRB,

I am wondering if your are still facing `Board d1_mini (platform esp8266, package esp8266) is unknown` error on your Arduino IDE 1.6.7 / IDE 1.6.10 installations?

If so, have you tried https://github.com/esp8266/Arduino/issues/1387#issuecomment-204865028?

Exact folder location (for your Ubuntu 14.0.4 LTS or any other O/S) can be checked after opening _File > Preferences_ as described [here](http://www.instructables.com/id/Programming-Esp8266-12e-With-Arduino-IDE/?ALLSTEPS#step5).

I guess location of this folder may be different between your IDE 1.6.5 and IDE 1.6.7 / IDE 1.6.10 installations.

Krzysztof

amauridsilva, 2017-03-25:
amigo tenho o mesmo poblema nao atualizo a ide

stevekramer99, 2017-08-15:
Hi I can't upload to my new d1 mini from Arduino IDE 1.8.3.  I am wondering is this because am running Win10 64bit Pro and Arduino is installed in C:\Program Files (x86)\Arduino\ but the sketch directory is C:\Users\Perigalacticon\Documents\Arduino\sketches  ???  The error is "error: espcomm_upload_mem failed".  Thanks.

suculent, 2017-08-15:
Sketches folder is configurable in preferences. 

The one you have is communication error on upload. Try to reset shortly before upload starts, or erase_flash using esptool. 

Odesláno z iPhonu

BluecloudNine, 2018-02-27:
Can anyone help here? I'm getting this error:
[image]

signal: bad system call (core dumped)
Error compiling for board WeMos D1 R2 & mini.

kwikiel, 2018-10-26:
This solved the issue (for future reference)
https://arduino-esp8266.readthedocs.io/en/latest/faq/a04-board-generic-is-unknown.html