esp8266/Arduino #1252: UDP broadcast to 255.255.255.255 not working

nicolasbisi, 2015-12-19:
I'm trying to send a broadcast to the network but it's not working.

Here is the simple version of the code I'm using:
```
IPAddress broadcastIp(192, 168, 1, 255);
//IPAddress broadcastIp(255, 255, 255, 255);
udp.beginPacket(broadcastIp,udpPort);
udp.write("hi");
udp.endPacket();
```

If I use the first ip (192.168.1.255) it works fine and I receive the data in multiple servers.
If I use the 255.255.255.255, nothing happens.
I connected wireshark and notice that the packets are not leaving the esp8266, although it returns an ok code from endPacket().

Am I doing something wrong?

penfold42, 2015-12-19:
When sending a multicast packet, replace udp.beginPacket(addr, port) with udp.beginPacketMulticast(addr, port, WiFi.localIP()). 

Try that

nicolasbisi, 2015-12-19:
Hi penfold42. Thank you for your reply.
I just tried that. Still the same thing. It worked for 192.168.1.255 and It didn't for 255.255.255.255 (return 1, but doesn't send the message (verified with wireshark))
Any ideias?

penfold42, 2015-12-19:
can you see what wifi_get_broadcast _if() returns ?

Looking at the espressos API docs it (and it's set friend) specify AP, STA or both for which interface to send broadcasts to

nicolasbisi, 2015-12-19:
Hi penfold42,

I could not find any function named  wifi_get_broadcast or similar. Is that right?
Just in case, I tested changing to AP, STA and both and got the same result.

Background:
Not that it matters since the message is not leaving the board, but I'm trying to broadcast udp messages using the esp8266 and receiving them using a python udp server.

penfold42, 2015-12-19:
Sorry, should have been clearer - it's provided by the espressif API and there's no arduino wrapper for it.

nicolasbisi, 2015-12-20:
Hum, thats bad.
Thank you penfold42 anyway.

penfold42, 2015-12-20:
Not really bad - I'm sure with the right included .h you could call it.
Most of the wrappers are straightforward.

pgollor, 2015-12-20:
@nicolasbisi Do you want to transmit broadcast or multicast messages?

If you want to transmit broadcast messages you can try something like this:
* 192.168.255.255
* 192.255.255.255

Maybe this could enough for your project?

And please remember that the most routers do not route limited broadcasts (255.255.255.255).

nicolasbisi, 2015-12-20:
Hi pgollor,
I'm trying to broadcast.
Yes, it works with 192.168.1.255, but I wanted something more general, that would work in any router. I think Ill just apply the net-mask then...

Out of curiosity, why the routers do not route limited broadcasts? Isn't that a requirement for getting a dynamic IP?
I don't know about my router, and I can't do many experiments with it's settings, but I could send an broadcast to 255.255.255.255 using a python udp client. If that matters.
Thanks for the reply

penfold42, 2015-12-20:
You need to separate the 2 functions your. "Router" is performing.
1) The wireless access point part will pass both local subnet and "all 1s" broadcast to the local subnet.
2) The routing part will not forward either broadcast types to any other network. For most home users this "other network" is the Internet.

pgollor, 2015-12-21:
If you want to use only broadcasts with aaa.bbb.ccc.255 you can do something like this as work around:

```
IPAddress ip = WiFi.localIP();
ip[3] = 255;
```

This will work in each subnet where your esp9266 is.

penfold42, 2015-12-21:
I wouldn't recommend that.

You're assuming every subnet in the world has a /24 mask - which is not true.

penfold42, 2015-12-21:
Call WiFi.subnetMask(); and bit wise OR the inverse of it with the Ip address

pgollor, 2015-12-21:
Okay you are right. I meant: It will work for the most private networks...
But this is not the major issue.

nicolasbisi, 2015-12-21:
What I'm doing right now is using IPAddress ip(192.168.1.255). It works fine, but I wanted something more generic.
I'll end up using the WiFi.subnetMask() like penfold42 said.
I just thought that using 255.255.255.255 would be easier... Guess I was wrong.
Thank  you all for the reply.

nicolasbisi, 2016-01-23:
To whoever is still wondering, I solved my problem this way.

IPAddress broadcastIp;
broadcastIp = ~WiFi.subnetMask() | WiFi.gatewayIP();

broadcast will now have the broadcast address for the current network.

closed 2016-01-23 by nicolasbisi

micw, 2016-03-20:
Thank you very much for the example. Using the local network broadcast IP works great.

One thing is very strange: I have had UDP broadcasts on 255.255.255.255 working. I have several ESP-01 modules installed that broadcasts to 255.255.255.255. Today I had to change a sketch and broadcast stopped to work. Switching to 192.168.1.255 solved the issue.

Since I flashed the devices with 255.255.255.255 I have a new computer and have re-installed arduino and esp stuff (probably with newer versions). So I assume that some change broke support for broadcasting to 255.255.255.255

micw, 2016-03-20:
I tried to write code to reliable get the current broadcast ip but there are lot of corner cases (e.g. wifi diconnectes and reconencts with new subnet before my code recognized it) - resulting in lot of complicated code. So I decided to calculate the broadcast ip before each message I send...

claudeheintz, 2016-03-21:
It appears to me that the current SDK will only receive broadcast packets sent with an address matching the subnet mask set with WiFi.config or WiFi.softAPConfig.  Which means that a wildcard 255 in the network portion of the address is not accepted.  In order to receive limited broadcast packets sent to 255.255.255.255, the subnet mask would need to be zero.  I haven't tried to see if that is possible.  But I have tried 192.168.1.255 with sub-netmask 255.255.255.0 returned with DHCP.  Setting a static address starting with 192 and subnet mask of 255.0.0.0 allows packets to be broadcast on 192.255.255.255.  While this is not necessarily correct, it seems that the SDK is accepting broadcast (255) octets in the host portion of and address but not in the network portion, insisting that part of the address matches the network portion of the IP address of the ESP8266.

gerardwr, 2016-04-03:
Have the same problem here.

A sketch that broadcasts UDP packets to 255.255.255.255 was running fine several months, the UDP messages were received OK by UDP listeners on other computers in my network.

Now, suddenly it has stopped working ;-(
The packets are not received by my UDP listeners anymore.

I upgraded recently from version 2.0.0 to 2.1.0 using Boards Manager, so I guess this has caused the malfunction.

The workaround from @nicolasbisi works fine to calculate the subnet broadcast address, but in my opinion the original issue was "closed", but never "solved".

@nicolasbisi : If you agree that your issue was not "solved", could you "reopen" the issue please? Maybe the issue will be investigated further then, and possibly solved.

gerardwr, 2016-04-03:
Update from my previous post.

I just upgraded from Arduino 1.6.7 to Arduino 1.6.8 and the UDP broadcast to 255.255.255.255 is sent by the ESP8266 and received by my UDP listeners.

So it seems that the issue is now really "solved" using Arduino 1.6.8.

nicolasbisi, 2016-04-03:
@gerardwr Thanks for testing and letting us know.

DAVIDBETAN2000, 2017-09-20:
yo lo solucione con el siguiente comando
AT+SAVETRANSLINK=1,"192.168.1.255",8080,"UDP",8080
asi envio mensajes a todos los dispositivos en udp
y si quiero salir en el Hecules escribo +++ y vuelve al modo AT

devyte, 2017-09-20:
@DAVIDBETAN2000 me alegro, but wrong repo, this is not the AT command firmware.