esp8266/Arduino #1255: WDT reset with SD card

rephansu, 2015-12-19:
When creating a new file , need a lot of time for search the FAT for free clusters.
If there are a large number of files in SD card, might lead WDT reset

Trigger WDT timeout at here:
https://github.com/esp8266/Arduino/blob/master/libraries/SD/src/utility/SdVolume.cpp : line 58
```sh
  // search the FAT for free clusters
  for (uint32_t n = 0;; n++, endCluster++) {
    // can't find space checked all clusters
    if (n >= clusterCount_) return false;
    /// if( n %1024 == 0)delay(0); // Add dealy for WDT

    // past end - start from beginning of FAT
    if (endCluster > fatEnd) {
      bgnCluster = endCluster = 2;
    }
    uint32_t f;
    if (!fatGet(endCluster, &f)) return false;

    if (f != 0) {
      // cluster in use try next cluster as bgnCluster
      bgnCluster = endCluster + 1;
    } else if ((endCluster - bgnCluster + 1) == count) {
      // done - found space
      /// Serial.printf("result:%d , %d\n" , n , bgnCluster); // result: 371012 , 371014
      break;
    }

  }
```

It found space at 371012 time.

me-no-dev, 2015-12-22:
did delay(0); fix the issue?
if so can you try using optimistic_yield(10000); without the "if" statement.

devyte, 2017-10-19:
The referenced code is still in latest git.

devyte, 2017-11-20:
Adding yield or delay means the function can't be used in async code anymore.

@igrr I'm not sure what can be done here, and I don't have the hardware to test. Any ideas?

igrr, 2017-12-31:
Don't have hardware to test this at the moment either, moving to 2.5.0

devyte, 2018-12-01:
Note: if the code here must yield, the correct way is with a delay(0) as originally proposed. That will yield when in CONT, and do nothing when in SYS. In contrast, optimistic_yield() only skips yield when called too soon after loop() entry.

devyte, 2018-12-01:
This requires investigation that won't fit into 2.5.0, mostly due to lack of hardware for testing. Pushing milestone back.

devyte, 2019-03-01:
Assigning to @earlephilhower in view of the new SDFS implementation.

earlephilhower, 2019-07-21:
The code paths are modified and in a different file, but they're generally the same.  There are two methods in FatVolume.cpp which can iterate over all clusters on a card that need attention:

````
bool FatVolume::allocateCluster(uint32_t current, uint32_t* next)
bool FatVolume::allocContiguous(uint32_t count, uint32_t* firstCluster)
````

Wackypedia lists FAT32 potentially having 268,435,445 clusters, so iterating over them all might take a wee bit of time.  A practical example of a 16GB card (formatted w/Linux VFAT default options) shows just under 500,000 clusters (30441472 sectors / 64 (sectors/cluster)).

As a simple, brute-force way of handling the WDT I propose calling `delay(0)` every 5,000 clusters iterated over in these methods.

earlephilhower, 2019-07-21:
See https://github.com/earlephilhower/ESP8266SdFat/pull/3 for the proposed solution.  That repo needs to be merged, then a submodule update done here on the core to pull it in.

closed 2019-07-25 by earlephilhower